laxnpander/OpenREALM #1: There are some errors when roslaunch

KeoChi, 2018-07-30:
Hello, I have compiled the code. When I roslaunch *.launch, there are some error.
`[ INFO] [1532931506.689198874]: STAGE_NODE [pose_estimation]: Received output directory, set to:
	/home/keo/catkin_ws/src/OpenREALM/realm_ros/output/18-07-30_14-18-25
[ INFO] [1532931506.689523671]: STAGE_NODE [pose_estimation]: Received frame.
**[ERROR] [1532931506.696714002]: Client [/rviz] wants topic /realm/alexa/general/gnss_base to have datatype/md5sum [nav_msgs/OccupancyGrid/3381f2d731d4076ec5c71b0759edbe4e], but our version has [sensor_msgs/NavSatFix/2d3a8cd499b9b4a0249fb98fd05cfa48]. Dropping connection.**
[ INFO] [1532931506.716948517]: STAGE_NODE [pose_estimation]: Received frame.
2018-07-30 14:18:26.747 (   1.066s) [        2E9C0700]             loguru.hpp:2056     0| Logging to '/home/keo/catkin_ws/src/OpenREALM/realm_ros/output/18-07-30_14-18-25/pose_estimation/publisher.log', mode: 'a', verbosity: 9
2018-07-30 14:18:26.747 (   1.066s) [        2E9C0700]    pose_estimation.cpp:452      0| Successfully initialized pose_estimation publisher!
2018-07-30 14:18:26.754 (   1.072s) [        2E1BF700]    pose_estimation.cpp:114      0| Processing frame #1 with timestamp 1532931506189585469!
2018-07-30 14:18:26.754 (   1.072s) [        2E1BF700]    pose_estimation.cpp:166      0| Tracking frame #1 in visual SLAM...!
[ INFO] [1532931506.772847002]: Image #4, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0004.jpg
[ INFO] [1532931506.824812175]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1532931506.973024976]: Image #5, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0005.jpg
[ INFO] [1532931507.018816604]: STAGE_NODE [pose_estimation]: Received frame.
**2018-07-30 14:18:27.060 (   1.379s) [        2E1BF700]    pose_estimation.cpp:179   WARN| No tracking.**
2018-07-30 14:18:27.065 (   1.384s) [        2E1BF700] worker_thread_base.cpp:62       0| Thread 'Stage [pose_estimation]' has processed data. Time elapsed: 0.31 [s]
2018-07-30 14:18:27.148 (   1.466s) [        2E9C0700]    pose_estimation.cpp:505      0| Publishing pose of frame #1...
2018-07-30 14:18:27.166 (   1.484s) [        2E1BF700]    pose_estimation.cpp:114      0| Processing frame #2 with timestamp 1532931506390519880!
2018-07-30 14:18:27.166 (   1.484s) [        2E1BF700]    pose_estimation.cpp:166      0| Tracking frame #2 in visual SLAM...!
[ INFO] [1532931507.173294898]: Image #6, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0006.jpg
[ INFO] [1532931507.216468794]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1532931507.373471682]: Image #7, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0007.jpg
[ INFO] [1532931507.417505693]: STAGE_NODE [pose_estimation]: Received frame.
RH: 0.47038 SH: 1776.77
**2018-07-30 14:18:27.528 (   1.847s) [        2E1BF700]    pose_estimation.cpp:179   WARN| No tracking.**
2018-07-30 14:18:27.532 (   1.851s) [        2E1BF700] worker_thread_base.cpp:62       0| Thread 'Stage [pose_estimation]' has processed data. Time elapsed: 0.37 [s]
2018-07-30 14:18:27.552 (   1.870s) [        2E9C0700]    pose_estimation.cpp:505      0| Publishing pose of frame #2...
[ INFO] [1532931507.573896666]: Image #8, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0008.jpg
[ INFO] [1532931507.618892480]: STAGE_NODE [pose_estimation]: Received frame.
2018-07-30 14:18:27.632 (   1.951s) [        2E1BF700]    pose_estimation.cpp:114      0| Processing frame #3 with timestamp 1532931506590496385!
2018-07-30 14:18:27.632 (   1.951s) [        2E1BF700]    pose_estimation.cpp:166      0| Tracking frame #3 in visual SLAM...!
[ INFO] [1532931507.774115317]: Image #9, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0009.jpg
[ INFO] [1532931507.820739724]: STAGE_NODE [pose_estimation]: Received frame.
RH: 0.436867 SH: 882.216
**2018-07-30 14:18:27.968 (   2.287s) [        2E1BF700]    pose_estimation.cpp:179   WARN| No tracking.**
2018-07-30 14:18:27.971 (   2.290s) [        2E1BF700] worker_thread_base.cpp:62       0| Thread 'Stage [pose_estimation]' has processed data. Time elapsed: 0.34 [s]
[ INFO] [1532931507.974523680]: Image #10, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0010.jpg
[ INFO] [1532931508.017728366]: STAGE_NODE [pose_estimation]: Received frame.
2018-07-30 14:18:28.054 (   2.373s) [        2E9C0700]    pose_estimation.cpp:505      0| Publishing pose of frame #3...
2018-07-30 14:18:28.071 (   2.390s) [        2E1BF700]    pose_estimation.cpp:114      0| Processing frame #6 with timestamp 1532931507192029979!
2018-07-30 14:18:28.072 (   2.390s) [        2E1BF700]    pose_estimation.cpp:166      0| Tracking frame #6 in visual SLAM...!
[ INFO] [1532931508.174775871]: Image #11, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0011.jpg
[ INFO] [1532931508.225445285]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1532931508.374993406]: Image #12, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0012.jpg
**2018-07-30 14:18:28.387 (   2.706s) [        2E1BF700]    pose_estimation.cpp:179   WARN| No tracking.**

**realm_stage_node: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
[ INFO] [1532931521.191556690]: Image #76, image Path: /media/keo/XA_vision/liupeng/OpenREALM/edm_big_overlap_50p/img_0076.jpg
[realm_pose_estimation-2] process has died [pid 9031, exit code -6, cmd /home/keo/catkin_ws/devel/lib/realm_ros/realm_stage_node __name:=realm_pose_estimation __log:=/home/keo/.ros/log/53f56d9c-93c0-11e8-9348-c83dd48417bd/realm_pose_estimation-2.log].
log file: /home/keo/.ros/log/53f56d9c-93c0-11e8-9348-c83dd48417bd/realm_pose_estimation-2*.log**

`

laxnpander, 2018-07-30:
Hmm, are you using the provided dataset or your own? The first error is weird. The message type should be “sensor_msgs/NavSatFix”. Do you have any idea where the occupancy grid message might come from? 

The second one is some weird eigen stuff. I will investigate what might help!

KeoChi, 2018-08-02:
Amazing！ I create and init another independent workspace. And then I implement the second option "Configuration WITH cuda" . It can work! I don't know why. May be some code packages in the original workspace affected the compilation of this project OpenREALM.

laxnpander, 2018-08-03:
Glad it works now for you. Let me know if you have any further questions or problems!

closed 2018-08-03 by laxnpander