laxnpander/OpenREALM #4: Is there any system requirement for this program?

MerlotQ, 2019-01-07:
The ros nodes started and then all of them died. I guess the program ran out of memory. So is there any minimum sys requirement?

laxnpander, 2019-01-07:
@MerlotQ: Hard to generalise, as it highly depends on what kind of mapping you want to use (with slam/without/3D). The example from the README ran with a 16 GB RAM i5 Laptop without a problem. What do you have?

MerlotQ, 2019-01-08:
@laxnpander Hi thanks for your reply. I am using an i5 destop with 8 GB ram. Should be ok. I found the problem is I lack the rviz plugin "rviz_plugins/AerialMapDisplay". Also I have problem with boost filesystem lib. The path and file check gives error.

MerlotQ, 2019-01-08:
The log info is like 

```
[ERROR] [1546935253.217805674]: PluginlibFactory: The plugin for class 'realm_rviz_plugin/GroundImage' failed to load.  Error: Failed to load library /media/hlf0055/4CB827A3B8278B12/stitching_datasets/realtime_map/devel/lib//librealm_rviz_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /media/hlf0055/4CB827A3B8278B12/stitching_datasets/realtime_map/devel/lib//librealm_rviz_plugin.so: undefined symbol: _ZN13QNetworkReply16staticMetaObjectE)
[ERROR] [1546935253.236573507]: PluginlibFactory: The plugin for class 'rviz_plugins/AerialMapDisplay' failed to load.  Error: MultiLibraryClassLoader: Could not create class of type rviz::AerialMapDisplay
[New Thread 0x7fffc3544700 (LWP 8843)]
[New Thread 0x7fffc2d43700 (LWP 8844)]
[New Thread 0x7fffc2542700 (LWP 8845)]
[New Thread 0x7fffc1d41700 (LWP 8850)]
profile path is :/media/hlf0055/4CB827A3B8278B12/stitching_datasets/realtime_map/src/OpenREALM/realm_ros/profiles/alexa_gnss
[ INFO] [1546935253.671858201]: Exiv2 Grabber Node [Ankommen]: Successfully loaded camera: 
	cx = 610.324
	cy = 516.917
	fx = 1209.07
	fy = 1210.66
	k1 = -0.198806
	k2 = 0.122544
	p1 = -0.000221
	p2 = -0.000877

Thread 1 "realm_exiv2_gra" received signal SIGSEGV, Segmentation fault.
0x00007ffff6420942 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::find_last_of(char const*, unsigned long, unsigned long) c--Type <RET> for more, q to quit, c to continue without paging--
onst () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) 
(gdb) 
(gdb) bt
#0  0x00007ffff6420942 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::find_last_of(char const*, unsigned long, unsigned long) const () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff529575e in boost::filesystem::path::m_parent_path_end() const ()
   from /usr/local/lib/libboost_filesystem.so.1.58.0
#2  0x00007ffff5295959 in boost::filesystem::path::remove_filename() ()
   from /usr/local/lib/libboost_filesystem.so.1.58.0
#3  0x00007ffff5291521 in boost::filesystem::detail::directory_iterator_increment(boost::filesystem::directory_iterator&, boost::system::error_code*) ()
   from /usr/local/lib/libboost_filesystem.so.1.58.0
#4  0x00007ffff7b8e79c in realm::Exiv2GrabberNode::getFileList(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /media/hlf0055/4CB827A3B8278B12/stitching_datasets/realtime_map/devel/.private/realm_ros/lib/librealm_ros.so
#5  0x00007ffff7b90481 in realm::Exiv2GrabberNode::Exiv2GrabberNode() ()
   from /media/hlf0055/4CB827A3B8278B12/stitching_datasets/realtime_map/devel/.private/realm_ros/lib/librealm_ros.so
#6  0x0000000000405643 in main ()
```

laxnpander, 2019-01-10:
@MerlotQ: Are you running Ubuntu 16.04? There seems to be something wrong with your ROS and/or Boost. Also the libstd error is a little weird. The Ground Image plugin is provided with the code. Don’t know why it should not be able to load. The AerialMapDisplay is just nice to have and not required.

MerlotQ, 2019-01-14:
@laxnpander  Thanks. It turned out there was something wrong with my ubuntu env. I now can run the program nicely on another pc. Is swri-robotics/mapviz the source github for AerialMapDisplay?

closed 2019-01-14 by MerlotQ

laxnpander, 2019-01-14:
@MerlotQ: No, it should be rviz_satellite. You can clone it here:
https://github.com/gareth-cross/rviz_satellite

laxnpander, 2019-01-24:
@MerlotQ: Hey, sorry for the late response. I am currently travelling and dont have access to the internet everywhere. You have to consider the following: For visual SLAM you need a lot more than just image data. For example a good quality camera, with high framerate and low distortion. It must be calibrated in advance. There is quite a few other requirements to setup a live mapping system, which can't be covered by OpenREALM (or any other live running system, state of the art). 

If you tell me more about your use case, I might be able to help you? :)

dannove, 2019-02-26:
> @MerlotQ: No, it should be rviz_satellite. You can clone it here:
> https://github.com/gareth-cross/rviz_satellite

Hey, I am a fresh man. I met the same question that lack the rviz plugin "rviz_plugins/AerialMapDisplay". I have cloned the 'rviz_satellite'. How can I install it in the project.

laxnpander, 2019-02-27:
Just put the cloned folder in your ROS working directory. This should be where all other top level open realm folders (realm_ros, real
_core, ...) are.

dannove, 2019-02-27:
> @dannove: Just put the cloned folder in your ROS working directory. This should be where all other top level open realm folders (realm_ros, real
> _core, ...) are.

     wow, thanks for your reply!  I fixed that question but when i run the test launch, it shows that 
 pose_estimation.cpp:179   WARN| No tracking.]  and ### GEOREFERENCE ABORTED ###
     It seems that can not pass through stage [pose_estimation] . I checked the ‘output/pose_estimation’ file. There are only 2 files in 'trajectory', and the other folder 'frames', 'keyframes', 'keyframes_full' just have nothing. I WONDE  may you can help me analyze what happened. Your reply will be higly appreciated!

laxnpander, 2019-03-02:
@dannove: Which launch file are you starting? The SLAM one? No tracking is not a problem in the beginning. It takes a bit of time until the SLAM and GPS are fused to a scale measurement. So if you let it run for 30 secs or so it should start tracking and compute a georeference. You can follow the progress observing the „pose_estimation/img_tracked“  ROS message. Let me know if you find it, I am not 100% sure about the name right now.

Output files can be defined in the profiles folder. You have to find „stage_settings.txt“ in the pose estimation profile folder and define what outputs you want to create. Keyframes are not always saved, only if activated. I think I deactivated it for the test launch.

dannove, 2019-03-05:
@laxnpander : Sorry for the late response. I just follow the 'Quick Start' in the mainpage and ran the realm_ros/launch/alexa_noreco.launch. I only modified the path in the launch file and I didn't change anything in the profile.

laxnpander, 2019-03-10:
@dannove: Hmmm, difficult to say without seeing it. Is it doing nothing? Do you see any other messages of the pose estimation being delivered? There should be an image stream for example, that shows the current tracked image.

rexsham, 2019-05-09:
> @dannove: Just put the cloned folder in your ROS working directory. This should be where all other top level open realm folders (realm_ros, real
> _core, ...) are.

Maybe I am dumb, I tried using rviz_satellite demo launch file and it worked loading mapbox tiles in that demo. But I still meet problem while using OpenREALM loading the AerialMapDisplay plugin.

I have cloned the rviz_satellite folder into the top level folder of OpenREALM where realm_ros, realm_core folder is located in. Then I catkin clean and catkin build again.

When I try to use the launch files of OpenREALM, I see the error that the other few people see:
(I have replaced the URI object to the one that worked with the rviz_satellite demo and it is still the same.

AerialMapDisplay
The class required for this display, 'rviz_plugins/AerialMapDisplay', could not be loaded.
Error:
According to the loaded plugin descriptions the class rviz_plugins/AerialMapDisplay with base class type rviz::Display does not exist. Declared types are realm_rviz_plugin/GroundImage rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu