laxnpander/OpenREALM #6: Can I Run alexa_reco.launch without CUDA?

rexsham, 2019-05-09:
I can run alexa_gnss.launch and alexa_noreco.launch without problem on a Virtual Box machine which do not have CUDA access.

When I run alexa_reco.launch, RViz showed No Image and here is the terminal log:

insight@openrealm:~/catkin_ws/src/OpenREALM/realm_ros/launch$ roslaunch realm_ros alexa_reco.launch
sh: 0: getcwd() failed: No such file or directory
... logging to /home/insight/.ros/log/ce99feee-7206-11e9-a3d9-080027e141bd/roslaunch-openrealm-4367.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://openrealm:33089/

SUMMARY
========

PARAMETERS
 * /realm_densification/config/id: alexa
 * /realm_densification/config/method: psl
 * /realm_densification/config/profile: alexa_reco
 * /realm_densification/stage/type: densification
 * /realm_densification/topics/input: /realm/alexa/pose...
 * /realm_densification/topics/output: /realm/alexa/dens...
 * /realm_exiv2_grabber/config/id: alexa
 * /realm_exiv2_grabber/config/input: /home/insight/Dow...
 * /realm_exiv2_grabber/config/profile: alexa
 * /realm_exiv2_grabber/config/rate: 10.0
 * /realm_mosaicing/config/id: alexa
 * /realm_mosaicing/config/profile: alexa_reco
 * /realm_mosaicing/stage/master: True
 * /realm_mosaicing/stage/type: mosaicing
 * /realm_mosaicing/topics/input: /realm/alexa/orth...
 * /realm_ortho_rectification/config/id: alexa
 * /realm_ortho_rectification/config/profile: alexa_reco
 * /realm_ortho_rectification/stage/type: ortho_rectification
 * /realm_ortho_rectification/topics/input: /realm/alexa/surf...
 * /realm_ortho_rectification/topics/output: /realm/alexa/orth...
 * /realm_pose_estimation/config/id: alexa
 * /realm_pose_estimation/config/method: orb_slam2
 * /realm_pose_estimation/config/profile: alexa_reco
 * /realm_pose_estimation/stage/type: pose_estimation
 * /realm_pose_estimation/topics/input: /realm/alexa/input
 * /realm_pose_estimation/topics/output: /realm/alexa/pose...
 * /realm_surface_generation/config/id: alexa
 * /realm_surface_generation/config/profile: alexa_reco
 * /realm_surface_generation/stage/type: surface_generation
 * /realm_surface_generation/topics/input: /realm/alexa/dens...
 * /realm_surface_generation/topics/output: /realm/alexa/surf...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    realm_densification (realm_ros/realm_stage_node)
    realm_exiv2_grabber (realm_ros/realm_exiv2_grabber)
    realm_mosaicing (realm_ros/realm_stage_node)
    realm_ortho_rectification (realm_ros/realm_stage_node)
    realm_pose_estimation (realm_ros/realm_stage_node)
    realm_surface_generation (realm_ros/realm_stage_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [4377]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ce99feee-7206-11e9-a3d9-080027e141bd
process[rosout-1]: started with pid [4390]
started core service [/rosout]
process[realm_exiv2_grabber-2]: started with pid [4407]
process[realm_pose_estimation-3]: started with pid [4408]
[... 211 lines cut ...]
2019-05-09 11:01:58.656 (   0.468s) [        E774A4C0] surface_generation.cpp:147      0| - save_elevation: 0
2019-05-09 11:01:58.656 (   0.468s) [        E774A4C0] surface_generation.cpp:148      0| - save_normals: 0
[ INFO] [1557370918.656132963]: STAGE_NODE [surface_generation]: Received output directory, set to:
	/home/insight/catkin_ws/src/OpenREALM/realm_ros/output/19-05-09_11-01-58
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]             loguru.hpp:2056     0| Logging to '/home/insight/catkin_ws/src/OpenREALM/realm_ros/output/19-05-09_11-01-58/ortho_rectification/stage.log', mode: 'a', verbosity: 9
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]         stage_base.cpp:49       0| Successfully initialized!
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]         stage_base.cpp:50       0| Stage path set to: /home/insight/catkin_ws/src/OpenREALM/realm_ros/output/19-05-09_11-01-58/ortho_rectification
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]ortho_rectification.cpp:169      0| ### Stage process settings ###
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]ortho_rectification.cpp:170      0| - GSD: 0.10
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]ortho_rectification.cpp:172      0| ### Stage save settings ###
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]ortho_rectification.cpp:173      0| - save_valid: 0
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]ortho_rectification.cpp:174      0| - save_ortho_rgb: 0
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]ortho_rectification.cpp:175      0| - save_ortho_gtiff: 0
2019-05-09 11:01:58.661 (   0.423s) [        1744B4C0]ortho_rectification.cpp:176      0| - save_elevation: 0
2019-05-09 11:01:58.662 (   0.423s) [        1744B4C0]ortho_rectification.cpp:177      0| - save_elevation_angle: 0
[ INFO] [1557370918.662080477]: STAGE_NODE [ortho_rectification]: Received output directory, set to:
	/home/insight/catkin_ws/src/OpenREALM/realm_ros/output/19-05-09_11-01-58
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
2019-05-09 11:01:58.727 (   0.619s) [         96D7700]             loguru.hpp:2056     0| Logging to '/home/insight/catkin_ws/src/OpenREALM/realm_ros/output/19-05-09_11-01-58/pose_estimation/publisher.log', mode: 'a', verbosity: 9
2019-05-09 11:01:58.727 (   0.619s) [         96D7700]    pose_estimation.cpp:452      0| Successfully initialized pose_estimation publisher!

laxnpander, 2019-05-29:
Jep, you are right! Thanks for pointing that out. I will fix it as soon as I have better connection (currently travelling abroad). Keep up the bug search!

laxnpander, 2019-07-10:
Should be fixed thanks ykhedar! 🙏🏻

closed 2019-07-10 by laxnpander