laxnpander/OpenREALM #7: alexa_reco shows no reconstruction image in RViz

rexsham, 2019-05-21:
I have been able to run alexa_gnss and alexa_noreco without problem.

But when I try to run alexa_reco, I do not see RViz showing anything on the reconstruction display area.

Image pane has images loaded without problem.

rostopic shows that VSLAM has been run and visual tracking can output coordinates of each camera location.  But there is no point cloud coming out.

I have been stuck on this for a while,  this problem happens on both cuda or no cuda compile settings.

laxnpander, 2019-05-29:
@rexsham: Hmm, can you see if the densification stage generates depth images? There is a topic for this as well. If depth images are generated, then reconstruction works and visualisation fails.

ykhedar, 2019-07-10:
@rexsham  I think this issue is related to the issue #6 . If it works for you now, we should close this issue.

rexsham, 2019-07-10:
I have tried to fix issue #6 by myself but the problem persists. I'll check
out your latest code to try again and respond here.

laxnpander, 2020-01-17:
Reopen if the problem still persists.

closed 2020-01-17 by laxnpander