laxnpander/OpenREALM #10: Build Failed

immuhammadadil, 2019-08-20:
Hi, I am getting this error on UBuntu 16+ROS Kinetic & Opencv3.3.1 I have fulfilled all the requirements and cross checked them

burqstream@burqstream-Z97-HD3:~/catkin_ws$ catkin build
---------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/burqstream/catkin_ws
---------------------------------------------------------------
Build Space:        [exists] /home/burqstream/catkin_ws/build
Devel Space:        [exists] /home/burqstream/catkin_ws/devel
Install Space:      [unused] /home/burqstream/catkin_ws/install
Log Space:          [exists] /home/burqstream/catkin_ws/logs
Source Space:       [exists] /home/burqstream/catkin_ws/src
DESTDIR:            [unused] None
---------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        psl
---------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------
[build] Found '11' packages in 0.0 seconds.                                        
[build] Updating package table.                                                    
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                                
___________________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /home/burqstream/catkin_ws/logs/catkin_tools_prebuild/build.cmake.001.log
ImportError: "from catkin_pkg.package import parse_package" failed: No module named 'catkin_pkg'
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/burqstream/.virtualenvs/cv/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/burqstream/catkin_ws/build/catkin_tools_prebuild/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:167 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

cd /home/burqstream/catkin_ws/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /home/burqstream/catkin_ws/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/burqstream/catkin_ws/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/burqstream/catkin_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
...................................................................................
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]          
Failed    <<< catkin_tools_prebuild                [ 0.2 seconds ]                 
Abandoned <<< orb_slam_2                           [ Unrelated job failed ]        
Abandoned <<< realm_core                           [ Unrelated job failed ]        
Abandoned <<< realm_io                             [ Unrelated job failed ]        
Abandoned <<< realm_densifier_base                 [ Unrelated job failed ]        
Abandoned <<< realm_msgs                           [ Unrelated job failed ]        
Abandoned <<< realm_ortho                          [ Unrelated job failed ]        
Abandoned <<< realm_rviz_plugin                    [ Unrelated job failed ]        
Abandoned <<< realm_vslam_base                     [ Unrelated job failed ]        
Abandoned <<< realm_stages                         [ Unrelated job failed ]        
Abandoned <<< realm_ros                            [ Unrelated job failed ]        
[build] Summary: 0 of 11 packages succeeded.                                       
[build]   Ignored:   1 packages were skipped or are blacklisted.                   
[build]   Warnings:  None.                                                         
[build]   Abandoned: 10 packages were abandoned.                                   
[build]   Failed:    1 packages failed.                                            
[build] Runtime: 0.3 seconds total.

laxnpander, 2019-08-25:
Hmm, looks like something is wrong with your catkin build tools. Did you try installing it with
`pip install catkin_pkg`
?

immuhammadadil, 2019-08-25:
Thanks for the reply. Yes, I have tried pip install catkin_pkg.
I am working to sort out my environment issues.
May I ask one more question. Did you implement Map2DFusion? It is not giving good results on custom dataset although it is working good with provided sample datasets.
I am implementing and benchmarking all the realtime mosaicking techniques. By the way, yours is most impressive one as I read from your thesis(yet to be tested)

laxnpander, 2019-08-26:
@immuhammadadil: I know Map2DFusion, but I never got into it deeply. My guess would be, that tuning the ORB SLAM is the main issue, if it does not perform properly. So make sure your custom dataset is suited for Orb slam and all the correct parameters are used by map2dfusion framework. It’s the same for OpenREALM by the way. Getting the pose estimator to work accurately is the most crucial part.

immuhammadadil, 2019-08-26:
Thanks for your expert opinion.
Can you please guide me on what method or tool I should use to calibrate the camera? 
I have captured dataset using mavic2 Pro. ORBSLAM doesn't processed all my frames for trajectory generation(Timestamp X Y Z Q1 Q2 Q3 Q4). That's why I have used MATLAB to generate a trajectory file.
Your opinion can help a lot.
Regards,
M Adil

laxnpander, 2019-08-26:
@immuhammadadil: We played around with DJI data, too. But based on our tests it is not suitable for visual SLAM. I can’t really say why exactly, maybe due to low frame rate or the overall specifications of the camera, but ORB SLAM was not able to detect enough stable features. Our tests were not comprehensive though, but you will definitely have a hard time getting it to work! Maybe if you fly high enough and have a very unique scenery throughout every picture. But jeah, it will be hard work!

At best you have a very good camera setup with high frame rate and low distortion. This is essential for all visual SLAM approaches and real-time mapping btw.

As for calibration: I used the ROS node for camera calibration of my GigE camera. This again will not work with your Mavic. You may have to find a tool that loads already captured images and uses them for calibration. The OpenCV example for calibration will help you do that!

immuhammadadil, 2019-08-26:
Thanks for giving such a detailed response. You really directed me to a new course of action.

laxnpander, 2019-08-30:
@immuhammadadil: No problem! Let me know if you have further questions or problems! I am currently on the road, so sorry I cant help with your build environment. Hope it is easy fixable!

closed 2019-09-02 by immuhammadadil

immuhammadadil, 2019-09-02:
Hi, 
I have successfully set up and ran openREALM. Results are also according to the expectation.
Can you please refer me to some more datasets like the one you have provided.
Thanks

laxnpander, 2019-09-05:
@immuhammadadil: Unfortunately I don’t know of any public datasets, which is why we captured this one.