laxnpander/OpenREALM #14: Misaligned Mosaic

immuhammadadil, 2019-09-06:
The output mosaic of the sample dataset is misaligned. The alignment issue exists between legs of flight i.e. vertical legs.
Can you tell what is the issue with alignment? Is this the best output of openREALM?

Secondly, if we compare image resolution with the mosaic resolution, the mosaic is a sample at 1/6 of the image size. Can we change this ratio to something else like 1/2 or 1/4 in code?
[image]

laxnpander, 2019-09-11:
@immuhammadadil: This looks like the GPS only Version. I am not sure right now if it can be better, did you have a look into my thesis? There I analysed all different versions (gps/vslam/dense).

For way better results try to run it with visual SLAM. Also you can change resolution of the final mosaic for example in:

realm_ros/profiles/alexa_noreco/ortho_rectification/stage_settings.yaml

There you will find the GSD (Ground Sampling Distance) parameter which indicates the resolution in m/pixel. But make sure you roslaunch with the correct profile! OpenREALM profiles are passed in with the roslaunch file and then look in realm_ros/profiles folder. This way you can make sure to test the same dataset with different parameter sets.

laxnpander, 2019-09-11:
Ah, a sidenote: I think you use a viewer that is not suited for the tiff export. The colours are wrong. You might want to check out QGIS. There you can load the tiff and overlay it with google maps satellite data. Very helpful!

laxnpander, 2019-09-11:
I just had a look. GPS only mosaic is not getting any better! I think the heading is a bit off in the data, but there is not much you can do about it. But again: this is gps only! When you use visual SLAM for pose estimation it will be much better.

immuhammadadil, 2019-09-14:
Hi, thanks for the reply, I have tried alexa_noreco.launch, but the result is not completed. It's processing just 1 leg of the flight. Only one row of the mosaic is covered. Is there any parameter to be tuned?

laxnpander, 2019-09-20:
@immuhammadadil: can you check how the ORB SLAM is performing? There is an output picture showing the features successfully tracked. Something like realm/pose_estimation/img_tracked.

My guess is your computer is a bit weaker than mine. You can reduce the resolution of the orb slam in realm_ros/profiles/alexa_noreco/pose_estimation/method/orb_slam_settings.yaml.

Resizing 1.0 means full size image.
Resizing 0.5 means 1/4 image size.

You can also vary the number of features tracked, but I think 3000 might be fine in any case.

I am working on more documentation for this stuff, sorry you have to find your way through on your own at the moment. Hope it helps!

closed 2019-11-27 by immuhammadadil