laxnpander/OpenREALM #15: realm_exiv2_grabber-2 process dies

pjende, 2020-01-15:
Hi all,
first of all, thank you for putting this software together. It looks really nice but I was unable to run it as of now.
I am using Ubuntu 16.04 in a VM and could compile the program successfully.

However, when I am trying to run the demo with one of your launch-files, I run into this error. I already tried to use other/older versions of exiv2 without any success.

[realm_exiv2_grabber-2] process has died [pid 16955, exit code -11, cmd /home/osboxes/openrealm/devel/lib/realm_ros/realm_exiv2_grabber __name:=realm_exiv2_grabber __log:=/home/osboxes/.ros/log/d2cbd3f0-37b0-11ea-8e07-0800272121fb/realm_exiv2_grabber-2.log]. log file: /home/osboxes/.ros/log/d2cbd3f0-37b0-11ea-8e07-0800272121fb/realm_exiv2_grabber-2*.log

Thank you for your help!

All the best

laxnpander, 2020-01-17:
@pjende: Hmm, do you have some more information? Is it crashing right from the start? Or does it do something and then crashes? Make sure the path you are reading from is your image folder and that there is not other data which might be read by accident.

pjende, 2020-01-20:
Thanks for your quick reply. It crashes right from the start, please see the terminal output below. I am using the example files you provided. The boost version I am using is 1.72. My ros installation seems to work just fine, I installed it from scratch, i.e. full installation, using a VM for your software. 

`$ roslaunch realm_ros alexa_gnss.launch
... logging to /home/osboxes/.ros/log/81c302bc-3b61-11ea-903e-0800272121fb/roslaunch-osboxes-16586.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://osboxes:37589/

SUMMARY
========

PARAMETERS
 * /realm_exiv2_grabber/config/id: alexa
 * /realm_exiv2_grabber/config/input: /home/osboxes/ope...
 * /realm_exiv2_grabber/config/profile: alexa_gnss
 * /realm_exiv2_grabber/config/rate: 10.0
 * /realm_mosaicing/config/id: alexa
 * /realm_mosaicing/config/profile: alexa_gnss
 * /realm_mosaicing/stage/master: True
 * /realm_mosaicing/stage/type: mosaicing
 * /realm_mosaicing/topics/input: /realm/alexa/orth...
 * /realm_ortho_rectification/config/id: alexa
 * /realm_ortho_rectification/config/profile: alexa_gnss
 * /realm_ortho_rectification/stage/type: ortho_rectification
 * /realm_ortho_rectification/topics/input: /realm/alexa/surf...
 * /realm_ortho_rectification/topics/output: /realm/alexa/orth...
 * /realm_pose_estimation/config/id: alexa
 * /realm_pose_estimation/config/method: orb_slam2
 * /realm_pose_estimation/config/profile: alexa_gnss
 * /realm_pose_estimation/stage/type: pose_estimation
 * /realm_pose_estimation/topics/input: /realm/alexa/input
 * /realm_pose_estimation/topics/output: /realm/alexa/pose...
 * /realm_surface_generation/config/id: alexa
 * /realm_surface_generation/config/profile: alexa_gnss
 * /realm_surface_generation/stage/type: surface_generation
 * /realm_surface_generation/topics/input: /realm/alexa/pose...
 * /realm_surface_generation/topics/output: /realm/alexa/surf...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    realm_exiv2_grabber (realm_ros/realm_exiv2_grabber)
    realm_mosaicing (realm_ros/realm_stage_node)
    realm_ortho_rectification (realm_ros/realm_stage_node)
    realm_pose_estimation (realm_ros/realm_stage_node)
    realm_surface_generation (realm_ros/realm_stage_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [16596]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 81c302bc-3b61-11ea-903e-0800272121fb
process[rosout-1]: started with pid [16609]
started core service [/rosout]
process[realm_exiv2_grabber-2]: started with pid [16612]
process[realm_pose_estimation-3]: started with pid [16627]
process[realm_surface_generation-4]: started with pid [16628]
process[realm_ortho_rectification-5]: started with pid [16629]
process[realm_mosaicing-6]: started with pid [16630]
process[rviz-7]: started with pid [16631]
[ INFO] [1579510058.099975668]: STAGE_NODE [mosaicing]: Loading stage settings from path:
	/home/osboxes/openrealm/src/OpenREALM/realm_ros/profiles/alexa_gnss/mosaicing/stage_settings.yaml
[ INFO] [1579510058.102887605]: STAGE_NODE [surface_generation]: Loading stage settings from path:
	/home/osboxes/openrealm/src/OpenREALM/realm_ros/profiles/alexa_gnss/surface_generation/stage_settings.yaml
[ INFO] [1579510058.113054324]: STAGE_NODE [ortho_rectification]: Loading stage settings from path:
	/home/osboxes/openrealm/src/OpenREALM/realm_ros/profiles/alexa_gnss/ortho_rectification/stage_settings.yaml
[... 117 lines cut ...]
2020-01-20 03:47:38.327 (   0.321s) [        641772C0] surface_generation.cpp:141      0| - try_use_elevation: 1
2020-01-20 03:47:38.327 (   0.321s) [        641772C0] surface_generation.cpp:142      0| - knn_radius_factor: 1.00
2020-01-20 03:47:38.327 (   0.321s) [        641772C0] surface_generation.cpp:143      0| - mode_surface_normals: 0
2020-01-20 03:47:38.327 (   0.321s) [        641772C0] surface_generation.cpp:145      0| ### Stage save settings ###
2020-01-20 03:47:38.327 (   0.321s) [        641772C0] surface_generation.cpp:146      0| - save_valid: 0
2020-01-20 03:47:38.327 (   0.321s) [        641772C0] surface_generation.cpp:147      0| - save_elevation: 0
2020-01-20 03:47:38.327 (   0.321s) [        641772C0] surface_generation.cpp:148      0| - save_normals: 0
[ INFO] [1579510058.327172844]: STAGE_NODE [surface_generation]: Received output directory, set to:
	/home/osboxes/openrealm/src/OpenREALM/realm_ros/output/20-01-20_03-47-38
2020-01-20 03:47:38.329 (   0.322s) [        5866F2C0]             loguru.hpp:2056     0| Logging to '/home/osboxes/openrealm/src/OpenREALM/realm_ros/output/20-01-20_03-47-38/ortho_rectification/stage.log', mode: 'a', verbosity: 9
2020-01-20 03:47:38.329 (   0.322s) [        5866F2C0]         stage_base.cpp:49       0| Successfully initialized!
2020-01-20 03:47:38.329 (   0.322s) [        5866F2C0]         stage_base.cpp:50       0| Stage path set to: /home/osboxes/openrealm/src/OpenREALM/realm_ros/output/20-01-20_03-47-38/ortho_rectification
2020-01-20 03:47:38.329 (   0.322s) [        5866F2C0]ortho_rectification.cpp:169      0| ### Stage process settings ###
2020-01-20 03:47:38.329 (   0.322s) [        5866F2C0]ortho_rectification.cpp:170      0| - GSD: 0.10
2020-01-20 03:47:38.329 (   0.322s) [        5866F2C0]ortho_rectification.cpp:172      0| ### Stage save settings ###
[... cut ...]
[ INFO] [1579510058.329824412]: STAGE_NODE [ortho_rectification]: Received output directory, set to:
	/home/osboxes/openrealm/src/OpenREALM/realm_ros/output/20-01-20_03-47-38
[realm_exiv2_grabber-2] process has died [pid 16612, exit code -11, cmd /home/osboxes/openrealm/devel/lib/realm_ros/realm_exiv2_grabber __name:=realm_exiv2_grabber __log:=/home/osboxes/.ros/log/81c302bc-3b61-11ea-903e-0800272121fb/realm_exiv2_grabber-2.log].
log file: /home/osboxes/.ros/log/81c302bc-3b61-11ea-903e-0800272121fb/realm_exiv2_grabber-2*.log
`

pjende, 2020-01-21:
Thanks again for your help! I really appreciate that.

1) I tried to provide more RAM (16GB), it did not change a thing, the crash also occurs. 
2) I isolated all the other nodes, the crash still occurs. Which version of exiv2 are you using? 

On another note, I have seen that you are planning to further develop the procedure to make it compatible with newer versions of ROS and Ubuntu. Is this still a thing?

laxnpander, 2020-01-21:
Hey @pjende,

I am currently using exiv2 version 0.27.99.0. And jeah sure, what kind of system do you run? My personal laptop where I use OpenREALM at the moment is Ubuntu 18.04 with ROS Melodic. I can provide it to you if it helps?

pjende, 2020-01-21:
That would be great if you could do that! I also run 18.04 and Melodic on my main machine. 
As you recommended in your installation instructions, I installed OpenCV 3.3.1 on the VM.

laxnpander, 2020-01-21:
@pjende: Here is the download link for my current version: https://drive.google.com/open?id=15Dit_H-2IzSt9e5IG64mtarHuNWIj59t

It's very close to the current git revision, only minor changes. Some things that I encountered while migrating to 18.04 and melodic:
1) PCL can be downloaded more comfortably through the repository: 
`sudo apt install libpcl-dev`
2) Geographic messages were missing:
`sudo apt install ros-melodic-geographic-msgs`
3) OpenCV from repository is 3.2, make sure you get 3.3 from the official OpenCV website
4) Geodesy package was missing
`sudo apt install ros-melodic-geodesy`
5) Get Exiv2 from the official sources https://www.exiv2.org/

That's it. Let me know if you have problems with compilation, I will try my best to guide you through and get you going!

pjende, 2020-01-23:
Hi, sorry for coming back to you so late!
First of all, thank you very much for providing your current version. 

I could successfully install and run it, however, there were some little issues I had to overcome. 
Apparently, exiv2 changed some details regarding UniquePtr and AutoPtr. So, I had to get the latest version from github directly, which can be found here: https://github.com/Exiv2/exiv2

Another little thing was a typo in the CMakeList of realm_io, where Exiv2 could not be found because the first letter was capitalised. According to the standard installation of Exiv2, the library can be accessed by "exiv2" and not "Exiv2". Maybe you have some sort of custom installation or they changed this detail over the course of their development. 

In any event, I am happy that it is running and would like to thank you again for your help. I do have further questions but I believe it may be better to open another issue for that if you do not mind :)

laxnpander, 2020-01-23:
@pjende: Thanks for the feedback! I will fix it for the next commit. It's sometimes really mysterious how things run on one computer setup, but not on another one with apparently the same parameters. :) Let me know if something doesn't work or if you need an explanation for something, it is a pleasure for me to help!

closed 2020-01-23 by laxnpander

LiShuaixin, 2022-06-09:
> Ah, and what kind of OpenCV version do you run? Often multiple versions of OpenCV can cause tremendous headache.

Hi, thanks for your sharing! I met the same issue that `realm_exiv2_grabber` node crashed from the start. I checked the code and found it crashed calling `cv::FileStorage fs(filepath, cv::FileStorage::READ);` My PC environment is Ubuntu20.04 with ROS noetic and Opencv 3.4.16. When I complied the OpenRealm_ROS wrapper, it threw out warnings below:
```
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /home/lee/RosWorkSpace/openrealm_ws/devel/lib/librealm_ros.so, may conflict with libopencv_core.so.3.4
/usr/bin/ld: warning: libopencv_calib3d.so.3.4, needed by /home/lee/RosWorkSpace/openrealm_ws/devel/lib/librealm_ros.so, may conflict with libopencv_calib3d.so.4.2
/usr/bin/ld: warning: libopencv_imgproc.so.4.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2.0, may conflict with libopencv_imgproc.so.3.4
```

It seems like the opencv lib from ROS noetic conflicts with the one I compiled from the source.

Any idea to solve the issue? many thanks!

laxnpander, 2022-06-09:
@LiShuaixin yes, this is due to conflicting libraries linked. You have to resolve the conflict either by removing your custom OpenCV version or by compiling your ROS (cv_bridge) with OpenCV 4.2.