laxnpander/OpenREALM #19: Not Tracking

scially, 2020-03-23:
hi, i use OpenREALM for my datas, but it always display : 'Not Tracking'.
[image]

laxnpander, 2020-03-25:
@scially: Hey, I use orb slam 2 for tracking. Everything that works with it, works also with OpenREALM. No tracking can have a lot of reasons unfortunately. You are processing your own dataset, correct? What FPS does the camera provide, what image resolution, whats your hardware setup?

Most of these problems come from a setup that is not suited for visual SLAM.

scially, 2020-04-07:
thanks.... my datasets has some error, and it solved ....
and i have another question...
the rviz has an error:
 ```
The class required for this display, 'rviz_plugins/AerialMapDisplay', could not be loaded.

Error:
According to the loaded plugin descriptions the class rviz_plugins/AerialMapDisplay  with base class type rviz::Display does not exist. 
Declared types are realm_rviz_plugin/GroundImage rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu
```
how do i solve it?

laxnpander, 2020-04-08:
Hey @scially,

this problem is not directly OpenREALM related. You can either
1) clone RVIZ Satellite in your catkin workspace: https://github.com/gareth-cross/rviz_satellite und use it to see how good your datasets align with google maps / open street maps
2) remove AerialMapDisplay in RVIZ as soon as it opens up and hit the save button, so it does not pop up anymore.

Let me know if this solved your issues,

greets,
Alex

scially, 2020-04-10:
i remove AerialMapDisplay in RVIZ ,and it's workd. thanks....

laxnpander, 2020-04-10:
@scially: You are welcome!

closed 2020-04-10 by laxnpander