laxnpander/OpenREALM #20: Using ROS sensor_msgs instead of Exif header information

pjende, 2020-04-20:
Hello all together,

thanks again for this really nice piece of software!

I was wondering whether there is a possibility to use direct sensor pose information within the process, i.e. absolute GNSS poses as a prerequisite for VSLAM. 

Probably, this is not built in yet, but I am willing to do my best and contribute this functionality. 

Any idea where to start?

PS: The issue from (https://github.com/laxnpander/OpenREALM/issues/16) is resolved! Thanks a lot!

laxnpander, 2020-04-20:
@pjende: Hey, nice to hear you are still on it. If I understand you correctly, you want to develop a custom grabber for image + gps information, right? So to get rid of exif tags?

That should be straightforward. I designed the pipeline so one can easily provide a custom grabber, as long as it creates a proper frame message. Have a look at the realm_ros/src/realm_ros_lib/grabber_exiv2_node.cpp. You can use it as prototype for your own node.

pjende, 2020-04-22:
Thank you for your fast response! 

> custom grabber for image + gps information, right? So to get rid of exif tags?
Yes, indeed! The idea is to run a rosbag with two messages (images and sensor information). 

Would you recommend using image_transport? As far as I can tell, you use a different method to publish the images.

laxnpander, 2020-04-22:
@pjende: And with sensor information you mean GPS + Magnetic Heading right? You can use image_transport, I don't see any downside on that. In my framework I bundle all data in a combined "frame" message, which is exchanged between the processing stages. But it also uses image_transport for compression and reading of the image data. The Exiv2 grabber was used, because it is easier to create small subsets of your dataset, if you have the images directly in a folder as .jpg or .png and can look at them. Rosbags are less handy in that case.

Just make sure you synchronize all incoming data with the timestamp, so you GPS location roughly matches the acquisition time of the image.

ubwolkers, 2020-04-24:
Hi all!

I think this is an interesting idea. I would like to do something similar and was wondering if you could give an example how to use the "frame" message properly! 
Thank you

laxnpander, 2020-04-25:
@ubwolkers: Hey, nice to hear you are interested in this. You can take a look at the Frame.msg in the realm_msgs package. The following parameters should be set, when the frame enters the pipeline:

**std_msgs/Header header**
This is the standard ROS message header. Set the timestamp to now when you create the message and you should be fine.

**std_msgs/String camera_id**
This is a unique name for the camera you can choose. For example "my_camera" or something like this.

**std_msgs/UInt32 stage_id**
This should be a unique ID for the frame. So basically you increment an integer every time you create a frame.

std_msgs/UInt64 timestamp
This is the timestamp of the acquisition. If you read directly from a camera, then it can be ros::Time::now(). But sometimes your image data was acquired somewhere else at a different time, that's why this exists. Just keep in mind, that the visual SLAM will predict the position based on the time passed between current and last frame.

sensor_msgs/CompressedImage imagedata
This should be straightforward the image data compressed.

realm_msgs/Pinhole camera_model
This is a custom message that I created for the camera pinhole model. I use the OpenCV standard, so fx, fy, cx, cy, width, height and the distortion parameters k1,k2,p1,p2 and k3 = 0. Note that fishhole cameras are currently not supported.

sensor_msgs/NavSatFix gpsdata
This is also a standard ROS message, so basically Latitude, Longitude, Altitude. Covariance is not important for my framework at the moment.

std_msgs/Float32 heading
This should be an optional parameter, but I would advise you to grab the info from your UAV if you can. It allows you to do GPS only based mapping and might stabilize the initialization procedure.

That's it! The rest of the parameters are computed within my framework and should not be your concern.
Let me know in case you need further help

laxnpander, 2020-05-23:
@ubwolkers @sev-of-nine: I added a ROS image grabber and a launch file to the repository. It is not yet final, but it should do the job. Let me know if you have trouble with it!

Note: One bug I experienced was, that when running data from a rosbag the RVIZ visualization does not seem to show anything. I have to resubscribe to the topics it should display. No idea why yet.

closed 2020-05-23 by laxnpander