laxnpander/OpenREALM #21: Issue with Exif header information

pjende, 2020-04-24:
Hello,
Although my raw JPG images can be tracked by OpenREALM, the georeferencing does not seem to work. I suspect this has to do with the Geotags in the exif header. This is an output snippet from the console, note the error message: **_Warning: Directory GPSInfo has an unexpected next pointer; ignored_**
```

> 2020-04-24 17:31:32.036 (  76.243s) [        19092700]    pose_estimation.cpp:117      0| Processing frame #5 with timestamp 1587742291567852154!
> 
> 2020-04-24 17:31:32.036 (  76.243s) [        19092700]    pose_estimation.cpp:169      0| Tracking frame #5 in visual SLAM...!
> [ INFO] [1587742292.098363955]: STAGE_NODE [pose_estimation]: Received frame.
> [ INFO] [1587742292.513321399]: Image #153, image Path: /frames_exif/2019-11-13T21:35:59.862979Z.jpg
> Warning: Directory GPSInfo has an unexpected next pointer; ignored.
> [ INFO] [1587742292.595857501]: STAGE_NODE [pose_estimation]: Received frame.
> 2020-04-24 17:31:32.614 (  76.821s) [        19092700]    pose_estimation.cpp:192      0| Key frame insertion.
> 2020-04-24 17:31:32.620 (  76.827s) [        19092700] worker_thread_base.cpp:64       0| Thread 'Stage [pose_estimation]' has processed data. Time elapsed: 0.58 [s]
> 2020-04-24 17:31:32.620 (  76.827s) [        13FFF700]geometric_referencer.cp:69       0| Starting georeferencing...
> 2020-04-24 17:31:32.621 (  76.828s) [        13FFF700]geometric_referencer.cp:135      0| ### GEOREFERENCE ABORTED ###
```

Comparing the exif information of my data set with the one provided, yields the following results, yet again with the same error message. Even though the GPSVersionID is not the same, I changed it to the one given in the demo data set. This won't do the trick. 

I created the exif headers using piexif (https://github.com/hMatoba/Piexif). After some googling, I have seen that someone already stumbled upon a similar issue regarding exif headers read by exiv2 (https://dev.exiv2.org/boards/3/topics/2159).   

Any idea how to resolve this issue?

Thank you very much!

DEMO:

```
`exiv2 -pv -g GPS img_0111.jpg`

> 0x8825 Image        GPSTag                      Long        1  122
> 0x0000 GPSInfo      GPSVersionID                Byte        4  2 2 2 2
> 0x0001 GPSInfo      GPSLatitudeRef              Ascii       2  N
> 0x0002 GPSInfo      GPSLatitude                 Rational    3  52/1 24/1 493/100
> 0x0003 GPSInfo      GPSLongitudeRef             Ascii       2  E
> 0x0004 GPSInfo      GPSLongitude                Rational    3  10/1 13/1 3826/100
> 0x0005 GPSInfo      GPSAltitudeRef              Byte        1  0
> 0x0006 GPSInfo      GPSAltitude                 Rational    1  98/1
```

My data set:

```
`exiv2 -pv -g GPS 2019-11-13T21:35:33.067425Z.jpg`

> Warning: Directory GPSInfo has an unexpected next pointer; ignored.
> 0x8825 Image        GPSTag                      Long        1  26
> 0x0000 GPSInfo      GPSVersionID                Byte        4  2 0 0 0
> 0x0001 GPSInfo      GPSLatitudeRef              Ascii       2  N
> 0x0002 GPSInfo      GPSLatitude                 Rational    3  47/1 58/1 243/20
> 0x0003 GPSInfo      GPSLongitudeRef             Ascii       2  E
> 0x0004 GPSInfo      GPSLongitude                Rational    3  8/1 29/1 2601/50
> 0x0005 GPSInfo      GPSAltitudeRef              Byte        1  0
> 0x0006 GPSInfo      GPSAltitude                 Rational    1  333/1
```

laxnpander, 2020-04-25:
Hmmm, I have never encountered such a problem. Can you send me a few of your images so I can try it out myself? 

A few thoughts apart from that:

- 5 Frames will probably be not enough to initialize the georeference. This might also be the reason, why it aborts? 

- There are some settings, I have to look up. But if I remember right, you have to fly at least ~50m or so to create enough spatial distance for the georeferencer to initialize. 

- Tracking of the orb slam must be stable enough to initialize the georeference.

pjende, 2020-04-26:
Thanks for your reply! Sure, I will attach some images including exif 
information as well as the corresponding camera IOs (calib.yaml). 

Regarding your remarks:
The tracking seems to work nicely (also for more than 50 metres), I used these images with OpenVSLAM as well and the ORB-SLAM preview in OpenREALM's RVIZ also indicates that it works. 

I strongly believe that it is the exif information. Maybe you have an idea to "repair" the header? 

In any event, I appreciate your help! Thanks!

Images + metadata:
[https://we.tl/t-sEz0y7FRGM](url)

laxnpander, 2020-04-26:
@pjende: Hey, I checked your dataset. I think you have saved GPS information in the metadata, but all of them seem to be the same? Are you sure, the images are geotagged properly? Might be induced by the corrupted header of course, but just to make sure. I get the same warning like you at least.

pjende, 2020-04-26:
Oh, you are right! Thanks! There was an issue in my converter code which I fixed now. The lat,lng,alt information are not interpolated or filtered, though, hence a few consecutive images share the same position. Since I have IMU information as well, I could integrate the relative updates in the state estimation. Will have a look at that tomorrow.

Certainly, this converter bug renders georeferencing impossible, OpenREALM, however, still does not work after fixing it. Could you have another look at the data? Thank you very much!

[https://we.tl/t-cEiHa4VBoI](url)

laxnpander, 2020-04-27:
@pjende: I checked your dataset and found the problem. The exiv2 grabber node expects a naming convention like: image_0001, image_0002, ... and so on. It creates the unique frame id based on the counter within the name. This caused some trouble inside the software. I didn't expect it to be a problem, so good thing you found it!

Fastest way for you to bypass this is rename your files. I will think about a fix, though I like the convention because you will be able to trace your image throughout the output data (depthmap_0001 will be based on image_0001 and so on). 

Also I played around with your dataset a little bit, the resolution is very high! In case you run into performance problems, here are some tweeks:
realm_ros/profiles/alexa_reco/densification/method/psl_settings.yaml 
-> resizing to ~0.15
realm_ros/profiles/alexa_reco/ortho_rectification/stage_settings.yaml
-> GSD to ~0.5

I will try to find some settings that work well on my computer and send them to you.

pjende, 2020-04-27:
Thank you very much for your effort! I changed the naming convention and it works now! 
I also edited the settings accordingly for my data set, although I am happy if you could share some best practice settings. 
The problem with the geotags does not seem to be an issue btw, so we could ignore that at the moment and close the topic if you want.

laxnpander, 2020-04-27:
You are welcome! Working with different datasets also helps me making the framework more robust, so it's a win-win :) You can open a new issue describing your setup / task, then I can help finding good settings.

closed 2020-04-27 by laxnpander