laxnpander/OpenREALM #23: could you please help me check if there is any problem with my data set?

scially, 2020-06-24:
Hello, could you please help me check if there is any problem with my data set? Except the camera parameters, all the other parameters are default, but the orthophoto results are not good, thanks~

Datas Share URL:  
[https://drive.google.com/file/d/18cfaPikdErhAKo0jISpyOH6OYdfrY941/view?usp=sharing](https://drive.google.com/file/d/18cfaPikdErhAKo0jISpyOH6OYdfrY941/view?usp=sharing)
---
the camera config is
``` YAML
type: pinhole
# Frames per second
fps: 5
# Image size
width: 1368
height: 912
# Principal point
cx: 689.97
cy: 456.58
# Focal length
fx: 2010.848
fy: 1340.565
# Distortion
k1: -0.266511765417987
k2: 0.10617945434786
p1: 0.000195369051756418
p2: 0.00143753730138743
k3: -0.0285408159411354
```
### my ortho result:
[image]

laxnpander, 2020-06-25:
@scially Hey hey, your dataset looks quite okay. Some things might need improvement:
- Try to fix the heading, at the turning points you have way too much movement. Orb slam is losing its tracking there and has a hard time to relocalize after that, resulting in some bad points and bad estimation quality in general. This should be an easy fix for the next dataset.
- Your frame rate is okay, but not overwhelming. If you want to get better data, you might consider flying slower or increase framerate. The tracking of the orb slam really needs **as little movement per frame as possible** 
- Your lens has a bit of fisheye, it seems to be okay but ideally it would be less. Usually you are not as flexible on the hardware, so I guess you can't improve that.

Unfortunately there is no magic algorithm to fix these issues with your dataset. My framework is at best as good as the orb slam, and it has a hard time with your data. Fixing the heading of the drone might already do the job.

Let me know if it helped and best regards,
Alex

scially, 2020-06-29:
Thank you~ 
I followed your suggestion, replaced a new UAV, and re-photographed an area, DJI FC6310S camera, the first route mosaicing perfect, but when it turned to the second route, although it was still Tracking, the mosaicing was not good, is my lateral overlap too high?

New Datas Share URL:
[https://drive.google.com/file/d/1xahiNa0fA_pKFx8Dr47Oltq20gaBAh_2/view?usp=sharing](https://drive.google.com/file/d/1xahiNa0fA_pKFx8Dr47Oltq20gaBAh_2/view?usp=sharing)  

the camera config is
```YAML
type: pinhole
# Frames per second
fps: 10.0
# Image size
width: 1368
height: 912
# Principal point
cx: 684.0
cy: 456.0
# Focal length
fx: 1989.818
fy: 1326.545
# Distortion
k1: 0.000001
k2: 0
p1: 0.000001
p2: 0.000001
k3: 0
```
[image]
[image]

laxnpander, 2020-06-29:
@scially: Your lateral overlap is indeed pretty high, but that is not a problem for OpenREALM. My guess is that your GPS data in the images is not accurate. Either that or the altitude is slightly wrong. It initializes the georeference in the first leg and then tries to correct it afterwards because it detects a mismatch between GPS/Altitude and visual position. That in consequence distorts the global map, because the georeference is changed.

Go to `realm_ros/profiles/your_profile/pose_estimation/stage_settings.yaml` and set the flag `update_georef: 0`. This will prevent the georeference from updating after it has been initialized once. Your map should look better. 
Let's do this first and then we will see about the other problem!

scially, 2020-07-01:
@laxnpander yes, it's right about the gps data in the images, 
I did not use RTK for this flight, I plan to use RTK to fly once.

After the parameter `update_georef: 0` setting, the effect does not change much
[image]

And, after this position, it has been No Tracking always....
[image]

laxnpander, 2020-07-01:
@scially: It looks a bit better when I run it:
[image]

Though, some artifacts will probably remain even with different settings. I assume in this case the blending is not doing its best job. But that's as good as it gets at the moment. To achieve the same results as me, you may try the most recent build. 

Note: The recent build moved to CMake only as build system. So please refer to the install instructions. Let me know if you encounter any problems.

As for your tracking problem: Doesn't happen to me, can you check if your computer is running out of memory? OpenREALM is quite resource hungry and this is something that happens typically.

laxnpander, 2020-07-07:
@scially Hey, how is it going? Are you happy with your results now or did you have problems with the installation?

scially, 2020-07-07:
@laxnpander , Hey，thanks ....   I plan to re-shoot an area with a drone with RKT, and then clone a copy of the latest project to try it out, and will notify you in time when the results are available,

[image]

laxnpander, 2020-08-04:
I will close this for now. Feel free to reopen or create a new issue if you have more problems / questions.

closed 2020-08-04 by laxnpander

scially, 2020-08-04:
thanks a lot... the OpenREALM is very great..It is worth learning