laxnpander/OpenREALM #26: roslaunch realm_ros alexa_noreco error

TinaBxt, 2020-07-21:
hi, thank you very much for your open source spirit. Now I compiled the project, but an error was reported during runtime. Can you help me find out what the problem is?

roslaunch realm_ros alexa_noreco.launch

error：
2020-07-21 08:59:44.906 (   0.067s) [        9A5D3340]      densification.cpp:52    WARN| Densification launched, but settings forbid.
2020-07-21 08:59:44.906 (   0.067s) [        9A5D3340]      densification.cpp:53    WARN| Try set 'use_sparse_depth' or 'use_dense_depth' in settings. All frames are redirected.
[realm_pose_estimation-3] process has died [pid 11341, exit code -6, cmd /home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/catkin_ws/devel/lib/realm_ros/realm_stage_node __name:=realm_pose_estimation __log:=/home/bxt/.ros/log/76f0445a-caed-11ea-a7c1-00d86120b817/realm_pose_estimation-3.log].
log file: /home/bxt/.ros/log/76f0445a-caed-11ea-a7c1-00d86120b817/realm_pose_estimation-3*.log

TinaBxt, 2020-07-21:
setting /run_id to 76f0445a-caed-11ea-a7c1-00d86120b817
process[rosout-1]: started with pid [11323]
started core service [/rosout]
process[realm_exiv2_grabber-2]: started with pid [11340]
process[realm_pose_estimation-3]: started with pid [11341]
process[realm_densification-4]: started with pid [11342]
process[realm_surface_generation-5]: started with pid [11343]
process[realm_ortho_rectification-6]: started with pid [11344]
process[realm_mosaicing-7]: started with pid [11345]
process[rviz-8]: started with pid [11346]
[ INFO] [1595293184.892678427]: STAGE_NODE [densification]: Loading stage settings from path:
	/home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/densification/stage_settings.yaml
[ INFO] [1595293184.892904989]: STAGE_NODE [densification]: Detected stage type: 'densification'
[ INFO] [1595293184.894360878]: STAGE_NODE [pose_estimation]: Loading stage settings from path:
	/home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/pose_estimation/stage_settings.yaml
[ INFO] [1595293184.894612457]: STAGE_NODE [pose_estimation]: Detected stage type: 'pose_estimation'
[ INFO] [1595293184.902519914]: Exiv2 Grabber Node [Ankommen]: Successfully loaded camera: 
	cx = 610.324
	cy = 516.917
	fx = 1209.07
	fy = 1210.66
	k1 = -0.198806
	k2 = 0.122544
	p1 = -0.000221
	p2 = -0.000877
[ INFO] [1595293184.905241342]: STAGE_NODE [pose_estimation]: : Loading camera from path:
	/home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/camera/calib.yaml
[ INFO] [1595293184.905352018]: STAGE_NODE [pose_estimation]: : Detected camera model: 'pinhole'
[ INFO] [1595293184.905373060]: STAGE_NODE [pose_estimation]: : Loading vslam settings from path:
	/home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/pose_estimation/method/dsm_settings.yaml
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Error: Loading visual slam settings failed. Method 'DSM' not recognized
[ INFO] [1595293184.906117139]: STAGE_NODE [densification]: : Loading camera from path:
	/home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/camera/calib.yaml
[ INFO] [1595293184.906223425]: STAGE_NODE [densification]: : Detected camera model: 'pinhole'
[ INFO] [1595293184.906241349]: STAGE_NODE [densification]: : Loading densifier settings from path:
	/home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/densification/method/dummy_settings.yaml
[ INFO] [1595293184.906308927]: STAGE_NODE [densification]: : Detected densifier type: 'DUMMY'
2020-07-21 08:59:44.906 (   0.067s) [        9A5D3340]      densification.cpp:52    WARN| Densification launched, but settings forbid.
2020-07-21 08:59:44.906 (   0.067s) [        9A5D3340]      densification.cpp:53    WARN| Try set 'use_sparse_depth' or 'use_dense_depth' in settings. All frames are redirected.
[ INFO] [1595293184.909657728]: STAGE_NODE [densification]: Started stage node successfully!
2020-07-21 08:59:44.909 (   0.070s) [        669CF700] worker_thread_base.cpp:57       0| Thread 'Stage [densification]' starting loop...
[ INFO] [1595293184.911362123]: STAGE_NODE [ortho_rectification]: Loading stage settings from path:
	/home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/ortho_rectification/stage_settings.yaml
[ INFO] [1595293184.911591026]: STAGE_NODE [ortho_rectification]: Detected stage type: 'ortho_rectification'
[ INFO] [1595293184.914034233]: Scanning input image for provided meta tags...
[ INFO] [1595293184.914454582]: [FOUND]		'Exif.GPSInfo.GPSAltitude'
[ INFO] [1595293184.914474502]: [FOUND]		'Exif.GPSInfo.GPSLatitude'
[ INFO] [1595293184.914487105]: [FOUND]		'Exif.GPSInfo.GPSLongitude'
[ INFO] [1595293184.914498593]: [FOUND]		'Exif.Image.Model'
[ INFO] [1595293184.914509656]: [FOUND]		'Xmp.exif.REALM.Heading'
[ WARN] [1595293184.914522011]: [NOT FOUND]	'Xmp.exif.REALM.Timestamp'
[ INFO] [1595293184.914551958]: Image #0, image Path: /home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/edm_big_overlap_50p/img_0000.jpg
[ INFO] [1595293184.918102882]: STAGE_NODE [ortho_rectification]: : Loading camera from path:
	/home/bxt/micROS_TAIIC/work/20200706_air_ground_sea_plan/OpenSourceCode/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/camera/calib.yaml
[ INFO] [1595293184.918208584]: STAGE_NODE [ortho_rectification]: : Detected camera model: 'pinhole'
Stage [ortho_rectification]: Created Stage with Settings: 
	<Param>[save_elevation_angle]: 0
	<Param>[save_elevation]: 0
	<Param>[save_valid]: 0
	<Param>[type]: ortho_rectification
	<Param>[save_ortho_gtiff]: 0
	<Param>[queue_size]: 5
	<Param>[save_ortho_rgb]: 0
	<Param>[path_output]: 
	<Param>[GSD]: 0.1
[ INFO] [1595293184.921860034]: STAGE_NODE [ortho_rectification]: Started stage node successfully!
2020-07-21 08:59:44.921 (   0.045s) [        3F7FE700] worker_thread_base.cpp:57       0| Thread 'Stage [ortho_rectification]' starting loop...
[... cut ...] 76f0445a-caed-11ea-a7c1-00d86120b817/realm_pose_estimation-3.log].
log file: /home/bxt/.ros/log/76f0445a-caed-11ea-a7c1-00d86120b817/realm_pose_estimation-3*.log
[ INFO] [1595293185.115047164]: Image #2, image Path: /home/work/OpenSourceCode/edm_big_overlap_50p/img_0002.jpg
[ INFO] [1595293185.215296634]: Image #3, image Path: /home/work/OpenSourceCode/edm_big_overlap_50p/img_0003.jpg
[ INFO] [1595293185.315579823]: Image #4, image Path: /home/work/OpenSourceCode/edm_big_overlap_50p/img_0004.jpg

laxnpander, 2020-07-21:
@TinaBxt: Hey, yes I can help you. Your actual error is here: 
```
terminate called after throwing an instance of 'std::invalid_argument'
what(): Error: Loading visual slam settings failed. Method 'DSM' not recognized
```
To fix it, go to your launch file realm_ros/launch/alexa_noreco.launch and edit the following line: 

`<param name="config/method" type="string" value="dsm"/>`
to
`<param name="config/method" type="string" value="orb_slam2"/>`

It's an error from my side. I will fix it in the master branch soon. Let me know if there are other problems that come up.

Best regards,

Alex

laxnpander, 2020-08-04:
If you have any more trouble feel free to reopen. For now I'll close this then.

closed 2020-08-04 by laxnpander