laxnpander/OpenREALM #28: Docker roslaunch: command not found

winsento, 2020-09-18:
1. Build and run docker
```
./docker/docker_build.sh
./docker/docker_run.sh
```

2. In docker container try to launch:
```
root@eef80d9e8e4e:/# roslaunch realm_ros alexa_reco.launch
bash: roslaunch: command not found
```

3. Install ROS1 brige inside docker container
```
apt-get upgrade && apt-get install -y catkin

mkdir catkin_ws && cd catkin_ws
mkdir src && cd src
git clone https://github.com/laxnpander/OpenREALM_ROS1_Bridge.git
cd .. && catkin_make -DCMAKE_BUILD_TYPE=Release
```
Result:
```
...
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - realm_msgs
-- ~~  - realm_ros
-- ~~  - realm_rviz_plugin
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realm_msgs'
-- ==> add_subdirectory(OpenREALM_ROS1_Bridge/realm_msgs)
-- Could NOT find roscpp (missing: roscpp_DIR)
-- Could not find the required component 'roscpp'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /usr/share/catkin/cmake/catkinConfig.cmake:82 (find_package):
  Could not find a package configuration file provided by "roscpp" with any
  of the following names:

    roscppConfig.cmake
    roscpp-config.cmake

  Add the installation prefix of "roscpp" to CMAKE_PREFIX_PATH or set
  "roscpp_DIR" to a directory containing one of the above files.  If "roscpp"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  OpenREALM_ROS1_Bridge/realm_msgs/CMakeLists.txt:4 (find_package)
```
Docker desktop (2.3.0.5 build 48029) on macos (10.15.6)

laxnpander, 2020-09-22:
Hey @winsento,

 to use the OpenREALM ROS1 Bridge inside your docker container you will need to install ROS as well.
Please refer to this site for more information on how to install it: http://wiki.ros.org/

Let me know if this resolves your problem.

Best regards,

Alex

laxnpander, 2020-10-07:
Reopen in case this didn't fix it for you.

closed 2020-10-07 by laxnpander