laxnpander/OpenREALM #31: docker_build.sh fails because of dependencies missing

donkikos, 2020-10-31:
It seems the error happens because the dependencies are not installed - Dockerfile does not run `install_deps.sh`.

The error:
```
Scanning dependencies of target realm_vslam_base
[ 70%] Building CXX object modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/src/visual_slam_factory.cpp.o
In file included from /cmake_ws/src/OpenREALM/modules/realm_vslam/realm_vslam_base/include/realm_vslam_base/visual_slam_factory.h:31:0,
                 from /cmake_ws/src/OpenREALM/modules/realm_vslam/realm_vslam_base/src/visual_slam_factory.cpp:21:
/cmake_ws/src/OpenREALM/modules/realm_vslam/realm_vslam_base/include/realm_vslam_base/open_vslam.h:29:10: fatal error: openvslam/system.h: No such file or directory
 #include <openvslam/system.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/src/visual_slam_factory.cpp.o] Error 1
modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/build.make:62: recipe for target 'modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/src/visual_slam_factory.cpp.o' failed
make[1]: *** [modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/all] Error 2
CMakeFiles/Makefile2:421: recipe for target 'modules/realm_vslam/realm_vslam_base/CMakeFiles/realm_vslam_base.dir/all' failed
make: *** [all] Error 2
Makefile:129: recipe for target 'all' failed
The command '/bin/sh -c cd cmake_ws/src/OpenREALM && mkdir build && cd build && cmake .. && make' returned a non-zero code: 2
```

laxnpander, 2020-11-02:
@donkikos I will ask my docker expert @ykhedar on this

ykhedar, 2020-11-02:
@donkikos thanks for pointing this out. In fact I knew this problem and had created a new Dockerfile for this and it was called Dockerfile.cmake but forgot to replace the old file with the new one. I have done this now in the latest commit. It works for me. Please check and let us know if the problem exists.

donkikos, 2020-11-03:
@laxnpander @ykhedar thank you! Now it does not fail. 

One related issue though. It does not support CUDA, PSL module is not installed and hence it is not possible to do actual SLAM as far as I understand. Correct me if I am wrong. Any chance we can get a working docker that supports all features?

ykhedar, 2020-11-03:
Sadly currently we have no support for CUDA based docker..For this one should ideally use the nvidia-docker base image and install the other dependecies as per the CPU docker. In addition to that, as you already mentioned, you need to also install PSL for the PSL based densification to work. If I find some more time in the next days, I can also add the nvidia-docker based docker image but dont hope for it to happen soon :). If you are able to do it yourself, we will be happy to assist you in the process.

donkikos, 2020-11-03:
I made a docker for the ROS bridge. Please find the Dockerfile attached to this issue - https://github.com/laxnpander/OpenREALM_ROS1_Bridge/issues/3 .

laxnpander, 2020-11-05:
@donkikos Thanks for your work. We will check it out! @ykhedar: What do you think? Can we use this docker for the ROS bridge? And if yes, where is a smart place to put it? Probably inside the ROS Bridge repo due to the fact it is installing ROS?

donkikos, 2020-11-05:
Maybe you can make it as a multistage Dockerfile and build different targets for library only or ROS bridge included. 

Also for no CUDA use Dockerfile based ok ROS images is much faster to build. OpenCV build is taking too long when CUDA is present. Another option is to switch to newer OpenCV version in ubuntu repos (if they support CUDA).

ykhedar, 2020-11-14:
> 
> 
> @donkikos Thanks for your work. We will check it out! @ykhedar: What do you think? Can we use this docker for the ROS bridge? And if yes, where is a smart place to put it? Probably inside the ROS Bridge repo due to the fact it is installing ROS?

Wonderful @donkikos ! @laxnpander yes you are right, it makes sense to put it in the ROS Bridge repo.

laxnpander, 2020-11-16:
@donkikos Alright, do you want to create a PR for that docker? Then you are listed as contributer

closed 2021-03-17 by laxnpander