laxnpander/OpenREALM #38: Race Condition if Open VSlam reinitializes resulting in crash

zthorson, 2021-01-06:
If OpenVslam ends up reinitializing after completely losing tracking, there appears to be a race condition that causes the map iterator to crash entirely.

This occurs more frequently when we are rapidly feeding frames to the system.

I will update it with the crash log at next occurance.

laxnpander, 2021-01-07:
Do you know if it's a crash inside OpenVSLAM or if it is located in some OpenREALM module?

zthorson, 2021-01-07:
I changed how I perform some calls and it hasn't been happening as often.  I will post a stack trace once I get it to reoccur.  It is relatively rare as it requires a dataset that tracks for a bit, fails, and regains tracking.  The more frequent occurrence when tracking is lost is that it is never regained.

Either way, I will post the full stack trace if I see it occur again.  It has occured less since I tweaked my camera configuration.

zthorson, 2021-01-07:
This might indicate a lock contention issue, or a lock that isn't quite being cleaned up properly.  

You can ignore the g3logger portion of the stack trace.  We use g3logger in our wrapper code, which is set up to catch crashes and create the stack trace.

```
[2021-01-07 15:49:44.062] [info] initialization succeeded with H
15:49:44.068497 I Using:/home/zach/Datasets/Quickstitch/2020.10.08/QuickStitchFrameCache/2020_10_08_13_30_22/9389.jpg
15:49:44.069547 I Pose Pose : output/pose/gnss
15:49:44.069562 I Queue depth: 1
15:49:44.329724 I Using:/home/zach/Datasets/Quickstitch/2020.10.08/QuickStitchFrameCache/2020_10_08_13_30_22/9520.jpg
15:49:44.330894 I Pose Pose : output/pose/gnss
15:49:44.330911 I Queue depth: 2
15:49:44.416330 I Using:/home/zach/Datasets/Quickstitch/2020.10.08/QuickStitchFrameCache/2020_10_08_13_30_22/9589.jpg
15:49:44.417379 I Pose Pose : output/pose/gnss
15:49:44.417395 I Queue depth: 3
[2021-01-07 15:49:44.439] [info] new map created with 113 points: frame 23 - frame 25
2021/01/07 15:49:44 446077

***** FATAL SIGNAL RECEIVED ******* 
Received fatal signal: SIGABRT(6)	PID: 196686

***** SIGNAL SIGABRT(6)

*******	STACKDUMP *******
	stack dump [1]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/g3log/libg3logger.so.1.3.2-122+0x5dc7a [0x7f5d8f0c3c7a]
	stack dump [2]  /lib/x86_64-linux-gnu/libpthread.so.0+0x153c0 [0x7f5d8f01c3c0]
	stack dump [3]  /lib/x86_64-linux-gnu/libc.so.6gsignal+0xcb [0x7f5d8d63e18b]
	stack dump [4]  /lib/x86_64-linux-gnu/libc.so.6abort+0x12b [0x7f5d8d61d859]
	stack dump [5]  /lib/x86_64-linux-gnu/libc.so.6+0x25729 [0x7f5d8d61d729]
	stack dump [6]  /lib/x86_64-linux-gnu/libc.so.6+0x36f36 [0x7f5d8d62ef36]
	stack dump [7]  /lib/x86_64-linux-gnu/libpthread.so.0+0xbed9 [0x7f5d8f012ed9]

	stack dump [8]  /usr/local/lib/libopenvslam.so : openvslam::data::keyframe::get_landmarks() const+0x2d [0x7f5d8cb031cd]

	stack dump [9]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_vslam/realm_vslam_base/libopen_realm_vslam_base-1.0.0.so : realm::OpenVslam::getTrackedMapPoints() const+0x64 [0x7f5d8ef7aac0]

	stack dump [10]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_vslam/realm_vslam_base/libopen_realm_vslam_base-1.0.0.so : realm::OpenVslam::track(std::shared_ptr<realm::Frame>&, cv::Mat const&)+0x460 [0x7f5d8ef7a67c]

	stack dump [11]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_stages/libopen_realm_stages-1.0.0.so : realm::stages::PoseEstimation::track(std::shared_ptr<realm::Frame>&)+0x17c [0x7f5d8f1f2032]

	stack dump [12]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_stages/libopen_realm_stages-1.0.0.so : realm::stages::PoseEstimation::process()+0x138 [0x7f5d8f1f1976]

	stack dump [13]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_core/libopen_realm_core-1.0.0.so : realm::WorkerThreadBase::run()+0x39e [0x7f5d8ebbb65c]

	stack dump [14]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_core/libopen_realm_core-1.0.0.so : void std::__invoke_impl<void, void (realm::WorkerThreadBase::*&)(), realm::WorkerThreadBase*&>(std::__invoke_memfun_deref, void (realm::WorkerThreadBase::*&)(), realm::WorkerThreadBase*&)+0x6a [0x7f5d8ebef01e]

	stack dump [15]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_core/libopen_realm_core-1.0.0.so : std::__invoke_result<void (realm::WorkerThreadBase::*&)(), realm::WorkerThreadBase*&>::type std::__invoke<void (realm::WorkerThreadBase::*&)(), realm::WorkerThreadBase*&>(void (realm::WorkerThreadBase::*&)(), realm::WorkerThreadBase*&)+0x4a [0x7f5d8ebeef56]

	stack dump [16]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_core/libopen_realm_core-1.0.0.so : void std::_Bind<void (realm::WorkerThreadBase::*(realm::WorkerThreadBase*))()>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>)+0x5b [0x7f5d8ebeeeb1]

	stack dump [17]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_core/libopen_realm_core-1.0.0.so : void std::_Bind<void (realm::WorkerThreadBase::*(realm::WorkerThreadBase*))()>::operator()<, void>()+0x37 [0x7f5d8ebeee35]

	stack dump [18]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_core/libopen_realm_core-1.0.0.so : void std::__invoke_impl<void, std::_Bind<void (realm::WorkerThreadBase::*(realm::WorkerThreadBase*))()>>(std::__invoke_other, std::_Bind<void (realm::WorkerThreadBase::*(realm::WorkerThreadBase*))()>&&)+0x24 [0x7f5d8ebeeddd]

	stack dump [19]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-debug/quickstitch/3rdparty/OpenREALM/modules/realm_core/libopen_realm_core-1.0.0.so : std::__invoke_result<std::_Bind<void (realm::WorkerThreadBase::*(realm::WorkerThreadBase*))()>>::type std::__invoke<std::_Bind<void (realm::WorkerThreadBase::*(realm::WorkerThreadBase*))()>>(std::_Bind<void (realm::WorkerThreadBase::*(realm::WorkerThreadBase*))()>&&)+0x33 [0x7f5d8ebeed72]

[... cut ...]
	stack dump [24]  /lib/x86_64-linux-gnu/libpthread.so.0+0x9609 [0x7f5d8f010609]
	stack dump [25]  /lib/x86_64-linux-gnu/libc.so.6clone+0x43 [0x7f5d8d71a293]

Exiting after fatal event  (FATAL_SIGNAL). Fatal type:  SIGABRT
Log content flushed sucessfully to sink

g3log g3FileSink shutdown at: 15:49:44 446739
Log file at: [./logs/stitching.log.20210107-154938.log]

exitWithDefaultSignalHandler:229. Exiting due to FATAL_SIGNAL, 6   

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

laxnpander, 2021-01-08:
@zthorson Hmm, now that you say it I think I am not even catching an internal reset of OpenVSLAM. I implemented it for ORB SLAM but had to dig quite deep into the original code base to do it. I wanted to avoid that with OpenVSLAM so you can basically always just pull the master branch. It shouldn’t be rocket science to implement, we just have to find a way to check whether or not a reset has happened. Maybe through the number of keyframes or something similar. I will think about it on Monday!

laxnpander, 2021-03-19:
@zthorson: I am having a rough look at it today. My datasets haven't caused this issue yet, but I will try to implement a check on OpenREALM side whether openvslam did reset or not. My guess would be, that this causes the segfault. If for example openvslam resets and OpenREALM is still trying to access deleted frames/keyframes, this could lead to something like this. 

The solution should be pretty straightforward, in this line https://github.com/laxnpander/openvslam/blob/84290a7dcdf93fbfdf56c98ab2eb6f2d3d98baae/src/openvslam/tracking_module.cc#L151 openvslam is resetting the map_db_, which is indirectly accessed by OpenREALM through the map_publisher in https://github.com/laxnpander/OpenREALM/blob/beeb3f0e4452a98b59fbd3a3dbe250e86b2ec8e2/modules/realm_vslam/realm_vslam_base/src/open_vslam.cpp#L83 . So by checking whether the databse is empty or not we can check if a reset has happened. Not sure if it definitely prevents your issue, but it is required anyway.

zthorson, 2021-03-22:
@laxnpander I spent some time on Friday trying to make datasets where the system would lose and regain tracking, but I wasn't able to reproduce the error.  I think it's mostly because the datasets I generated would lose tracking, but never reset to try to regain it.  Even the dataset that originally was generating errors wasn't cooperating with my attempt to recreate it.  So whatever it is, I'm guessing there is a subtle timing element involved as well.

I'll probably try putting the fix you found to try in and see if the issue occurs during some of our other testing.

laxnpander, 2021-03-22:
Hmm okay, that makes it definitely harder to debug. I could imagine that a keyframe is being accessed, that was already deleted. Or is being deleted after I grab it. One may need to make sure this is either a deep copy or it lives long enough to live through the function.

laxnpander, 2021-03-22:
Just had a look at the openvslam code. The good news is: There seems to be no danger of the keyframe objects being deleted before an access. The bad news: objects are not deleted at all, so openvslam is leaking: https://github.com/laxnpander/openvslam/blob/84290a7dcdf93fbfdf56c98ab2eb6f2d3d98baae/src/openvslam/data/map_database.cc#L39

Not sure why they didn’t go with smart pointers here, would eradicate problems like these. But for our problem this means the segfault is not coming from deleted keyframes.

zthorson, 2021-03-25:
@laxnpander I'll have to look at that pull request and see if I should fold it in to my build.

In the meantime, I can artificially trigger the stack trace failure with the following code in the OpenVslam track function.  In the past, this reset of OpenVSLAM occurred on it's own, without an external reset call.  

When that happens, it appears that the OpenVslamKeyframeUpdater eventually accesses a bad landmark pointer while inside it's read loop.

I think we should be able to detect this internal reset by checking for a transition from Tracking/Lost to Uninitialized/Initialized.  If that transition occurs after feeding a keyframe, then we know the system reset on us.  I'll work on it a bit and put up a pull request once I've tested it a bit.

```
static long force_reset_cnt = 0;

VisualSlamIF::State OpenVslam::track(Frame::Ptr &frame, const cv::Mat &T_c2w_initial)
{
  force_reset_cnt++;
  if (force_reset_cnt > 100) {
      force_reset_cnt = 0;
      LOG_F(INFO, "Forcing vslam reset!");
      m_vslam->request_reset();
      //reset();
  }
```

zthorson, 2021-03-25:
I've managed to detect a reset by watching the current tracker state for a transition, and this drastically reduces the number of crashes.   

See my changes here:
https://github.com/zthorson/OpenREALM/commit/95b0e460e17c9a574f8360671b9fe3a553512f39

We now only segfault 1/10 runs with the trigger code above, rather than every run.   I think we are getting into a condition where our vslam has reset, but the process command in the OpenVslamKeyframeUpdater has already started.  So the keyframes get removed out from under the updater.

The heavy handed approach would be to use a shared lock across the OpenVslamKeyframeUpdater::process() and OpenVslam::track(), though this would likely have a small performance impact. 

This results in the slightly different stacktrace:
```
2021-03-25 13:07:46.161 (   5.704s) [Stage [pose_esti]         open_vslam.cpp:60       0| Forcing vslam reset!
2021-03-25 13:07:46.161 (   5.704s) [        3BFF7700]geometric_referencer.cp:119      0| ### GEOREFERENCE ABORTED ###
2021-03-25 13:07:46.161 (   5.704s) [        3BFF7700]geometric_referencer.cp:120      0| Unique frames: 7
13:07:46.161324 I Pose Image: debug/tracked
[2021-03-25 13:07:46.163] [info] resetting system
13:07:46.168191 I Using: /home/zach/Datasets/Quickstitch/TestFlights/2021_03_17_14_14_40/39612.jpg
13:07:46.170201 I Pose Pose : output/pose/gnss
[2021-03-25 13:07:46.275] [info] reset mapping module
[2021-03-25 13:07:46.280] [info] reset global optimization module
[2021-03-25 13:07:46.282] [info] clear BoW database
2021/03/25 13:07:46 292781

***** FATAL SIGNAL RECEIVED ******* 
Received fatal signal: SIGABRT(6)	PID: 188314

***** SIGNAL SIGABRT(6)

*******	STACKDUMP *******
	stack dump [1]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-linux-release/quickstitch/3rdparty/g3log/libg3logger.so.1.3.2-122+0x152bb [0x7f87cccc92bb]
	stack dump [2]  /lib/x86_64-linux-gnu/libpthread.so.0+0x153c0 [0x7f87ccc693c0]
	stack dump [3]  /lib/x86_64-linux-gnu/libc.so.6gsignal+0xcb [0x7f87cbd2818b]
	stack dump [4]  /lib/x86_64-linux-gnu/libc.so.6abort+0x12b [0x7f87cbd07859]
	stack dump [5]  /lib/x86_64-linux-gnu/libstdc++.so.6+0x9e951 [0x7f87cc0de951]
	stack dump [6]  /lib/x86_64-linux-gnu/libstdc++.so.6+0xaa47c [0x7f87cc0ea47c]
	stack dump [7]  /lib/x86_64-linux-gnu/libstdc++.so.6+0xaa4e7 [0x7f87cc0ea4e7]
	stack dump [8]  /lib/x86_64-linux-gnu/libstdc++.so.6+0xaa799 [0x7f87cc0ea799]

	stack dump [9]  /lib/x86_64-linux-gnu/libstdc++.so.6 : std::__throw_system_error(int)+0x98 [0x7f87cc0e177f]

	stack dump [10]  /usr/local/lib/libopenvslam.so : openvslam::data::landmark::will_be_erased()+0x76 [0x7f87c7bca966]

	stack dump [11]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-linux-release/quickstitch/3rdparty/OpenREALM/modules/realm_vslam/realm_vslam_base/libopen_realm_vslam_base-1.0.0.so : realm::OpenVslamKeyframeUpdater::process()+0x351 [0x7f87cbc66f01]

	stack dump [12]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-linux-release/quickstitch/3rdparty/OpenREALM/modules/realm_core/libopen_realm_core-1.0.0.so : realm::WorkerThreadBase::run()+0x378 [0x7f87cbbd8408]
	stack dump [13]  /lib/x86_64-linux-gnu/libstdc++.so.6+0xd6d84 [0x7f87cc116d84]
	stack dump [14]  /lib/x86_64-linux-gnu/libpthread.so.0+0x9609 [0x7f87ccc5d609]
	stack dump [15]  /lib/x86_64-linux-gnu/libc.so.6clone+0x43 [0x7f87cbe04293]

Exiting after fatal event  (FATAL_SIGNAL). Fatal type:  SIGABRT
Log content flushed sucessfully to sink

terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
2021/03/25 13:07:46 292781

***** FATAL SIGNAL RECEIVED ******* 
Received fatal signal: SIGABRT(6)	PID: 188314

***** SIGNAL SIGABRT(6)

*******	STACKDUMP *******
	stack dump [1]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-linux-release/quickstitch/3rdparty/g3log/libg3logger.so.1.3.2-122+0x152bb [0x7f87cccc92bb]
	stack dump [2]  /lib/x86_64-linux-gnu/libpthread.so.0+0x153c0 [0x7f87ccc693c0]
	stack dump [3]  /lib/x86_64-linux-gnu/libc.so.6gsignal+0xcb [0x7f87cbd2818b]
	stack dump [4]  /lib/x86_64-linux-gnu/libc.so.6abort+0x12b [0x7f87cbd07859]
	stack dump [5]  /lib/x86_64-linux-gnu/libstdc++.so.6+0x9e951 [0x7f87cc0de951]
	stack dump [6]  /lib/x86_64-linux-gnu/libstdc++.so.6+0xaa47c [0x7f87cc0ea47c]
	stack dump [7]  /lib/x86_64-linux-gnu/libstdc++.so.6+0xaa4e7 [0x7f87cc0ea4e7]
	stack dump [8]  /lib/x86_64-linux-gnu/libstdc++.so.6+0xaa799 [0x7f87cc0ea799]

	stack dump [9]  /lib/x86_64-linux-gnu/libstdc++.so.6 : std::__throw_system_error(int)+0x98 [0x7f87cc0e177f]

	stack dump [10]  /usr/local/lib/libopenvslam.so : openvslam::data::landmark::will_be_erased()+0x76 [0x7f87c7bca966]

	stack dump [11]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-linux-release/quickstitch/3rdparty/OpenREALM/modules/realm_vslam/realm_vslam_base/libopen_realm_vslam_base-1.0.0.so : realm::OpenVslamKeyframeUpdater::process()+0x351 [0x7f87cbc66f01]

	stack dump [12]  /home/zach/Source/quickstitch/real-time-stitching/cmake-build-linux-release/quickstitch/3rdparty/OpenREALM/modules/realm_core/libopen_realm_core-1.0.0.so : realm::WorkerThreadBase::run()+0x378 [0x7f87cbbd8408]
	stack dump [13]  /lib/x86_64-linux-gnu/libstdc++.so.6+0xd6d84 [0x7f87cc116d84]
	stack dump [14]  /lib/x86_64-linux-gnu/libpthread.so.0+0x9609 [0x7f87ccc5d609]
	stack dump [15]  /lib/x86_64-linux-gnu/libc.so.6clone+0x43 [0x7f87cbe04293]

Exiting after fatal event  (FATAL_SIGNAL). Fatal type:  SIGABRT
Log content flushed sucessfully to sink

g3log g3FileSink shutdown at: 13:07:46 293033
Log file at: [./stitching.log.20210325-130740.log]

exitWithDefaultSignalHandler:229. Exiting due to FATAL_SIGNAL, 6   

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

zthorson, 2021-03-25:
@laxnpander 
Am I misunderstanding something, or does this iterator never increment?  At least on my copy, once the first 'process' is called, the function just spins on the first iterator item over and over.  Eventually, that item is deleted and we get a fault.

https://github.com/laxnpander/OpenREALM/blob/465ecc8a68a55dd7fee956af01b56169ea6db231/modules/realm_vslam/realm_vslam_base/src/open_vslam.cpp#L234-L282

laxnpander, 2021-03-25:
@zthorson: Good catch! Looks like there is an it++ missing! Maybe it got deleted by accident. I don’t see any  reason why it shouldn’t increment

zthorson, 2021-03-25:
@laxnpander 
It doesn't resolve our keyframe directly, but since the iterator no longer locks onto the first keyframe and examines it over and over, we have less of a chance of iterating over a keyframe marked as deleted or one that is missing.  You are right in that there isn't any explicit cleanup, so I'm not sure why the data is missing.

Good catch on the logical OR, I'll add that to my pull request.  I looked straight at that and didn't catch it.

zthorson, 2021-03-25:
@laxnpander Fixes short of the lm one discussed above are in pull request https://github.com/laxnpander/OpenREALM/pull/49

laxnpander, 2021-03-26:
> @laxnpander
> It doesn't resolve our keyframe directly, but since the iterator no longer locks onto the first keyframe and examines it over and over, we have less of a chance of iterating over a keyframe marked as deleted or one that is missing. You are right in that there isn't any explicit cleanup, so I'm not sure why the data is missing.
> 
> ```
> if (!lm || lm->will_be_erased())
> ```
> 
> Looking closer, I think that statement is correct.
> If lm is null, !lm is TRUE, which should short circuit the second call, which is the behavior you would want.

Hmm, yes I guess you are right about that. I don't see why it would fail at this call then. It would only make sense, if the object behind it is deleted and not set to NULL properly but points just to some place in memory not associated with a landmark object.

But first I will merge your PR, looks good to me!

zthorson, 2021-04-23:
I haven't see this particular issue reoccur since https://github.com/laxnpander/OpenREALM/pull/49 was merged.  We can probably close this out for now.

laxnpander, 2021-04-24:
@zthorson Actually, in my current data set I get some fatal locking error as soon as openvslam resets. Haven’t found the time to investigate yet. Gonna reopen until I had a look into it

laxnpander, 2021-05-11:
Maybe it is resolved. Error does not happen anymore. Not sure if I was missing this PR back then. Closing until it reoccures.

laxnpander, 2021-05-11:
As suggested by @zthorson the keyframe updater is started while an internal reset is happening. OpenVSLAM invalidates data and consequently all hell breaks lose (system_error). This only happens on challenging datasets with lots of resets. 

Option 1: Shared mutex for OpenVslam::track() and OpenVslamKeyframeUpdater::process()

Sharing a mutex seems to be an option, though we have a bit of a problem here. Let's assume the updater tries to acquire the lock, but it is currently locked by the tracker. It'll sit and wait. The tracker however detects a slam reset and requests the updater to reset as well. This thread however still sits and waits for the lock. So its process is not initiated until the lock is free and only after that it resets as well. 

Option 2: Get rid of parallel threading of keyframe updater and move function into OpenVslam class

I currently prefer this option. Robustness will always be more important than performance. Also what purpose has multi threading if they are not actually computing in parallel? 

=> Will implement Option 2 and see if I can measure some performance differences.

laxnpander, 2021-05-11:
We could think about triggering the update function only after a keyframe insertion and then let it process in a detached thread. That way it only becomes a problem if it takes longer to update the keyframes than it takes the slam to do a full tracking cycle. Though, if the stars align and it actually happens, it might be even harder to debug this.

zthorson, 2021-05-17:
@laxnpander I'll be taking a closer look at some of the crashes this week.    Have you made any progress in a branch I can start with, or should I start from scratch?

laxnpander, 2021-05-17:
@zthorson Current dev build is an implementation without the keyframe updater. Also pose estimation is now correctly reset when openvslam resets. I don’t think I had any more issues since then.

zthorson, 2021-05-17:
@laxnpander I had a chance to test this on one of the datasets that would reset frequently.  I ran the set 30 times with the old dev branch, and 30x with the new:

```
Old Branch: 10 Good, 15 Hangs (no crash, just stopped processing), 1 Crash
New Branch: 30 Good
```

It looks very promising to me and performance on my local machine doesn't seem affected.  I'll work on getting it deployed to iOS and see if I get the same good results.

laxnpander, 2021-05-18:
@zthorson Sounds good, glad to hear! I think performance is not affected, because keyframe updates mostly happen in the first few minutes. After a while there is no updates anymore. Not exactly sure why. My guess is that frames are published so quickly, that they are not in memory anymore to be updated. This is a bit weird, because I intended it to be delayed behind real-time so there is at least a few updates. But that's something for another issue. 

One last thing I'd suggest is making the current keyframe update thread a member, so in the unlikely event that the update process takes very long we are not creating new threads and detach them indefinitely. One update thread should be enough.

zthorson, 2021-05-18:
@laxnpander
I can think of a few ways to handle it.  If we think it's very unlikely to occur, then we could just reject updates that come in if we are still busy updating a previous thread.  Or we could ensure our last update finishes before starting the next one.  

If we think it's rare, skipping an update might be preferred, otherwise we could block until the next one occurs.

Otherwise we could go with the heavier approach of using a worker_thread_base and a separate queue, but that seems like it might be overkill for this.

There is a pretty straightforward way to check if a thread has finished running using std::async if we are looking for a simple approach, I had a sample commit below.  If this fufills what you are thinking, I can make a pull request:
https://github.com/zthorson/OpenREALM/commit/33d147649cd4bd7eaeded718a1ce5d1e0f26b0ff

laxnpander, 2021-05-19:
@zthorson Looks good to me! Let's do it!

closed 2021-05-19 by laxnpander