laxnpander/OpenREALM #43: g2o build error

Airplane5, 2021-03-02:
While install_deps.sh building g2o, CMake reported error:

/usr/bin/ld: 找不到(**cannot find**) -lopengl_helper
collect2: error: ld returned 1 exit status
g2o/apps/g2o_viewer/CMakeFiles/viewer_library.dir/build.make:231: recipe for target '../lib/libg2o_viewer.so' failed
make[2]: *** [../lib/libg2o_viewer.so] Error 1
CMakeFiles/Makefile2:504: recipe for target 'g2o/apps/g2o_viewer/CMakeFiles/viewer_library.dir/all' failed
make[1]: *** [g2o/apps/g2o_viewer/CMakeFiles/viewer_library.dir/all] Error 2

My CMake version is 3.15.4.
Because unknown reason, I have to delete
`
if [ "${CMAKE_VERSION_MINOR}" -le 9 ]; then

  echo 'CMake Version is too old! Trying to download newer version '
  if [ $ARCH == "aarch64" ]
    CMAKE_FILE="cmake-3.10.3"

    echo 'Architecture is aarch64. There is no cmake binaries available. Compiling from source is required. This takes a while.'
    while true
    do
      read -p "Should cmake be compiled from sources now? (yes/no)" yn
      case $yn in
        [Yy]* ) break;;
        [Nn]* ) exit;;
        * ) echo "Please answer yes or no.";;
      esac
    done
  else if [ $ARCH == "x86_64" ] #; then
    CMAKE_FILE="cmake-3.10.3-Linux-x86_64"
  fi

  # Check if file already exists
  if [ ! -e "${CMAKE_FILE}.tar.gz" ]; then
    wget https://cmake.org/files/v3.10/${CMAKE_FILE}.tar.gz
  

  # Remove existing unpacked cmake folder
  if [ -d "${CMAKE_FILE}" ]; then
    rm -r ${CMAKE_FILE}
  fi

  tar xvzf ${CMAKE_FILE}.tar.gz

  if [ $ARCH == "aarch64" ]
    ./configure
    sudo make install
  else
    export PATH="`pwd`/${CMAKE_FILE}/bin:$PATH"
  fi
fi
`

I also deleted 
`git checkout 9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a`

how can I solve this error?

Airplane5, 2021-03-02:
there is a warning too:
`In file included from /usr/local/include/eigen3/Eigen/Core:450,
                 from /home/wlh/g2o/g2o/core/eigen_types.h:30,
                 from /home/wlh/g2o/build/g2o/config.h:46,
                 from /home/wlh/g2o/g2o/stuff/misc.h:31,
                 from /home/wlh/g2o/g2o/examples/tutorial_slam2d/se2.h:30,
                 from /home/wlh/g2o/g2o/examples/tutorial_slam2d/simulator.h:30,
                 from /home/wlh/g2o/g2o/examples/tutorial_slam2d/simulator.cpp:27:
/usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h: In member function ‘void g2o::tutorial::Simulator::simulate(int, const g2o::tutorial::SE2&)’:
/usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h:194:66: warning: ‘*((void*)(&<anonymous>)+80).Eigen::DenseStorage<double, 9, 3, 3, 0>::m_data’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     DenseStorage(const DenseStorage& other) : m_data(other.m_data) {
                                                                  ^
/usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h:194:66: warning: ‘*((void*)(&<anonymous>)+16).Eigen::DenseStorage<double, 2, 2, 1, 0>::m_data’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     DenseStorage(const DenseStorage& other) : m_data(other.m_data) {
                                                                  ^
/usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h:194:66: warning: ‘*((void*)(&<anonymous>)+64).Eigen::DenseStorage<double, 4, 2, 2, 0>::m_data’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     DenseStorage(const DenseStorage& other) : m_data(other.m_data) {
                                                                  ^
`

laxnpander, 2021-03-04:
@Airplane5 A bit messy your problem description. Looks like your main error is "-lopengl_helper" missing. Can you try this: https://github.com/RainerKuemmerle/g2o/issues/362

Airplane5, 2021-03-04:
Thank U very much and apology for my poor English!
Actually the problem is "-lopengl_helper" missing. I will try solution in the link.
Thanks again for your reply.

laxnpander, 2021-03-04:
@Airplane5 No problem, let me know if this resolves the issue. What kind of system do you use? Ubuntu 18.04? 20.04? Different flavor? At least in my builds this problem never occured.

Airplane5, 2021-03-05:
I try the solution but there is a new error now. 
System version is 18.04. My senior fellow also told me he hadn't meet error when compiling g2o. 
Thanks for your concern.

------------------ 原始邮件 ------------------

 
@Airplane5 No problem, let me know if this resolves the issue. What kind of system do you use? Ubuntu 18.04? 20.04? Different flavor? At least in my builds this problem never occured.

laxnpander, 2021-03-16:
@Airplane5: What does the new error look like? Without some better description I can't help you fix your problem.

Airplane5, 2021-03-16:
@laxnpander 
Sorry for not telling u that I had built g20 successfully.
I just have some problem with network when rosdep update. I think it's easy to tackle.
Thank you so much!!!!!!
You can close now.

laxnpander, 2021-03-17:
@Airplane5: You are welcome :) Let me know if you have further problems to get it running.

closed 2021-03-17 by laxnpander