laxnpander/OpenREALM #46: Running on test dataset error.

xycai-pku, 2021-03-13:
When running roslaunch realm_ros alexa_noreco.launch on dataset edm_big_overlap_50p using default profiles, I didn't see any ouput in the Rviz window.
[image]

And here's the terminal's output:
[image]

Do I need to change some configs to run the test dataset?

laxnpander, 2021-03-15:
@xycai-pku: Usually not, the first processing stage seems to fail. But there is no indication why. Can you do the following:
- Edit the launch file you are starting (realm_ros/launch/alexa_noreco.launch)
- Replace 
```
<node pkg="realm_ros" type="realm_stage_node" name="realm_pose_estimation" output="screen">
```
with
```
<node pkg="realm_ros" type="realm_stage_node" name="realm_pose_estimation" output="screen" launch-prefix="xterm -e gdb --args">
```
- Start the launch file again. There will be a terminal popping up, inside you enter "run", wait for it to crash and then type "bt". Then send me the output of the console.

Another suggestion, send me this file: realm_ros/output/some date and time/pose_estimation/stage.log.

xycai-pku, 2021-03-16:
@laxnpander Here is the terminal‘s output：
```#0  0x00007fffed82e69d in Eigen::JacobiSVD<Eigen::Matrix<double, -1, 9, 0, -1, 9>, 2>::compute(Eigen::Matrix<double, -1, 9, 0, -1, 9> const&, unsigned int) () from /usr/local/lib/libopenvslam.so  

#1  0x00007fffed8141ee in openvslam::solve::homography_solver::compute_H_21(std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&) ()
   from /usr/local/lib/libopenvslam.so  

#2  0x00007fffed814e69 in openvslam::solve::homography_solver::find_via_ransac(unsigned int, bool) () from /usr/local/lib/libopenvslam.so  

#3  0x00007ffff70826df in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6  

#4  0x00007ffff52a56db in start_thread (arg=0x7fffb8a6a700)  at pthread_create.c:463  

#5  0x00007ffff6add71f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95  
```
And here's realm_ros/output/some date and time/pose_estimate/stage.log:

```File verbosity level: 9
date       time         ( uptime  ) [ thread name/id ]                   file:line     v| 
2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]               loguru.h:2056     0| Logging to '/home/xycai/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/output/21-03-16_01-52-04/pose_estimation/stage.log', mode: 'a', verbosity: 9

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]         stage_base.cpp:38       0| Successfully initialized!

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]         stage_base.cpp:39       0| Stage path set to: 

/home/xycai/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/output/21-03-16_01-52-04/pose_estimation

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:319      0| ### Stage process settings ###  

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:320      0| - use_vslam: 1

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:321      0| - use_fallback: 0

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:322      0| - do_update_georef: 0

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:323      0| - do_suppress_outdated_pose_pub: 0

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:324      0| - th_error_georef: 1.00

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:325      0| - overlap_max: 100.00

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:326      0| - overlap_max_fallback: 95.00

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:328      0| ### Stage save settings ###

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:329      0| - save_trajectory_gnss: 1

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:330      0| - save_trajectory_visual: 1

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:331      0| - save_frames: 0

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:332      0| - save_keyframes: 0

2021-03-16 01:52:04.434 (   0.301s) [        F7F47680]    pose_estimation.cpp:333      0| - save_keyframes_full: 0

2021-03-16 01:52:04.434 (   0.302s) [Stage [pose_esti] worker_thread_base.cpp:43       0| Thread 'Stage [pose_estimation]' starting loop...

2021-03-16 01:52:04.485 (   0.352s) [Stage [pose_esti]    pose_estimation.cpp:139      0| Processing frame #111 with timestamp 1615884724407366389!

2021-03-16 01:52:04.485 (   0.353s) [Stage [pose_esti]    pose_estimation.cpp:199      0| Tracking frame #111 in visual SLAM...!

2021-03-16 01:52:04.530 (   0.397s) [Publisher [pose_]               loguru.h:2056     0| Logging to 
'/home/xycai/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/output/21-03-16_01-52-04/pose_estimation/publisher.log', mode: 'a', verbosity: 9

2021-03-16 01:52:04.530 (   0.397s) [Publisher [pose_]    pose_estimation.cpp:513      0| Successfully initialized pose_estimation publisher!

2021-03-16 01:52:04.579 (   0.446s) [Stage [pose_esti]    pose_estimation.cpp:225   WARN| No tracking.

2021-03-16 01:52:04.579 (   0.447s) [Stage [pose_esti]    pose_estimation.cpp:144      0| Timing [Tracking]: 94 ms

2021-03-16 01:52:04.579 (   0.447s) [Stage [pose_esti] worker_thread_base.cpp:84       0| Timing [Total]: 94 ms

2021-03-16 01:52:04.579 (   0.447s) [Stage [pose_esti]    pose_estimation.cpp:139      0| Processing frame #112 with timestamp 1615884724508501556!

2021-03-16 01:52:04.579 (   0.447s) [Stage [pose_esti]    pose_estimation.cpp:199      0| Tracking frame #112 in visual SLAM...!

2021-03-16 01:52:04.630 (   0.498s) [Publisher [pose_]    pose_estimation.cpp:566      0| Publishing pose of frame #111...
```

I'm using Ubuntu 18.04 with ros melodic in VMware.

laxnpander, 2021-03-17:
@xycai-pku: Hmmm, virtual machines come with their own challenges. I can't really provide any help there. Your problem is nothing I am aware of, so it could be related to the VM. Maybe not enough memory assigned? OpenREALM is quite hungry with the visualization. Another thing you could try is to remove open vslam and G2O and rebuild them with ```-DBUILD_WITH_MARCH_NATIVE=OFF```. Currently it is turned ON and sometimes causes problems with Eigen.

xycai-pku, 2021-03-17:
@laxnpander Thanks for your advice! I may compile OpenREALM on a PC later, now I'm going to close this issue.

closed 2021-03-17 by laxnpander

kintzhao, 2022-04-25:
roslaunch realm_ros  alexa_noreco.launch 
I have the same error.

 
I0425 23:09:25.294509 30821 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.298725 30820 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.300248 30822 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
E0425 23:09:25.295591 31085 stage_node_main.cpp:32] *** Aborted at 1650899365 (unix time) try "date -d @1650899365" if you are using GNU date ***
I0425 23:09:25.302570 30819 stage_node.cpp:112] _dir_date_time:22-04-25_23-09-21
I0425 23:09:25.304400 30819 stage_node.cpp:122] pub _gnss_base:
I0425 23:09:25.308743 30820 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.310401 30822 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.312570 30819 stage_node.cpp:112] _dir_date_time:22-04-25_23-09-21
I0425 23:09:25.312892 30819 stage_node.cpp:122] pub _gnss_base:
I0425 23:09:25.315364 30821 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.315534 30821 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.318637 30820 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.320111 30822 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.322489 30819 stage_node.cpp:112] _dir_date_time:22-04-25_23-09-21
I0425 23:09:25.322605 30819 stage_node.cpp:122] pub _gnss_base:
E0425 23:09:25.323042 31085 stage_node_main.cpp:32] PC: @     0x7fa53d14e909 Eigen::JacobiSVD<>::compute()
I0425 23:09:25.324543 30821 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
E0425 23:09:25.327814 31085 stage_node_main.cpp:32] *** SIGSEGV (@0x0) received by PID 30819 (TID 0x7fa4eaffd700) from PID 0; stack trace: ***
I0425 23:09:25.330236 30822 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
E0425 23:09:25.330626 31085 stage_node_main.cpp:32]     @     0x7fa53f4e2980 (unknown)
I0425 23:09:25.330794 30820 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.332466 30819 stage_node.cpp:112] _dir_date_time:22-04-25_23-09-21
E0425 23:09:25.332513 31085 stage_node_main.cpp:32]     @     0x7fa53d14e909 Eigen::JacobiSVD<>::compute()
I0425 23:09:25.332547 30819 stage_node.cpp:122] pub _gnss_base:
I0425 23:09:25.334646 30821 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
E0425 23:09:25.337004 31085 stage_node_main.cpp:32]     @     0x7fa53d15778a openvslam::solve::fundamental_solver::compute_F_21()
I0425 23:09:25.338773 30820 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.340220 30822 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.342584 30819 stage_node.cpp:112] _dir_date_time:22-04-25_23-09-21
I0425 23:09:25.342881 30819 stage_node.cpp:122] pub _gnss_base:
E0425 23:09:25.347088 31085 stage_node_main.cpp:32]     @     0x7fa53d1585b3 openvslam::solve::fundamental_solver::find_via_ransac()
I0425 23:09:25.347350 30821 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.348809 30820 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.354588 30822 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.354601 30819 stage_node.cpp:112] _dir_date_time:22-04-25_23-09-21
I0425 23:09:25.354709 30819 stage_node.cpp:122] pub _gnss_base:
I0425 23:09:25.358893 30820 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.360095 30822 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.362471 30819 stage_node.cpp:112] _dir_date_time:22-04-25_23-09-21
I0425 23:09:25.362550 30819 stage_node.cpp:122] pub _gnss_base:
E0425 23:09:25.362911 31085 stage_node_main.cpp:32]     @     0x7fa53f2046df (unknown)
I0425 23:09:25.365231 30821 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.365290 30821 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
I0425 23:09:25.368727 30820 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
E0425 23:09:25.364534 31085 stage_node_main.cpp:32]     @     0x7fa53f4d76db start_thread
I0425 23:09:25.370132 30822 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21
E0425 23:09:25.370927 31085 stage_node_main.cpp:32]     @     0x7fa53ec5f61f clone
[ INFO] [1650899366.047243176]: Image #2, image Path: /media/kint/E06A82386A820B8A/dataSet/open_realm_edm_dataset/edm_big_overlap_50p/img_0002.jpg
[realm_pose_estimation-3] process has died [pid 30819, exit code -11, cmd /home/kint/kint/slam/OpenREALM_ws/devel/lib/realm_ros/realm_stage_node __name:=realm_pose_estimation __log:=/home/kint/.ros/log/ae35196a-c4a9-11ec-9e17-58961d965e0b/realm_pose_estimation-3.log].
log file: /home/kint/.ros/log/ae35196a-c4a9-11ec-9e17-58961d965e0b/realm_pose_estimation-3*.log
[ INFO] [1650899366.566138707]: Image #3, image Path: /media/kint/E06A82386A820B8A/dataSet/open_realm_edm_dataset/edm_big_overlap_50p/img_0003.jpg
[ INFO] [1650899367.554202664]: Image #4, image Path: /media/kint/E06A82386A820B8A/dataSet/open_realm_edm_dataset/edm_big_overlap_50p/img_0004.jpg
[ INFO] [1650899368.077298056]: Image #5, image Path: /media/kint/E06A82386A820B8A/dataSet/open_realm_edm_dataset/edm_big_overlap_50p/img_0005.jpg
[ INFO] [1650899368.781517269]: Image #6, im

kintzhao, 2022-04-25:
link [](https://github.com/laxnpander/OpenREALM_ROS1_Bridge/issues/7)

> roslaunch realm_ros alexa_noreco.launch I have the same error.
>
> I0425 23:09:25.294509 30821 stage_node.cpp:304] subOutputPath 1:22-04-25_23-09-21 I0425 23:09:25.298725 30820 stage_node.cpp:304] [...]

https://github.com/laxnpander/OpenREALM_ROS1_Bridge/issues/7