laxnpander/OpenREALM #51: ov2_slam config file missing

ksd777, 2021-04-12:
Hi!
Could you please commit ov2_slam config file as well?

laxnpander, 2021-04-13:
@ksd777: Hey, how come you need this config file? Ov2slam is currently not supported by OpenREALM. I have it running in my private build and added the Interface for a potential switch in the future. But right now it doesn’t work with the github branches.

ksd777, 2021-04-13:
@laxnpander: I just saw there is a support for Ov2Slam and I wanted to give it a try. I added modified config file based on https://github.com/ov2slam/ov2slam/blob/main/parameters_files/accurate/tartanair/tartanair_mono.yaml but run into opengv library crash:
```
***** FATAL SIGNAL RECEIVED ******* 
Received fatal signal: SIGSEGV(11)	PID: 476080

***** SIGNAL SIGSEGV(11)

*******	STACKDUMP *******
	stack dump [1]  /usr/local/lib/libg3log.so.2.1.2-123+0x156cb [0x7f8ee427d6cb]
	stack dump [2]  /usr/lib/x86_64-linux-gnu/libc.so.6+0x46210 [0x7f8ee2e8a210]
	stack dump [3]  /usr/lib/x86_64-linux-gnu/libc.so.6cfree+0x20 [0x7f8ee2ee1870]

	stack dump [4]  /usr/local/lib/libov2slam-1.0.2.so : opengv::relative_pose::fivept_nister(opengv::relative_pose::RelativeAdapterBase const&, opengv::Indices const&)+0x67c [0x7f8edede6eac]

	stack dump [5]  /usr/local/lib/libov2slam-1.0.2.so : opengv::relative_pose::fivept_nister(opengv::relative_pose::RelativeAdapterBase const&, std::vector<int, std::allocator<int> > const&)+0x41 [0x7f8edede7191]

	stack dump [6]  /usr/local/lib/libov2slam-1.0.2.so : opengv::sac_problems::relative_pose::CentralRelativePoseSacProblem::computeModelCoefficients(std::vector<int, std::allocator<int> > const&, Eigen::Matrix<double, 3, 4, 0, 3, 4>&) const+0x17c [0x7f8ededbb36c]

	stack dump [7]  /usr/local/lib/libov2slam-1.0.2.so : opengv::sac::Ransac<opengv::sac_problems::relative_pose::CentralRelativePoseSacProblem>::computeModel(int)+0x11e [0x7f8eded682ee]

	stack dump [8]  /usr/local/lib/libov2slam-1.0.2.so : MultiViewGeometry::opengv5ptEssentialMatrix(std::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > > const&, std::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > > const&, int, float, bool, bool, float, float, Eigen::Matrix<double, 3, 3, 0, 3, 3>&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&, std::vector<int, std::allocator<int> >&)+0x3ac [0x7f8eded659bc]

	stack dump [9]  /usr/local/lib/libov2slam-1.0.2.so : VisualFrontEnd::epipolar2d2dFiltering()+0xa38 [0x7f8eded1d748]

	stack dump [10]  /usr/local/lib/libov2slam-1.0.2.so : VisualFrontEnd::trackMono(cv::Mat&, double)+0x5ad [0x7f8eded1e4dd]

	stack dump [11]  /usr/local/lib/libov2slam-1.0.2.so : VisualFrontEnd::visualTracking(cv::Mat&, double)+0x86 [0x7f8eded1e9d6]

	stack dump [12]  /usr/local/lib/libov2slam-1.0.2.so : SlamManager::spin()+0x2d1 [0x7f8eded10b21]
```

So I wanted to make sure it is not a parametric issue, and would be great if I can compare the set params you are using in your experiments.

laxnpander, 2021-04-14:
@ksd777: Okay try this: https://github.com/laxnpander/ov2slam
You need to clone this repository, install thirdparty dependencies as you would with ov2slam (the build script). But when building ov2slam you install it afterwards to root (sudo make install). This package is modified by me so it can be installed as cmake package.

You can check if it was succesfully installed looking at /usr/local/include and /usr/local/lib for ov2slam files. Next thing you have to do is build OpenREALM with ov2slam support. For that go to https://github.com/laxnpander/OpenREALM/blob/465ecc8a68a55dd7fee956af01b56169ea6db231/modules/realm_vslam/realm_vslam_base/CMakeLists.txt#L45 and uncomment all lines for ov2slam and comment block all open vslam lines.
Ideally you should now be able to compile OpenREALM without an issue. It will print a message saying something like "Selected ov2slam for visual pose estimation". 

If this is all going well, you only have to compile the OpenREALM_ROS1_BRIDGE with ov2slam as well.  Go to: 
https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/a364858df04debf7e5eceb22c79caa45fd49a0a0/realm_ros/CMakeLists.txt#L67 and replace it with `find_package(ov2slam REQUIRED)`. Then build the ros bridge with catkin_make again.

If you made it this far, next step is to modify the launch files for example at https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/a364858df04debf7e5eceb22c79caa45fd49a0a0/realm_ros/launch/alexa_reco.launch#L29 
Replace open_vslam with ov2slam. It will then automatically try to detect a settings file for ov2slam in `realm_ros/profiles/YOUR_PROFILE/pose_estimation/method`. There you paste this in: 
https://www.file-upload.net/download-14551389/ov2slam_settings.yaml.html

Let me know if I forgort a step somewhere in between. Have fun!

laxnpander, 2021-04-18:
Closed for now. Let me know if there are more questions.

closed 2021-04-18 by laxnpander