laxnpander/OpenREALM #52: ov2_slam spin missing

ksd777, 2021-04-12:
Hi!

When I have to call https://github.com/laxnpander/OpenREALM/blob/465ecc8a68a55dd7fee956af01b56169ea6db231/modules/realm_vslam/realm_vslam_base/src/ov2_slam.cpp#L117

There is an error that no such function is available. 
When we are checking https://github.com/ov2slam/ov2slam/blob/main/src/ov2slam.cpp, there is only function run() which opens infinite new frame listening loop

laxnpander, 2021-04-13:
@ksd777 Yes, this is a custom version of ov2slam which required some modifications to be integrated into OpenREALM. I can provide you with my current version, but I have no time currently to bugfix / support it.

ksd777, 2021-04-13:
It would be a great start for me if you could provide your buggy dev version of ov2slam together with config files

laxnpander, 2021-04-16:
see https://github.com/laxnpander/OpenREALM/issues/51

closed 2021-04-16 by laxnpander