laxnpander/OpenREALM #55: links to target "FLANN::FLANN" but the target was  not found.

Mengke-Yuan, 2021-04-18:
Hi Laxnpander,

        I got error in building OpenREALM_ROS1_Bridge with Ubuntu16.04 and ROS Kinetic. The first 'catkin_make -DCMAKE_BUILD_TYPE=Release' gives :
[catkin_make_1st.txt](https://github.com/laxnpander/OpenREALM/files/6330329/catkin_make_1st.txt)
and the second 'catkin_make -DCMAKE_BUILD_TYPE=Release' gives:
[catkin_make_2nd.txt](https://github.com/laxnpander/OpenREALM/files/6330330/catkin_make_2nd.txt)
It seems relative to 'FLANN'. I will be appreciated if you can give some hints. Regards!

laxnpander, 2021-04-18:
@Mengke-Yuan: Hey, what OS are you using? Ubuntu 20.04? There are a couple of these issues, also Vtk is causing some headache on 20.04 when compiled with PCL. I will have a look into it.

Mengke-Yuan, 2021-04-19:
> 
> 
> @Mengke-Yuan: Hey, what OS are you using? Ubuntu 20.04? There are a couple of these issues, also Vtk is causing some headache on 20.04 when compiled with PCL. I will have a look into it.

I am using Ubuntu16.04 and Kinetic. Thanks!

laxnpander, 2021-04-19:
@Mengke-Yuan: A quick fix should be to add `find_package(FLANN REQUIRED)` in the CMakeLists somewhere around here: https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/a364858df04debf7e5eceb22c79caa45fd49a0a0/realm_ros/CMakeLists.txt#L68

Mengke-Yuan, 2021-04-20:
Adding `find_package(FLANN REQUIRED)` worked. But new errors appears as shown in:
[log.txt](https://github.com/laxnpander/OpenREALM/files/6339786/log.txt)
Representative errors are :
1. /usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options. 
But I see the cmake setting at https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/a364858df04debf7e5eceb22c79caa45fd49a0a0/CMakeLists.txt#L51
2. usr/local/include/OpenREALM/realm_core/cv_grid_map.h:35:11: error: expected nested-name-specifier before ‘Ptr’
     using Ptr = std::shared_ptr<CvGridMap>;
           ^
/usr/local/include/OpenREALM/realm_core/cv_grid_map.h:36:11: error: expected nested-name-specifier before ‘ConstPtr’
     using ConstPtr = std::shared_ptr<const CvGridMap>;
           ^
/usr/local/include/OpenREALM/realm_core/cv_grid_map.h:38:11: error: expected nested-name-specifier before ‘Overlap’
     using Overlap = std::pair<CvGridMap::Ptr, CvGridMap::Ptr>;
3.error: ‘cv::InterpolationFlags’ is not a class or namespace
     void add(const std::string &layer_name, const cv::Mat &layer_data, int interpolation = cv::InterpolationFlags::INTER_LINEAR);
This may relate to OpenCV version?

Are there any dependences missing in my configuration? Thanks!

laxnpander, 2021-04-20:
@Mengke-Yuan: Hmm, this is weird. First time I see this error! This is not on ARM, is it? Error 2) might be related to the c++11 standard missing. So I'd try to fix the first issue and then see how it goes. However, it is difficult to debug without having your same hardware. I assume your compiler is old or outdated. Also, can you check what cmake version you use (terminal: cmake --version)?

I also found this which you could try: https://stackoverflow.com/questions/10851247/how-do-i-activate-c-11-in-cmake

Mengke-Yuan, 2021-04-20:
@laxnpander : My cmake version is 3.16.6 with an Intel CPU.

laxnpander, 2021-04-20:
@Mengke-Yuan: Hmmm, can you try to set the c++ standard to 14 for the ROS bridge? The main library is using c++14 https://github.com/laxnpander/OpenREALM/blob/465ecc8a68a55dd7fee956af01b56169ea6db231/CMakeLists.txt#L118
I have not experienced an issue with it yet, but maybe there is with your compiler version.

Mengke-Yuan, 2021-04-20:
@laxnpander : The compiler outputs seem unchanged after using c++14.

laxnpander, 2021-04-20:
@Mengke-Yuan: Alright, there is not much I can do unfortunately. You will have to find out what causes this problem on your specific system. Alternatively you can try to use the docker instead. Not sure if this is suitable for your application?

Mengke-Yuan, 2021-04-21:
@laxnpander: Thanks for your help.

laxnpander, 2021-04-21:
@Mengke-Yuan Sure, you are welcome

closed 2021-04-21 by laxnpander

Ahmad-Drak, 2022-08-01:
@Mengke-Yuan did you manage to solve this after all?

laxnpander, 2023-05-18:
It definitely builds on Ubuntu 20.04. Under some configurations (installed libraries, library versions) errors like yours occur. Currently, I have no time to fix this for everyone. So I am afraid you are on your own. 

One solution might be to build it without PCL, which is the root of a lot of trouble. I think there is a build flag for that. I’ll let you know when I find the time to look at it:

howz-chen, 2023-05-19:
> It definitely builds on Ubuntu 20.04. Under some configurations (installed libraries, library versions) errors like yours occur. Currently, I have no time to fix this for everyone. So I am afraid [...]

Thanks for your quick reply, 

I change to use Ubuntu18.04 docker image based on arm64(NVIDIA Jetson AGX Orin), and follow the steps, and now can run test demo successfully.

KanonWY, 2023-09-25:
@howz-chen hi,  Is there a solution in Ubuntu20?  I also encountered this problem!

aurovoy, 2024-03-27:
this error may solved by locate a "findFLANN.cmake" file, copy the code to the CmakeList.txt