laxnpander/OpenREALM #63: Dockerfile problems

gsenkowski, 2021-07-19:
First the git submodule init and git submodule update commands for the OpenREALM repo are missing.
After adding these, the build process runs up to RUN set -ex && . /opt/ros/melodic/setup.sh && catkin_make -DCMAKE_BUILD_TYPE=Release

some Warnings and errors occur during the catkin_make
```
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.
```

and then 

```
[ 88%] Linking CXX shared library /catkin_ws/devel/lib/librealm_ros.so
/usr/bin/ld: cannot find -lvtkChartsCore
/usr/bin/ld: cannot find -lvtkCommonColor
/usr/bin/ld: cannot find -lvtkCommonDataModel
/usr/bin/ld: cannot find -lvtkCommonMath
/usr/bin/ld: cannot find -lvtkCommonCore
/usr/bin/ld: cannot find -lvtksys
/usr/bin/ld: cannot find -lvtkCommonMisc
/usr/bin/ld: cannot find -lvtkCommonSystem
/usr/bin/ld: cannot find -lvtkCommonTransforms
/usr/bin/ld: cannot find -lvtkInfovisCore
/usr/bin/ld: cannot find -lvtkFiltersExtraction
/usr/bin/ld: cannot find -lvtkCommonExecutionModel
/usr/bin/ld: cannot find -lvtkFiltersCore
/usr/bin/ld: cannot find -lvtkFiltersGeneral
/usr/bin/ld: cannot find -lvtkCommonComputationalGeometry
/usr/bin/ld: cannot find -lvtkFiltersStatistics
/usr/bin/ld: cannot find -lvtkImagingFourier
/usr/bin/ld: cannot find -lvtkImagingCore
/usr/bin/ld: cannot find -lvtkalglib
/usr/bin/ld: cannot find -lvtkRenderingContext2D
/usr/bin/ld: cannot find -lvtkRenderingCore
/usr/bin/ld: cannot find -lvtkFiltersGeometry
/usr/bin/ld: cannot find -lvtkFiltersSources
/usr/bin/ld: cannot find -lvtkRenderingFreeType
/usr/bin/ld: cannot find -lvtkftgl
/usr/bin/ld: cannot find -lvtkDICOMParser
/usr/bin/ld: cannot find -lvtkDomainsChemistry
/usr/bin/ld: cannot find -lvtkIOXML
/usr/bin/ld: cannot find -lvtkIOGeometry
/usr/bin/ld: cannot find -lvtkIOCore
/usr/bin/ld: cannot find -lvtkIOXMLParser
/usr/bin/ld: cannot find -lvtkFiltersAMR
/usr/bin/ld: cannot find -lvtkParallelCore
/usr/bin/ld: cannot find -lvtkIOLegacy
/usr/bin/ld: cannot find -lvtkFiltersFlowPaths
/usr/bin/ld: cannot find -lvtkFiltersGeneric
/usr/bin/ld: cannot find -lvtkFiltersHybrid
/usr/bin/ld: cannot find -lvtkImagingSources
/usr/bin/ld: cannot find -lvtkFiltersHyperTree
/usr/bin/ld: cannot find -lvtkFiltersImaging
/usr/bin/ld: cannot find -lvtkImagingGeneral
/usr/bin/ld: cannot find -lvtkFiltersModeling
[... 226 lines cut ...]
/usr/bin/ld: cannot find -lvtkTestingGenericBridge
/usr/bin/ld: cannot find -lvtkTestingIOSQL
/usr/bin/ld: cannot find -lvtkTestingRendering
/usr/bin/ld: cannot find -lvtkViewsContext2D
/usr/bin/ld: cannot find -lvtkViewsGeovis
/usr/bin/ld: cannot find -lvtkWrappingJava
collect2: error: ld returned 1 exit status
OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/build.make:463: recipe for target '/catkin_ws/devel/lib/librealm_ros.so' failed
make[2]: *** [/catkin_ws/devel/lib/librealm_ros.so] Error 1
CMakeFiles/Makefile2:2021: recipe for target 'OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/all' failed
make[1]: *** [OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 90%] Linking CXX executable /catkin_ws/devel/lib/realm_rviz_plugin/ground_image_test_node
/usr/bin/ld: warning: libopencv_imgproc.so.3.3, needed by /usr/local/lib/libopencv_imgcodecs.so.3.3.1, may conflict with libopencv_imgproc.so.3.2
/usr/bin/ld: warning: libopencv_core.so.3.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.3.2.0, may conflict with libopencv_core.so.3.3
[ 90%] Built target ground_image_test_node
make: *** [all] Error 2
Makefile:140: recipe for target 'all' failed
Invoking "make -j8 -l8" failed
```

Something seems to be wrong with vtk but i was unable to fix it (trying to install different versions of vtk).
The warning about OpenCV could point to additional issues with the Dockerfile.
for the full output see the attached file.
[terminal_output.txt](https://github.com/laxnpander/OpenREALM/files/6840024/terminal_output.txt)

laxnpander, 2021-07-24:
@gsenkowski: Heu, are you using Ubuntu 20.04? I am currently on vacation and can not look into it. But can you try to put a „find_package(vtk REQUIRED)“ in the realm_ros/CMakeLists.txt and rerun it?

laxnpander, 2021-08-02:
@gsenkowski Any updates on that?

p-fj, 2021-08-04:
Hi all, 
I had a similar issue. I am not sure yet where it does come from, however, it is probably connected to VTK and PCL. Installing VTK gets you a bit further in the process but also does not do the trick. 

```
cd ~ && mkdir vtk && cd vtk
git clone https://github.com/Kitware/VTK.git
cd VTK
mkdir build
cd build
cmake ..
make -j$(($(nproc) - 1))
sudo make install
```

@laxnpander Have you tried reproducing the problem?

laxnpander, 2021-08-04:
> @gsenkowski: Heu, are you using Ubuntu 20.04? I am currently on vacation and can not look into it. But can you try to put a „find_package(vtk REQUIRED)“ in the realm_ros/CMakeLists.txt and rerun it?

@p-fj Did you try this? I think the problem could be related to mismatching vtk versions for opencv and ROS.

closed 2021-11-02 by laxnpander