laxnpander/OpenREALM #64: GPS point location in image

ksd777, 2021-07-21:
I found out that by using test dataset with old airfield, the stitched image matches the map very well. When I am using my own dataset, the constant (40-50m) offset is observable (the same objects in google map and stitched image does not match together). 
When I was writing EXIF to my dataset, I was using drone central longitude and latitude, which matches with the center of the frame (in case of downward facing camera and flat ground surface). So the question is whether in your dataset longitude/latitude is in the center of frame or corner of the image is used (top-left, bottom-left, bottom-right, top-right)?

laxnpander, 2021-07-24:
@ksd777 Hey, the gps coordinates are always the coordinates of the drone. Where do you observe this offset? Do you see it in rviz with the aerial map display overlay? If yes, then this is some old bug I have to look into. It’s using some wrong transformation for the overlay. The stitched map should be correct. So when you open the Geotiffs in the results folder (realm_ros/output/.../mosaicing) in a tool like QGIS everything should be displayed correctly.

ksd777, 2021-07-26:
@laxnpander Yes, I did observe offset in rviz plugin. May I ask you what was the workaround for rviz plugin to make edm_big_overlap_50p appear there without an offset?

laxnpander, 2021-07-30:
@ksd777 Aaah, I have to look into it. This will not happen until next week. I think the transformation set in the rviz plugin is wrong. Maybe I have fixed it in the Dev build already. But then you also need to checkout the Dev build of the bridge aaand I am not sure if it compiles out of the box. But you may try this first. So basically checkout dev branch of OpenREALM and ROS bridge and try again. Another suggestion might be checking out and older build of the rviz plugin. They changed something and since then we have this offset. But you will have to try which version works.

ksd777, 2021-07-30:
@laxnpander I am already living with dev branches (both the OpenREALM and OpenREALM_ROS1_Bridge) but the same offset. Older rviz plugin did not try yet..

laxnpander, 2021-08-02:
@ksd777 Do you have some test data I can use to check on my local version?

ksd777, 2021-08-02:
@laxnpander May I share my test images with you privately?

laxnpander, 2021-08-02:
@ksd777 Sure! I guess one image would be sufficient to see what’s going on?! Laxnpander@gmail.com

laxnpander, 2021-08-04:
@ksd777 I am having a look at you dataset right now. I do observe an offset but not 40-50m. More like a few meters. I sent you the results. Is this the same that you observe? The remaining offset that you see in the image is hard to judge. Might be a bad georeference, might be bad SLAM behavious. An small deviation in the altitude might also explain what you see in my pictures. Then the scaling is off and images shrink or bloat.

laxnpander, 2021-08-04:
@ksd777 I just checked my rviz satellite version. I modified the following line:
https://github.com/nobleo/rviz_satellite/blob/40ec791d09b0d276c2dce9ec934695c7faf31f28/src/aerialmap_display.cpp#L650
to
``` tf_buffer_->lookupTransform(ref_fix_->header.frame_id, MAP_FRAME, ref_fix_->header.stamp, ros::Duration(0.1));```

So I switched the frames. Does this solve your issue?

ksd777, 2021-08-04:
> @ksd777 I just checked my rviz satellite version. I modified the following line:
> https://github.com/nobleo/rviz_satellite/blob/40ec791d09b0d276c2dce9ec934695c7faf31f28/src/aerialmap_display.cpp#L650 [...]

I just tried that line with swapped first two params, but seems no visible effect on result

ksd777, 2021-08-04:
@laxnpander 

> @ksd777 I am having a look at you dataset right now. I do observe an offset but not 40-50m. More like a few meters. I sent you the results. Is this the same that you observe? The remaining offset [...]

As i replied in email to you, my offset is way larger than yours. For me it seems not very much related to the SLAM (I am using ORBSLAM), because even at the beginning (the 2 path lines are being drawed - one from GPS, another from visual estimation), the mission starting point is being visualized with that 40-50m offset. Maybe I have error in some pipeline params? Actually I am running out of ideas where to look for the problem..

laxnpander, 2021-08-04:
@ksd777 I don’t think there should be a parameter influencing that. Just to make sure: you checked the resulting GTiff in e.g. QGIS? Would be very good to know if the results are misaligned or if only the visualisation is misaligned.

ksd777, 2021-08-04:
@laxnpander 
Just played with generated gtiff files in QGIS - they are aligning with the OSM map more or less OK (very similar to yours). So we can conclude the error is related to the rviz visualization..

laxnpander, 2021-08-05:
@ksd777 Alright, I am going to have a look at the changes I made to the rviz satellite map in more detail today. Please also try catkin clean and catkin_make again. Sometimes changes made are getting lost somewhere in the catkin build system.

laxnpander, 2021-08-05:
@ksd777 I just realized I am using an upstream to my own fork:
https://github.com/laxnpander/rviz_satellite
So there might be more changes somewhere in the code compared to the original master. I don't have the time to investigate now, but feel free to use my version of the satellite maps to check if this clears your issues as well.

ksd777, 2021-08-05:
@laxnpander 

Looks like replacing the whole rviz_sattelite directory did the trick. Now the offset is very similar as on your result. Would be interesting to compare the codes and find out where is the difference.. But currently the most important was to find the cause and solution :)

laxnpander, 2021-08-06:
@ksd777 Haha, exactly! Unfortunately there is not enough time to know everything :dancers: 

About the remaining offset: I think it could be more accurate. Not sure what's the issue here, maybe bad GPS signal? Not in sync with the image? Heading or altitude offsets?

closed 2021-08-06 by laxnpander