laxnpander/OpenREALM #67: realm_pose_estimation-3 process has died

yafenggithub, 2021-09-22:
I can run "roslaunch realm_ros alexa_gnss.launch " very well,but when run "roslaunch realm_ros alexa_noreco.launch "and "roslaunch realm_ros alexa_reco.launch ",I get the error "[realm_pose_estimation-3] process has died [pid 23631, exit code -9, cmd /home/ubuntu/catkin_ws/devel/lib/realm_ros/realm_stage_node __name:=realm_pose_estimation __log:=/home/ubuntu/.ros/log/51d3957e-1bae-11ec-afdd-3cf862d9f68e/realm_pose_estimation-3.log].
log file: /home/ubuntu/.ros/log/51d3957e-1bae-11ec-afdd-3cf862d9f68e/realm_pose_estimation-3*.log"   but I cannot find the log.Please help me. Thanks
I try it on Ubuntu18.04,try VMware  and WSL and PC,the error alway on,the all ouput message is:

... logging to /home/ubuntu/.ros/log/51d3957e-1bae-11ec-afdd-3cf862d9f68e/roslaunch-ubuntu-23606.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:43223/

SUMMARY
========

PARAMETERS
 * /realm_exiv2_grabber/config/id: ubuntu
 * /realm_exiv2_grabber/config/input: /media/ubuntu/DAT...
 * /realm_exiv2_grabber/config/profile: alexa_noreco
 * /realm_exiv2_grabber/config/rate: 10.0
 * /realm_mosaicing/config/id: ubuntu
 * /realm_mosaicing/config/profile: alexa_noreco
 * /realm_mosaicing/stage/type: mosaicing
 * /realm_mosaicing/topics/input/frame: /realm/ubuntu/ort...
 * /realm_ortho_rectification/config/id: ubuntu
 * /realm_ortho_rectification/config/profile: alexa_noreco
 * /realm_ortho_rectification/stage/type: ortho_rectification
 * /realm_ortho_rectification/topics/input/frame: /realm/ubuntu/sur...
 * /realm_ortho_rectification/topics/output: /realm/ubuntu/ort...
 * /realm_pose_estimation/config/id: ubuntu
 * /realm_pose_estimation/config/method: open_vslam
 * /realm_pose_estimation/config/profile: alexa_noreco
 * /realm_pose_estimation/stage/master: False
 * /realm_pose_estimation/stage/type: pose_estimation
 * /realm_pose_estimation/topics/input/frame: /realm/ubuntu/input
 * /realm_pose_estimation/topics/output: /realm/ubuntu/pos...
 * /realm_surface_generation/config/id: ubuntu
 * /realm_surface_generation/config/profile: alexa_noreco
 * /realm_surface_generation/stage/type: surface_generation
 * /realm_surface_generation/topics/input/frame: /realm/ubuntu/pos...
 * /realm_surface_generation/topics/output: /realm/ubuntu/sur...
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    realm_exiv2_grabber (realm_ros/realm_exiv2_grabber)
    realm_mosaicing (realm_ros/realm_stage_node)
    realm_ortho_rectification (realm_ros/realm_stage_node)
    realm_pose_estimation (realm_ros/realm_stage_node)
    realm_surface_generation (realm_ros/realm_stage_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [23616]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 51d3957e-1bae-11ec-afdd-3cf862d9f68e
process[rosout-1]: started with pid [23627]
started core service [/rosout]
process[realm_exiv2_grabber-2]: started with pid [23630]
process[realm_pose_estimation-3]: started with pid [23631]
process[realm_surface_generation-4]: started with pid [23632]
process[realm_ortho_rectification-5]: started with pid [23637]
process[realm_mosaicing-6]: started with pid [23638]
process[rviz-7]: started with pid [23639]
[ INFO] [1632319606.205668389]: STAGE_NODE [pose_estimation]: Loading stage settings from path:
	/home/ubuntu/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/pose_estimation/stage_settings.yaml
[ INFO] [1632319606.206835631]: STAGE_NODE [pose_estimation]: Detected stage type: 'pose_estimation'
[ INFO] [1632319606.211654002]: STAGE_NODE [pose_estimation]: : Loading camera from path:
	/home/ubuntu/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/camera/calib.yaml
[... 189 lines cut ...]
[ INFO] [1632319617.459302860]: Image #11, image Path: /media/ubuntu/DATA/4slam/OpenREALM/open_data/edm_big_overlap_50p/img_0011.jpg
[ INFO] [1632319617.657814070]: Image #12, image Path: /media/ubuntu/DATA/4slam/OpenREALM/open_data/edm_big_overlap_50p/img_0012.jpg
[ INFO] [1632319617.877983860]: Image #13, image Path: /media/ubuntu/DATA/4slam/OpenREALM/open_data/edm_big_overlap_50p/img_0013.jpg
[ INFO] [1632319618.073904561]: Image #14, image Path: /media/ubuntu/DATA/4slam/OpenREALM/open_data/edm_big_overlap_50p/img_0014.jpg
[ INFO] [1632319618.305228749]: Image #15, image Path: /media/ubuntu/DATA/4slam/OpenREALM/open_data/edm_big_overlap_50p/img_0015.jpg
[ INFO] [1632319618.515402430]: Image #16, image Path: /media/ubuntu/DATA/4slam/OpenREALM/open_data/edm_big_overlap_50p/img_0016.jpg
[ INFO] [1632319618.759004022]: Image #17, image Path: /media/ubuntu/DATA/4slam/OpenREALM/open_data/edm_big_overlap_50p/img_0017.jpg
[ INFO] [1632319618.955550030]: Image #18, image Path: /media/ubuntu/DATA/4slam/OpenREALM/open_data/edm_big_overlap_50p/img_0018.jpg
^C[rviz-7] killing on exit
[realm_mosaicing-6] killing on exit
[realm_ortho_rectification-5] killing on exit
[realm_surface_generation-4] killing on exit
[realm_exiv2_grabber-2] killing on exit
2021-09-22 14:06:59.369 (  13.137s) [        E42E2680] worker_thread_base.cpp:129      0| Thread 'Stage [surface_generation]' received finish request...
2021-09-22 14:06:59.372 (  13.115s) [        2CB75680] worker_thread_base.cpp:129      0| Thread 'Stage [mosaicing]' received finish request...
2021-09-22 14:06:59.372 (  13.115s) [        2CB75680]          mosaicing.cpp:219    ERR| No global map created, skipping saveAll()
2021-09-22 14:06:59.378 (  13.121s) [        CC338680] worker_thread_base.cpp:129      0| Thread 'Stage [ortho_rectification]' received finish request...
2021-09-22 14:06:59.441 (  13.183s) [Stage [ortho_rec] worker_thread_base.cpp:101      0| Thread 'Stage [ortho_rectification]' finished!
2021-09-22 14:06:59.441 (  13.210s) [Stage [surface_g] worker_thread_base.cpp:101      0| Thread 'Stage [surface_generation]' finished!
2021-09-22 14:06:59.441 (  13.184s) [Stage [mosaicing] worker_thread_base.cpp:101      0| Thread 'Stage [mosaicing]' finished!
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

yafenggithub, 2021-10-03:
The  orb vocabulary cause this error,I have solved this problem,thanks.I will colse it.

laxnpander, 2021-10-04:
@yafenggithub Hey, Sorry I didn’t respond quick enough. I was on vacation. Glad you could resolve your issues! I am aware of the problem and thought I patched it already. It is related to the ros bridge not so much repo. But thanks for, pointing it out!

zxr12748, 2021-10-08:
> The orb vocabulary cause this error,I have solved this problem,thanks.I will colse it.

exactly same problem here,
could you give more detail information about how you solve it? thx~

csm-coder, 2022-03-19:
@zxr12748 i have the same problem and have no idea about it. do you solve it? thanks.

laxnpander, 2022-03-19:
@zxr12748 Can you post the terminal output? Or run the pose estimation node in gdb and Post the stacktrace?
http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

laxnpander, 2022-03-24:
No updates, so closing until reopened.

closed 2022-03-24 by laxnpander