laxnpander/OpenREALM #69: OpenVSlam config file

roblab-wh-ge, 2022-02-18:
Hi,

how can the parameter for OpenVSlam be changed (fx,fy,cx,cy....). Could not find the config file.
Output for the example is:
Camera:
  model: perspective
  setup: monocular
  cx: 1150.4143848
  color_order: RGB
  fx: 1680
  fy: 1680
  cy: 646.6886903999999
  k1: -0.00209
  k2: 0.014828
  p1: 0.0006579999999999999
  p2: 0.000242
  k3: 0
 Thx

laxnpander, 2022-02-18:
@roblab-wh-ge Hey, config files are in the profiles folder of the ROS package e.g., https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/65a86d465b5ba0b5256f5c8f1bb8090ec3a027f8/realm_ros/profiles/alexa_reco/camera/calib.yaml#L6

laxnpander, 2022-02-20:
@roblab-wh-ge It defines on what resolution you want to run the vslam. Resizing of 0.5 means half width and height of the original image. The original camera calibration is then scaled appropriately. So in the calib.yaml you have to but the full resolution parameters and if vslam is too slow you only modify the resizing.

laxnpander, 2022-02-21:
@roblab-wh-ge Difficult to say without seeing anything. Can you post a log of the terminal? Do you see a map being built in the examples and your custom data? Custom data is definitely tricky, because you have to follow several rules. For example your setup must be able to properly run with visual SLAM algorithms. This typically requires frame rates of at least 10 FPS and high quality cameras as well as some parameter tuning. What is your setup?

Notes: Currently the publishing of 3D point clouds is disabled, I haven't maintained this functionality in a while. See here: 
https://github.com/laxnpander/OpenREALM/blob/e9dce8ba669b197f35da59398fad65ae421ade64/modules/realm_stages/src/mosaicing.cpp#L262
I simply didn't need it and quality of the point cloud is also rather bad in most cases. It is, however, internally still used to rectify the images for the ortho. Also you should get an elevation map if everything works will.