laxnpander/OpenREALM #70: How to get the mesh output

ffy747426112, 2022-04-07:
Hello, when I use the 2.5D example（alexa_reco.launch）, I can only see the plane stitching result of the orthographic image in RViz, but not the mesh result.
I found that in the Display column, the status of the **Marker** is displayed as an ERROR, I wonder if this means that the Mesh result is not calculated.
Finally, I found under the createMeshFaces() function of **mosaicing.cpp** that the return result was set to be empty, and mesh generation was not performed. Besides that, the mesh in the entire mosaic.cpp is not performing the generation, you commented it all out, why is that? What should I do to get the mesh result?
Thank you

laxnpander, 2022-04-10:
@ffy747426112 Hey, yes I rarely used it and didn’t have the time to maintain it. I think it is bugged atm, so not sure what happens if you uncomment it. The general logic should be fine. I can see what I can do there, but it will take some time.

ffy747426112, 2022-04-11:
@laxnpander Thank you for your reply, I did encounter some bugs after uncommenting, so I still can't get the mesh result. I will continue to try to fix them.

closed 2022-04-13 by ffy747426112

laxnpander, 2022-04-13:
@ffy747426112 Have you been successful?

ffy747426112, 2022-04-13:
@laxnpander After uncommenting, and then slightly correcting the bug of the mesh result output function, the mesh result can be obtained after exiting the program, but there are still some bugs that I am not very clear about.
The results of mesh cannot be visualized in real time, and the visualization part still reports an error, but this is not my current focus, so I haven't delved into it. In addition, the output of mesh results is also unstable, and the wrong .ply file is often obtained, which cannot be read normally, and the problem has not been found for the time being.
At present, there is no problem with the general logic of mesh generation. I will solve these unstable bugs slowly according to the needs of the work.
Finally, thanks again for your help.