laxnpander/OpenREALM #71: pose_estimation.cpp:291   WARN:NO tracking and RVIZ-GroundImage-Message:No image received.

dililida, 2022-04-17:
: hi, thank you very much for your open source spirit. Now I compiled the project, but an error was reported during runtime. Can you help me ？
pose_estimation.cpp:291   WARN:NO tracking and RVIZ-GroundImage-Message:No image received.
[image]
[image]
[image]

dililida, 2022-04-22:
> @dililida Hey, no tracking is not a problem per se. There are always some stretches that might not be trackable due to lack of good features. But it looks like the examples are not working properly for you. So it is never tracking. Hard to say what's going on. Which example did you start? alexa_noreco? alexa_reco? alexa_gnss? Do you see features being tracked in the image view at some point? Can you post a full terminal output?

Ok,I have been able to run alexa_gnss without problem.
[image]
But when I try to run alexa_reco and alexa_noreco ，I get the same error.

dililida, 2022-04-22:
Here is the full terminal  output.
[output.txt](https://github.com/laxnpander/OpenREALM/files/8541779/output.txt)

laxnpander, 2022-04-25:
@dililida Hmm, strange. There is a couple of directories that could not be created. You can fix this by commenting this line
https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/65a86d465b5ba0b5256f5c8f1bb8090ec3a027f8/realm_ros/launch/alexa_reco.launch#L71
in the launch files. However, this will not cause your observed behaviour. It looks a little bit like openvslam causes problems. What kind of system is this running on?

kintzhao, 2022-04-26:
> @dililida Hmm, strange. There is a couple of directories that could not be created. You can fix this by commenting this line [...]

I have the same errod, and I run it on docker

kintzhao, 2022-04-26:
2022-04-26 09:46:50.303 ( 980.102s) [        3B9CF700]         stage_base.cpp:103      0| } 0.000 s: Stage [mosaicing] statistics
^C[realm_mosaicing-6] killing on exit
[realm_ortho_rectification-5] killing on exit
[realm_surface_generation-4] killing on exit
[realm_exiv2_grabber-2] killing on exit
2022-04-26 09:46:54.166 ( 984.301s) [        AC523680] worker_thread_base.cpp:129      0| Thread 'Stage [surface_generation]' received finish request...
2022-04-26 09:46:54.240 ( 984.375s) [Stage [surface_g] worker_thread_base.cpp:101      0| Thread 'Stage [surface_generation]' finished!
2022-04-26 09:46:54.330 ( 984.129s) [        748C0680] worker_thread_base.cpp:129      0| Thread 'Stage [mosaicing]' received finish request...
2022-04-26 09:46:54.330 ( 984.129s) [        748C0680]          mosaicing.cpp:219    ERR| No global map created, skipping saveAll()
2022-04-26 09:46:54.350 ( 984.333s) [        4CEED680] worker_thread_base.cpp:129      0| Thread 'Stage [ortho_rectification]' received finish request...
2022-04-26 09:46:54.380 ( 984.362s) [Stage [ortho_rec] worker_thread_base.cpp:101      0| Thread 'Stage [ortho_rectification]' finished!
2022-04-26 09:46:54.427 ( 984.226s) [Stage [mosaicing] worker_thread_base.cpp:101      0| Thread 'Stage [mosaicing]' finished!

kintzhao, 2022-04-27:
> @dililida Hmm, strange. There is a couple of directories that could not be created. You can fix this by commenting this line [...]

@laxnpander  which commit id of openvslam  for this

laxnpander, 2022-04-27:
@kintzhao I use a fork in my github: https://github.com/laxnpander/openvslam

The crucial part is this:
```c
[ INFO] [1650634697.318298862]: STAGE_NODE [pose_estimation]: Received frame.
[2022-04-22 21:38:17.318] [info] initialization succeeded with H
[ INFO] [1650634697.389721218]: Image #103, image Path: /home/dila/open_realm_edm_dataset/edm_big_overlap_50p/img_0103.jpg
[ INFO] [1650634697.418823065]: STAGE_NODE [pose_estimation]: Received frame.
```

The pose estimation stage is getting new frames, but it does not seem to progress beyond the initialization of openvslam. Looks like it gets stuck somewhere.

laxnpander, 2022-04-28:
Closed due to https://github.com/laxnpander/OpenREALM/issues/72

closed 2022-04-28 by laxnpander

dililida, 2022-04-28:
> @dililida嗯，奇怪。有几个目录无法创建。您可以通过在启动文件中注释此行 https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/65a86d465b5ba0b5256f5c8f1bb8090ec3a027f8/realm_ros/launch/alexa_reco.launch#L71 来解决此问题。但是，这不会导致您观察到的行为。它看起来有点像openvslam导致问题。这是在什么样的系统上运行的？

Ubuntu18.04.

dililida, 2022-04-28:
> @dililida嗯，奇怪。有几个目录无法创建。您可以通过在启动文件中注释此行 https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/65a86d465b5ba0b5256f5c8f1bb8090ec3a027f8/realm_ros/launch/alexa_reco.launch#L71 来解决此问题。但是，这不会导致您观察到的行为。它看起来有点像openvslam导致问题。这是在什么样的系统上运行的？

 Hi! I run it on Ubuntu18.04.

dubing-bit, 2023-10-26:
> > @dililida嗯，奇怪。有几个目录无法创建。您可以通过在启动文件中注释此行https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/65a86d465b5ba0b5256f5c8f1bb8090ec3a027f8/realm_ros/launch/alexa_reco.launch#L71来解决此问题但是，这不会导致您观察到的行为。它看起来有点像 openvslam 导致问题。这是在什么样的系统上运行的？
> 
> 你好！我在Ubuntu18.04上运行它。

Hi, I am having the same problem, GNSS only mapping is working. But running alexa_noreco and alexa_reco doesn't accept the images.

hfwuyanzu666, 2023-12-07:
> Here is the full terminal output. [output.txt](https://github.com/laxnpander/OpenREALM/files/8541779/output.txt)

Hello, have you solved this problem? My problem is the same as yours. If you have solved it, please share your solution. Thank you very much