laxnpander/OpenREALM #72: why it track failed , and can not create map

kintzhao, 2022-04-26:
roslaunch realm_ros alexa_noreco.launch   . just track at first, then without  topic: /realm/alexa/pose_estimation/tracked
without CUDA . using the  gived dataset:https://drive.google.com/open?id=1-2h0tasI4wzxZKLBbOz3XbJ7f5xlxlMe

roslaunch realm_ros alexa_gnss.launch  can get the map 

[result.txt](https://github.com/laxnpander/OpenREALM/files/8565413/result.txt)
[image]
[image]
[image]

laxnpander, 2022-04-28:
@kintzhao It's the same as in the other issue you posted. It all comes down to this line:
```c
[2022-04-27 01:08:39.545] [info] initialization succeeded with F
```
and nothing after that. So openvslam seems to freeze after initialization, but I can't tell you why with the given information. Can you try to run it on a desktop computer with Ubuntu 18.04 or 20.04 installed?

laxnpander, 2022-05-16:
No updates in 20 days, so closing.

closed 2022-05-16 by laxnpander