laxnpander/OpenREALM #74: Error when run: roslaunch realm_ros alexa_noreco.launch

wuxiangchao, 2022-07-20:
While roslaunch realm_ros alexa_gnss.launch is working whitout any problem, but when i run roslaunch realm_ros alexa_noreco.launch with yaml exception:

[2022-07-20 16:33:55.521] [info] config file loaded: 
terminate called after throwing an instance of 'YAML::InvalidNode'
  what():  invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa

=========================== 

sirope@ubuntu20-04:~/SourceCode/OpenREALM/catkin_ws$ roslaunch realm_ros alexa_noreco.launch
... logging to /home/sirope/.ros/log/b0fa8658-0806-11ed-bba5-17ed01860011/roslaunch-ubuntu20-04-40087.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu20-04:35557/

SUMMARY
========

PARAMETERS
 * /realm_exiv2_grabber/config/id: alexa
 * /realm_exiv2_grabber/config/input: //home/sirope/Dat...
 * /realm_exiv2_grabber/config/profile: alexa_noreco
 * /realm_exiv2_grabber/config/rate: 10.0
 * /realm_mosaicing/config/id: alexa
 * /realm_mosaicing/config/profile: alexa_noreco
 * /realm_mosaicing/stage/type: mosaicing
 * /realm_mosaicing/topics/input/frame: /realm/alexa/orth...
 * /realm_ortho_rectification/config/id: alexa
 * /realm_ortho_rectification/config/profile: alexa_noreco
 * /realm_ortho_rectification/stage/type: ortho_rectification
 * /realm_ortho_rectification/topics/input/frame: /realm/alexa/surf...
 * /realm_ortho_rectification/topics/output: /realm/alexa/orth...
 * /realm_pose_estimation/config/id: alexa
 * /realm_pose_estimation/config/method: open_vslam
 * /realm_pose_estimation/config/profile: alexa_noreco
 * /realm_pose_estimation/stage/master: True
 * /realm_pose_estimation/stage/type: pose_estimation
 * /realm_pose_estimation/topics/input/frame: /realm/alexa/input
 * /realm_pose_estimation/topics/output: /realm/alexa/pose...
 * /realm_surface_generation/config/id: alexa
 * /realm_surface_generation/config/profile: alexa_noreco
 * /realm_surface_generation/stage/type: surface_generation
 * /realm_surface_generation/topics/input/frame: /realm/alexa/pose...
 * /realm_surface_generation/topics/output: /realm/alexa/surf...
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    realm_exiv2_grabber (realm_ros/realm_exiv2_grabber)
    realm_mosaicing (realm_ros/realm_stage_node)
    realm_ortho_rectification (realm_ros/realm_stage_node)
    realm_pose_estimation (realm_ros/realm_stage_node)
    realm_surface_generation (realm_ros/realm_stage_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [40110]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b0fa8658-0806-11ed-bba5-17ed01860011
process[rosout-1]: started with pid [40135]
started core service [/rosout]
process[realm_exiv2_grabber-2]: started with pid [40142]
process[realm_pose_estimation-3]: started with pid [40143]
process[realm_surface_generation-4]: started with pid [40144]
process[realm_ortho_rectification-5]: started with pid [40145]
process[realm_mosaicing-6]: started with pid [40146]
process[rviz-7]: started with pid [40147]
[... 151 lines cut ...]
[ INFO] [1658306038.845953287]: Image #33, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0033.jpg
[ INFO] [1658306038.946444969]: Image #34, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0034.jpg
[ INFO] [1658306039.046825003]: Image #35, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0035.jpg
[ INFO] [1658306039.147366247]: Image #36, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0036.jpg
[ INFO] [1658306039.247837382]: Image #37, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0037.jpg
[ INFO] [1658306039.348962415]: Image #38, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0038.jpg
[ INFO] [1658306039.449809985]: Image #39, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0039.jpg
[ INFO] [1658306039.550191706]: Image #40, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0040.jpg
[ INFO] [1658306039.650699972]: Image #41, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0041.jpg
[ INFO] [1658306039.751108932]: Image #42, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0042.jpg
[ INFO] [1658306039.851400824]: Image #43, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0043.jpg
[ INFO] [1658306039.951703346]: Image #44, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0044.jpg
[ INFO] [1658306040.052935162]: Image #45, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0045.jpg
[ INFO] [1658306040.153778543]: Image #46, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0046.jpg
^C[rviz-7] killing on exit
[realm_mosaicing-6] killing on exit
[ INFO] [1658306040.254277434]: Image #47, image Path: //home/sirope/Datasets/open_realm_edm_dataset/edm_big_overlap_50p/img_0047.jpg
[realm_ortho_rectification-5] killing on exit
[realm_surface_generation-4] killing on exit
[realm_exiv2_grabber-2] killing on exit
2022-07-20 16:34:00.402 (   4.890s) [         2087000] worker_thread_base.cpp:129      0| Thread 'Stage [surface_generation]' received finish request...
^C2022-07-20 16:34:00.441 (   4.930s) [Stage [surface_g] worker_thread_base.cpp:101      0| Thread 'Stage [surface_generation]' finished!
QFileSystemWatcher::removePaths: list is empty
QFileSystemWatcher::removePaths: list is empty
2022-07-20 16:34:00.531 (   5.005s) [        C41E2000] worker_thread_base.cpp:129      0| Thread 'Stage [mosaicing]' received finish request...

Ahmad-Drak, 2022-07-26:
This might be related to the grid_map? maybe this helps
https://github.com/ANYbotics/grid_map#iterators

wuxiangchao, 2022-07-26:
> This might be related to the grid_map? maybe this helps https://github.com/ANYbotics/grid_map#iterators

Thanks, let me try.

laxnpander, 2022-08-05:
@wuxiangchao Hey, did you find a solution to your problem? First of all the error indicates it is related to the yaml parser. Here I use opencv, so this is not related to OpenREALM directly. What happens very often is that ROS and OpenREALM link to different versions of opencv. Please make sure both use the same version. This can be tricky sometimes as cmake can be a pain in this regard. What works best is having only the ros opencv version. So remove all custom opencv installs from your computer.

laxnpander, 2022-08-10:
@wuxiangchao Reopen if problem persists.

closed 2022-08-10 by laxnpander