laxnpander/OpenREALM #75: build error, realted to fbow

Ahmad-Drak, 2022-07-22:
Dependencies installed just fine.
When running `cmake ..` inside OpenRealm/build i get the following error about fbow

```
-- Build type not specified, using Release
-- CUDA Support disabled.
-- Found OpenCV: /usr (found suitable version "4.2.0", minimum required is "4") 
CMake Warning at modules/realm_core/CMakeLists.txt:19 (message):
  OpenCV 4 Support is experimental, use at your own risk!

CMake Warning at modules/realm_io/CMakeLists.txt:20 (message):
  OpenCV 4 Support is experimental, use at your own risk!

-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem 
-- Eigen found (include: /usr/local/include/eigen3, version: 3.3.9)
-- Found Boost: /usr/include (found suitable version "1.71.0", minimum required is "1.55.0") found components: system filesystem date_time iostreams regex 
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- OpenNI2 found (include: /usr/include/openni2, lib: /usr/lib/libOpenNI2.so)
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- OpenNI2 found (include: /usr/include/openni2, lib: /usr/lib/libOpenNI2.so)
-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- looking for PCL_COMMON
-- looking for PCL_KDTREE
-- looking for PCL_OCTREE
-- looking for PCL_SEARCH
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_2D
-- looking for PCL_GEOMETRY
-- looking for PCL_IO
-- looking for PCL_FEATURES
-- looking for PCL_ML
-- looking for PCL_SEGMENTATION
-- looking for PCL_VISUALIZATION
-- looking for PCL_SURFACE
-- looking for PCL_REGISTRATION
-- looking for PCL_KEYPOINTS
-- looking for PCL_TRACKING
-- looking for PCL_RECOGNITION
-- looking for PCL_STEREO
-- looking for PCL_APPS
-- looking for PCL_IN_HAND_SCANNER
-- looking for PCL_POINT_CLOUD_EDITOR
-- looking for PCL_OUTOFCORE
-- looking for PCL_PEOPLE
-- Found OpenCV: /usr (found suitable version "4.2.0", minimum required is "4") found components: core imgproc imgcodecs highgui 
CMake Warning at modules/realm_densifier/realm_densifier_impl/psl/CMakeLists.txt:20 (message):
  OpenCV 4 Support is experimental, use at your own risk!

-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system filesystem 
-- CUDA not found. Skipping PSL package...

-- CUDA not found. Skipping PSL package...

-- Found OpenCV: /usr (found suitable version "4.2.0", minimum required is "4") 
CMake Warning at modules/realm_densifier/realm_densifier_base/CMakeLists.txt:28 (message):
  OpenCV 4 Support is experimental, use at your own risk!

CMake Warning at modules/realm_ortho/CMakeLists.txt:22 (message):
  OpenCV 4 Support is experimental, use at your own risk!

-- Using header-only CGAL
-- Targetting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.48")  
-- Boost include dirs: /usr/include
-- Boost libraries:    
-- Using gcc version 4 or later. Adding -frounding-math
CMake Warning at modules/realm_stages/CMakeLists.txt:20 (message):
  OpenCV 4 Support is experimental, use at your own risk!

CMake Warning at modules/realm_vslam/realm_vslam_base/CMakeLists.txt:19 (message):
  OpenCV 4 Support is experimental, use at your own risk!

-- Found OpenCV: /usr (found version "4.2.0") 
-- Found OpenMP_C: -fopenmp  
-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE   
-- Selected OpenVSLAM for visual pose estimation
-- With OpenREALM Modules: core;io;densifier;ortho;stages;vslam_base
-- Configuring done
CMake Error at modules/realm_stages/CMakeLists.txt:105 (add_library):
  Target "realm_stages" links to target "fbow::fbow" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at modules/realm_vslam/realm_vslam_base/CMakeLists.txt:113 (add_library):
  Target "realm_vslam_base" links to target "fbow::fbow" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at modules/realm_stages/CMakeLists.txt:105 (add_library):
  Target "realm_stages" links to target "fbow::fbow" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at modules/realm_vslam/realm_vslam_base/CMakeLists.txt:113 (add_library):
  Target "realm_vslam_base" links to target "fbow::fbow" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

```

It most likely has something to do with the CMakeLists.txt, but cant figure out how to fix it. Any ideas?

Ahmad-Drak, 2022-07-22:
fbow seems to be installed where it should be

```
/usr/local/bin/fbow_test_cpu
/usr/local/include/fbow
/usr/local/include/fbow/bow_feat_vector.h
/usr/local/include/fbow/bow_vector.h
/usr/local/include/fbow/cpu.h
/usr/local/include/fbow/fbow.h
/usr/local/include/fbow/fbow_exports.h
/usr/local/include/fbow/type.h
/usr/local/include/fbow/vocabulary.h
/usr/local/include/fbow/vocabulary_creator.h
/usr/local/lib/libfbow.so
/usr/local/lib/libfbow.so.0.0
/usr/local/lib/libfbow.so.0.0.1
/usr/local/share/cmake/fbow
/usr/local/share/cmake/fbow/fbowConfig.cmake
/usr/local/share/cmake/fbow/fbowTargets-release.cmake
/usr/local/share/cmake/fbow/fbowTargets.cmake
```

wuxiangchao, 2022-07-26:
try add following content to CMakeLists.txt file.

set( FBOW_INCLUDE_DIRS "/usr/local/include" )
set( FBOW_LIBS "/usr/local/lib/libfbow.so" )

target_link_libraries(${PROJECT_NAME}  ${FBOW_LIBS} )

Ahmad-Drak, 2022-07-26:
@wuxiangchao i included the following to the CMakelists.txt

```
set(FBOW_INCLUDE_DIRS "/usr/local/include")
set(FBOW_LIBS "/usr/local/lib/libfbow.so")
target_link_libraries(${PROJECT_NAME} ${FBOW_LIBS})
```
however I get the error

```
CMake Error at CMakeLists.txt:27 (target_link_libraries):
  Cannot specify link libraries for target "OpenREALM" which is not built by
  this project.
```
Did you manage to compile? how does your CMakelist.txt look like?

Ahmad-Drak, 2022-07-26:
The following makes it compile

```
set(FBOW_INCLUDE_DIRS "/usr/local/include")
set(FBOW_LIBS "/usr/local/lib/libfbow.so")
find_package(fbow REQUIRED)
```

laxnpander, 2022-08-05:
@Ahmad-Drak You found a solution that makes you happy? I am aware of this problem, but haven’t found the time to wrap my head around the cmake internals causing this issue for some people / versions.

Ahmad-Drak, 2022-08-08:
@laxnpander the above comment allows it to compile. There is an issue now when installing the ROS bridge. I dont know if it is related to this or something seperate.
This error shows up at `cakin_make`

```
usr/bin/ld: cannot find -lCGAL
/usr/bin/ld: cannot find -lFLANN::FLANN
collect2: error: ld returned 1 exit status
make[2]: *** [OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/build.make:448: /home/ahmad/workspace/catkin_ws/devel/lib/librealm_ros.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:2072: OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

laxnpander, 2022-08-08:
@Ahmad-Drak You can solve this by adding 
```c
find_package(CGAL REQUIRED)
find_package(FLANN REQUIRED)
```
around this line:
https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/65a86d465b5ba0b5256f5c8f1bb8090ec3a027f8/realm_ros/CMakeLists.txt#L77

It's an issue with bad target exports in the main library of OpenREALM, but I haven't found the root of the problem yet.

Ahmad-Drak, 2022-08-09:
@laxnpander Thanks for the followup.
I have add the lines as you suggested.  I only get the lFLANN as error now
```

/usr/bin/ld: cannot find -lFLANN::FLANN
collect2: error: ld returned 1 exit status
make[2]: *** [OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/build.make:451: /home/ahmad/workspace/catkin_ws/devel/lib/librealm_ros.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:8007: OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 96%] Linking CXX executable /home/ahmad/workspace/catkin_ws/devel/lib/realm_rviz_plugin/ground_image_test_node
[ 96%] Built target ground_image_test_node
make: *** [Makefile:141: all] Error 2
```

It might be an issue of where flann is installed, I think. libflann can be found in 2 locations:

```
usr/lib/x86_64-linux-gnu/libflann.so
/usr/lib/x86_64-linux-gnu/libflann.so.1.9
/usr/lib/x86_64-linux-gnu/libflann.so.1.9.1
/usr/lib/x86_64-linux-gnu/libflann_cpp.so
/usr/lib/x86_64-linux-gnu/libflann_cpp.so.1.9
/usr/lib/x86_64-linux-gnu/libflann_cpp.so.1.9.1
/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a
/usr/lib/x86_64-linux-gnu/libflann_s.a
/usr/local/lib/libflann.so
/usr/local/lib/libflann.so.1.9
/usr/local/lib/libflann.so.1.9.1
/usr/local/lib/libflann_cpp.so
/usr/local/lib/libflann_cpp.so.1.9
/usr/local/lib/libflann_cpp.so.1.9.1
/usr/local/lib/libflann_cpp_s.a
/usr/local/lib/libflann_cuda.so
/usr/local/lib/libflann_cuda.so.1.9
/usr/local/lib/libflann_cuda.so.1.9.1
/usr/local/lib/libflann_cuda_s.a
/usr/local/lib/libflann_s.a
```
I set the `CMAKE_PREFIX_PATH` to `/usr/local/lib` : 

```
set(CMAKE_PREFIX_PATH "/usr/local/lib")
find_package(FLANN REQUIRED)
find_package(CGAL REQUIRED)
```
but still ld complains.
The `LD_LIBRARY_PATH` variable is already set in bashrc (even tried just setting it locally)
```
$LD_LIBRARY_PATH
bash: /usr/local/cuda-11.7/lib64:/usr/local/lib/:/home/ahmad/workspace/catkin_ws/devel/lib:/home/ahmad/workspace/catkin_ws/devel_cb/lib:/opt/ros/noetic/lib: No such file or directory

```

Ahmad-Drak, 2022-08-09:
some interesting, and probably related warning I get
```
CMake Warning (dev) at grid_map_geo/CMakeLists.txt:34 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "/usr/lib/libgdal.so" of target "test_tif_loader"
  does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at OpenREALM_ROS1_Bridge/realm_ros/CMakeLists.txt:167 (add_executable):
  Cannot generate a safe runtime search path for target realm_stage_node
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libflann_cpp.so.1.9] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

CMake Warning at OpenREALM_ROS1_Bridge/realm_ros/CMakeLists.txt:163 (add_executable):
  Cannot generate a safe runtime search path for target realm_exiv2_grabber
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libflann_cpp.so.1.9] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

CMake Warning at OpenREALM_ROS1_Bridge/realm_ros/CMakeLists.txt:131 (add_library):
  Cannot generate a safe runtime search path for target realm_ros because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libflann_cpp.so.1.9] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

CMake Warning at OpenREALM_ROS1_Bridge/realm_ros/CMakeLists.txt:159 (add_executable):
  Cannot generate a safe runtime search path for target realm_ros_grabber
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libflann_cpp.so.1.9] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.
```

laxnpander, 2022-08-10:
@Ahmad-Drak Never seen this warnings, but looks like it is related to something on your system. Hard to tell what exactly goes wrong. Is it possible, that OpenREALM main library links to a different libflann than the ROS1 bridge?

wuxiangchao, 2022-08-10:
> some interesting, and probably related warning I get
>
> ```
> CMake Warning (dev) at grid_map_geo/CMakeLists.txt:34 (add_dependencies):
>   Policy CMP0046 is not set: Error on non-existent dependency in [...]

the solutions of myself is:
1. mkdir cmake_modules under realm_ros
2. cp /usr/lib/x86_64-linux-gnu/cmake/pcl/Modules to cmake_modules
3. add the following to CMakeLists.txt which under realm_ros:
   > find_package(FLANN REQUIRED)

Ahmad-Drak, 2022-08-11:
@wuxiangchao sadly that doesnt work either. I will try to do a new clean install on 20.04 and report back.

```
usr/bin/ld: cannot find -lFLANN::FLANN
collect2: error: ld returned 1 exit status
make[2]: *** [OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/build.make:451: /home/ahmad/workspace/catkin_ws/devel/lib/librealm_ros.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:8007: OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j32 -l32" failed
```

wuxiangchao, 2022-08-11:
> @wuxiangchao sadly that doesnt work either. I will try to do a new clean install on 20.04 and report back.
>
> ```
> usr/bin/ld: cannot find -lFLANN::FLANN
> collect2: error: ld returned 1 exit status [...]

maybe add:
```
TARGET_LINK_LIBRARIES(
    FLANN::FLANN
)

Ahmad-Drak, 2022-08-11:
@wuxiangchao same result sadly.

wuxiangchao, 2022-08-11:
> @wuxiangchao same result sadly.

the following is my CMakeLists.txt which under relam_ros dir:

```
cmake_minimum_required(VERSION 3.15)

project(realm_ros)

set(cv_bridge_DIR ~/catkin_ws/devel/share/cv_bridge/cmake)# add my cv_bridge
set(OpenCV_DIR "/usr/local/include/opencv4/opencv2")

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++17 -pthread")

list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_SOURCE_DIR}/cmake_modules)

################################################################################
# Include CMake dependencies
################################################################################

include(CheckLanguage)

################################################################################
# Compiler specific configuration
################################################################################

check_language(CUDA)
if(CMAKE_CUDA_COMPILER)
    enable_language(CUDA)
    message(STATUS "CUDA Support enabled.")

    include(FindCUDA)
    set(CUDA_ARCH_LIST Auto CACHE STRING  "List of CUDA architectures (e.g. Pascal, Volta, etc) or \
                                           compute capability versions (6.1, 7.0, etc) to generate code for. \
                                           Set to Auto for automatic detection (default).")
    cuda_select_nvcc_arch_flags(CUDA_ARCH_FLAGS ${CUDA_ARCH_LIST})
    list(APPEND CUDA_NVCC_FLAGS ${CUDA_ARCH_FLAGS})
else()
    message(STATUS "CUDA Support disabled.")
endif()

################################################################################
# Dependencies
################################################################################

find_package(catkin REQUIRED COMPONENTS
        roscpp
        roslib
        pcl_ros
        geodesy
        cv_bridge
        tf
        std_msgs
        nav_msgs
        sensor_msgs
        geographic_msgs
        realm_msgs
)

find_package(exiv2 REQUIRED)

#boost
find_package(Boost COMPONENTS thread filesystem date_time system REQUIRED)

set(THREADS_PREFER_PTHREAD_FLAG ON)
find_package (Threads REQUIRED)

# Fix to avoid OpenCV package confusion with ROS melodic
find_package(OpenCV 3.3.1 EXACT QUIET)
if (NOT OpenCV_FOUND)
    find_package(OpenCV 3 QUIET)
endif()
if (NOT OpenCV_FOUND)
    find_package(OpenCV 4 REQUIRED)
    message(WARNING "OpenCV 4 Support is experimental, use at your own risk!")
endif()

#find_package(Pangolin REQUIRED)

find_package(cmake_modules REQUIRED)

find_package(OpenREALM REQUIRED)
find_package(openvslam REQUIRED)
find_package(OpenMP REQUIRED)

find_package(FLANN REQUIRED)

#find_package(ORB_SLAM2 REQUIRED)
#find_package(ORB_SLAM3 REQUIRED)
#find_package(ov2slam REQUIRED)

find_package(VTK REQUIRED)
find_package(Eigen3 3.3 REQUIRED NO_MODULE)

################################################################################
# Catkin Package
################################################################################

catkin_package(
        INCLUDE_DIRS
            include
        LIBRARIES
            ${PROJECT_NAME}
        CATKIN_DEPENDS
            roscpp
            roslib
            pcl_ros
            geodesy
            cv_bridge
            tf
            std_msgs
            nav_msgs
            sensor_msgs
            geographic_msgs
            realm_msgs
            cmake_modules
        DEPENDS
            OpenCV
)

################################################################################
# Build
################################################################################

add_library(${PROJECT_NAME} SHARED
        src/realm_ros_lib/grabber_ros_node.cpp
        src/realm_ros_lib/grabber_exiv2_node.cpp
        src/realm_ros_lib/stage_node.cpp
        src/realm_ros_lib/conversions.cpp
)

target_include_directories(${PROJECT_NAME} PUBLIC
        include
        ${catkin_INCLUDE_DIRS}
)

target_link_libraries(${PROJECT_NAME} PUBLIC
        ${catkin_LIBRARIES}
        ${cmake_modules_LIBRARIES}
        ${OpenCV_LIBRARIES}
        ${Boost_LIBRARIES}
        exiv2lib
        Threads::Threads
	    realm_stages
)

add_dependencies(${PROJECT_NAME}
        realm_msgs_generate_messages_cpp
)

################################################################################
# Executables
################################################################################

add_executable(realm_ros_grabber src/grabber_ros_node_main.cpp)
add_dependencies(realm_ros_grabber ${catkin_EXPORTED_TARGETS})
target_link_libraries(realm_ros_grabber PUBLIC ${PROJECT_NAME})

add_executable(realm_exiv2_grabber src/grabber_exiv2_node_main.cpp)
add_dependencies(realm_exiv2_grabber ${catkin_EXPORTED_TARGETS})
target_link_libraries(realm_exiv2_grabber PUBLIC ${PROJECT_NAME})

add_executable(realm_stage_node src/stage_node_main.cpp)
add_dependencies(realm_stage_node ${catkin_EXPORTED_TARGETS})
target_link_libraries(realm_stage_node PUBLIC ${PROJECT_NAME})

################################################################################
# Install
################################################################################

# Mark executables and/or libraries for installation
install(
        TARGETS ${PROJECT_NAME}
        ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
        LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
        RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)

# Mark cpp header files for installation
install(
        DIRECTORY include/${PROJECT_NAME}/
        DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION}
        FILES_MATCHING
            PATTERN "*.h"
            PATTERN "*.hpp"
)

```

laxnpander, 2022-08-30:
Due to the inactivity I assume this is somewhat resolved? Reopen if problems persist.

closed 2022-08-30 by laxnpander

KunB-Fighting, 2023-03-06:
Hi,
`find_package(CGAL REQUIRED)`
works for the error `usr/bin/ld: cannot find -lCGAL`

and I add
`set(CMAKE_MODULE_PATH "/usr/lib/x86_64-linux-gnu/cmake")
find_package(FLANN REQUIRED)
add_library(FLANN::FLANN INTERFACE IMPORTED)`

then the errors about flann disappeared.

Ahmad-Drak, 2024-08-08:
For anyone looking into this in the future:

Thanks to @wuxiangchao and @KunB-Fighting comments on this issue. A combination of both answers worked for compiling the openREALM main library and the ROS bridge as well.
ubuntu 20.04 (theoretically can be applied to 22.04 as well)

One major thing to note is that when installing Pangolin for the dependency for OpenVSLAM, make sure to download and install version 0.6, not the latest version (0.9).

Ahmad-Drak, 2024-08-08:
> Hello friends, thank you for sharing. I encountered a conflict between opencv4.2 and version 3.3.1 when compiling with ros. I introduced cv_bridge to see you, but I also tried to modify it. The error is as follows
> `/Project/catkin_ws/devel/lib/librealm_ros.so: undefined reference to cv_bridge::toCvCopy(sensor_msgs::Image_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'`

I didnt use the cv_bridge, therefore I dont know the nature of the error, sorry.