laxnpander/OpenREALM #77: What's the mean of this code?

GY56723-lab, 2022-10-05:
I’m a novice of slam. when i read the this code, I can't realize what it means:

cv::Mat mask;
      if (to.type() == CV_32F || to.type() == CV_64F)
        mask = (to != to) & (from == from);
      else
        mask = (to == 0) & (from > 0);

This segment appear in this function:

void CvGridMap::mergeMatrices(const cv::Mat &from, cv::Mat &to, int flag_merge_handling)
{
  switch(flag_merge_handling)
  {
    case REALM_OVERWRITE_ALL:
      to = from;
      break;
    case REALM_OVERWRITE_ZERO:
      cv::Mat mask;
      if (to.type() == CV_32F || to.type() == CV_64F)
        mask = (to != to) & (from == from);
      else
        mask = (to == 0) & (from > 0);
      from.copyTo(to, mask);
      break;
  }
}

If you responce, i would appreciate it

laxnpander, 2022-10-09:
@GY56723-debug Hey, this is not SLAM related. This is just logical operators with one speciality: to != to checks if the matrix "to" is not equal to matrix "to". So it is basically a comparison with itself. The speciality here is: We check for NaNs. As per C++ standard definition comparing a NaN with itself gives false in the logical world. 

The whole purpose of this function is to merge two layers (matrices) in a grid map. 

- REALM_OVERWRITE_ALL says we simply overwrite the content in grid "to" with the content of grid "from".
- REALM_OVERWRITE_ZERO says we overwrite only zero values in grid "to" with non-zero values of grid "from".

A grid map is the result of the mapping process, one layer could be RGB for example. Another one contains the elevation etc.

laxnpander, 2022-10-10:
@GY56723-debug You are welcome! Reopen if questions persist.

closed 2022-10-10 by laxnpander