laxnpander/OpenREALM #79: alexa_noreco.launch and alexa_reco.launch

pyn1998, 2022-11-28:
hi @laxnpander ,really impressived to see your amazing work ,it helps me a lot!
Now i have run all gnss/reco/noreco launch file with your test dataset ,and they look great.When i use my own dataset ,it worked well in alexa_gnss.launch,however there are no ortho results showed in alexa_noreco.launch and alexa_reco.launch.Would you please help me check if I have done something wrong? Here are the terminal outputs:

... logging to /home/pyn/.ros/log/9580b374-6ebe-11ed-abd3-fc4482124945/roslaunch-pyn-XPS-8940-17788.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pyn-XPS-8940:42293/

SUMMARY
========

PARAMETERS
 * /realm_exiv2_grabber/config/id: alexa
 * /realm_exiv2_grabber/config/input: /home/pyn/nanji_p...
 * /realm_exiv2_grabber/config/profile: alexa_noreco
 * /realm_exiv2_grabber/config/rate: 10.0
 * /realm_mosaicing/config/id: alexa
 * /realm_mosaicing/config/profile: alexa_noreco
 * /realm_mosaicing/stage/type: mosaicing
 * /realm_mosaicing/topics/input/frame: /realm/alexa/orth...
 * /realm_ortho_rectification/config/id: alexa
 * /realm_ortho_rectification/config/profile: alexa_noreco
 * /realm_ortho_rectification/stage/type: ortho_rectification
 * /realm_ortho_rectification/topics/input/frame: /realm/alexa/surf...
 * /realm_ortho_rectification/topics/output: /realm/alexa/orth...
 * /realm_pose_estimation/config/id: alexa
 * /realm_pose_estimation/config/method: open_vslam
 * /realm_pose_estimation/config/profile: alexa_noreco
 * /realm_pose_estimation/stage/master: True
 * /realm_pose_estimation/stage/type: pose_estimation
 * /realm_pose_estimation/topics/input/frame: /realm/alexa/input
 * /realm_pose_estimation/topics/output: /realm/alexa/pose...
 * /realm_surface_generation/config/id: alexa
 * /realm_surface_generation/config/profile: alexa_norecof
 * /realm_surface_generation/stage/type: surface_generation
 * /realm_surface_generation/topics/input/frame: /realm/alexa/pose...
 * /realm_surface_generation/topics/output: /realm/alexa/surf...
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    realm_exiv2_grabber (realm_ros/realm_exiv2_grabber)
    realm_mosaicing (realm_ros/realm_stage_node)
    realm_ortho_rectification (realm_ros/realm_stage_node)
    realm_pose_estimation (realm_ros/realm_stage_node)
    realm_surface_generation (realm_ros/realm_stage_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [18038]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9580b374-6ebe-11ed-abd3-fc4482124945
process[rosout-1]: started with pid [18073]
started core service [/rosout]
process[realm_exiv2_grabber-2]: started with pid [18076]
process[realm_pose_estimation-3]: started with pid [18077]
process[realm_surface_generation-4]: started with pid [18078]
process[realm_ortho_rectification-5]: started with pid [18079]
process[realm_mosaicing-6]: started with pid [18080]
process[rviz-7]: started with pid [18084]
[ INFO] [1669600041.136448059]: STAGE_NODE [ortho_rectification]: Loading stage settings from path:
[ INFO] [1669600041.136447757]: STAGE_NODE [mosaicing]: Loading stage settings from path:
	/home/pyn/openREALM/OpenREALM/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/ortho_rectification/stage_settings.yaml
	/home/pyn/openREALM/OpenREALM/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/mosaicing/stage_settings.yaml
[ INFO] [1669600041.137005181]: STAGE_NODE [surface_generation]: Loading stage settings from path:
	/home/pyn/openREALM/OpenREALM/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/surface_generation/stage_settings.yaml
[... 510 lines cut ...]
2022-11-28 09:47:45.760 (  24.731s) [Stage [pose_esti]    pose_estimation.cpp:250      0| } 0.806 s: track
2022-11-28 09:47:45.760 (  24.731s) [Stage [pose_esti] worker_thread_base.cpp:84       0| Timing [Total]: 806 ms
2022-11-28 09:47:45.760 (  24.732s) [Stage [pose_esti]    pose_estimation.cpp:482      0| Input frame buffer at 20.00%
2022-11-28 09:47:45.760 (  24.732s) [Stage [pose_esti]    pose_estimation.cpp:250      0| { track
2022-11-28 09:47:45.760 (  24.732s) [Stage [pose_esti]    pose_estimation.cpp:251      0| .   Frame id: #23, timestamp: 1669600063572
[ INFO] [1669600065.937838835]: Image #26, image Path: /home/pyn/nanji_photo/openrealm/sys/img_0026.jpg
[ INFO] [1669600065.939590694]: STAGE_NODE [pose_estimation]: Received frame.
2022-11-28 09:47:45.948 (  24.920s) [Publisher [pose_]    pose_estimation.cpp:651      0| .   Publishing pose of frame #19...
2022-11-28 09:47:46.545 (  25.517s) [Stage [pose_esti]    pose_estimation.cpp:291   WARN| .   No tracking.
2022-11-28 09:47:46.546 (  25.518s) [Stage [pose_esti]    pose_estimation.cpp:250      0| } 0.786 s: track
2022-11-28 09:47:46.546 (  25.518s) [Stage [pose_esti] worker_thread_base.cpp:84       0| Timing [Total]: 786 ms
2022-11-28 09:47:46.546 (  25.518s) [Stage [pose_esti]    pose_estimation.cpp:482      0| Input frame buffer at 20.00%
2022-11-28 09:47:46.546 (  25.518s) [Stage [pose_esti]    pose_estimation.cpp:250      0| { track
2022-11-28 09:47:46.546 (  25.518s) [Stage [pose_esti]    pose_estimation.cpp:251      0| .   Frame id: #24, timestamp: 1669600064720
2022-11-28 09:47:46.955 (  25.927s) [Publisher [pose_]    pose_estimation.cpp:651      0| .   Publishing pose of frame #20...
^C[rviz-7] killing on exit
[realm_mosaicing-6] killing on exit
[realm_ortho_rectification-5] killing on exit
[realm_surface_generation-4] killing on exit
[realm_pose_estimation-3] killing on exit
[realm_exiv2_grabber-2] killing on exit
2022-11-28 09:47:47.288 (  26.260s) [Stage [pose_esti]    pose_estimation.cpp:291   WARN| .   No tracking.
20

thanks a lot!

pyn1998, 2022-11-28:
[image]

pyn1998, 2022-11-28:
And in the image window of RVIZ, I can see the photos ,but it seems that there are no feature points detected:
!(PICTURE:/home/pyn/.config/tencent-qq//AppData/file//sendpix1.jpg)

pyn1998, 2022-11-28:
And in the image window of RVIZ, I can see the photos ,but it seems that there are no feature points detected:
[image]

laxnpander, 2022-11-28:
@pyn1998 Hey, glad you like OpenREALM. It is very likely, that something in your setup is causing trouble. What frame rate does your camera / dataset have? How fast do you fly? And did you provide OpenREALM with the camera intrinsics? You must be very careful with all of that. Under the roof we use a visual SLAM (OpenVSLAM) which requires a certain image quality and high overlap between consecutive images (>90%). Also it takes some tuning of the pipeline parameters to get the results you want.

pyn1998, 2022-11-28:
> @pyn1998 Hey, glad you like OpenREALM. It is very likely, that something in your setup is causing trouble. What frame rate does your camera / dataset have? How fast do you fly? And did you provide [...]

Thank you for your professional and detailed reply! My dataset has 80%/60% overlap,appears that it's not enough for openvslam; and I have already  provide OpenREALM with my camera intrinsics.As for the fly speed ,I'm no sure about that at this moment,could you please tell me what's the proper range of fly speed?And does the flypath matter the ortho result?I have some other dataset which flew in a unarranged path ,it can not get a satisfied result,either.
By the way,by changing another dataset which has higher overlap,I got the ortho result showed in the RVIZ, so it must be the problem of my dataset.
Thank you for your reply and your work,again!

laxnpander, 2022-11-29:
@pyn1998 The flight path does not matter too much except for completeness of the result. Also when flying lawnmower pattern the overall integrity should be better due to better geometry of triangulation. But results should be shown in any case. 

There is no universal rule what works and what not unfortunately. It very much depends on the image quality and the reproducibility of the features. This is really hard to identify as human and also depends very much on the feature extractor / matcher. The more overlap you can provide, the better! At best, test some configurations (frame rate, altitude, flight speed) with your specific hardware and tune the pipeline to it. Then use this config for other flights.

laxnpander, 2022-12-12:
Reopen if questions / problems persist.

closed 2022-12-12 by laxnpander