laxnpander/OpenREALM #80: Question about Vslam

fulizi123, 2023-02-07:
I know Vslam need camera intrinsics to run. How does OPENrealm get camera parameters from the image stream ?

zhangjd1029, 2023-02-08:
> 
Hello, classmates, I am also running this open source project recently. Could you please add wechat for communication

fulizi123, 2023-02-08:
Sure, it's my pleasure. My wechat is 'c442022734'.

laxnpander, 2023-02-08:
@fulizi123 Hey all, assuming you are using the OpenREALM ROS1 Bridge (https://github.com/laxnpander/OpenREALM_ROS1_Bridge) there are several parameter files to adjust for custom datasets. You will find example profiles in this directory: https://github.com/laxnpander/OpenREALM_ROS1_Bridge/tree/master/realm_ros/profiles

A profile is a set of configurations for a specific camera + processing hardware. Camera calibration is handled here https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/master/realm_ros/profiles/alexa_reco/camera/calib.yaml 

Please note, that processing your own dataset comes with a learning curve. The default values might not work good for you depending on the resolution of your images, your frame rate, your flight altitude, scene, etc.

fulizi123, 2023-02-09:
@laxnpander OK，I know. Thank you very much.

closed 2023-02-09 by fulizi123