laxnpander/OpenREALM #82: Error running alexa_noreco.launch

tongtongliu0809, 2023-03-21:
Hello, thank you for always maintaining the OpenRealm framework.When running roslaunch realm_ros alexa_noreco.launch on dataset edm_big_overlap_50p using default profiles, I didn't see any ouput in the Rviz window.
[image]
[image]
Looking forward to your reply! Thank you very much!

laxnpander, 2023-03-22:
@tongtongliu0809 Hey, the missing rviz plugin is fine. You can clone it to your catkin_ws from here, but it is not mandatory: https://github.com/laxnpander/rviz_satellite 

The wrong path is a problem though. You should find the vocabulary here: https://github.com/laxnpander/OpenREALM_ROS1_Bridge/tree/master/realm_ros/profiles/alexa_noreco/pose_estimation/method/open_vslam 

Do you have any other info on what might go wrong?

laxnpander, 2023-04-05:
Reopen if issues persist.

closed 2023-04-05 by laxnpander

dubing-bit, 2023-10-25:
hello,thank you your work! i had a similar error.
[image] 
I had clone the https://github.com/laxnpander/rviz_satellite to the "catkin_ws" directory,but the problem still exists, I'm a newbie and can't solve it.