laxnpander/OpenREALM #83: OpenVSLAM integration issue - 'openvslam::system' has no member named 'get_tracker_state'

eki-Shalll, 2023-04-06:
Hello, @laxnpander! I am trying to integrate OpenVSLAM with OpenREALM. I have encountered an issue while compiling the OpenREALM project. The error message is as follows:
`
/home/OpenREALM/modules/realm_vslam/realm_vslam_base/src/open_vslam.cpp:89:55: error: 'using element_type = class openvslam::system {aka class openvslam::system}' has no member named 'get_tracker_state'
   openvslam::tracker_state_t tracker_state = m_vslam->get_tracker_state();
`
I have already included the tracking_module.h file from OpenVSLAM in the open_vslam.h file in the realm_vslam_base module. However, the openvslam::system class does not have a member named get_tracker_state.

Could you please guide how to properly access the tracker state or suggest an alternative method to resolve this issue? Thank you.
[image]

laxnpander, 2023-04-06:
@eki-Shalll Hey, did you pull my fork? 

https://github.com/laxnpander/openvslam

I think I added the getter for the tracker state myself.

eki-Shalll, 2023-04-07:
OH, I pulled forks from others for some reason. Now I solved that problem, and I appreciate your help. However, I have encountered a new issue:

I am experiencing an issue with the OpenREALM library when creating a ROS workspace for compilation at the last step:
`catkin_make -DCMAKE_BUILD_TYPE=Release`

The error message I encounter is as follows:
```
CMake Error at /home/OpenREALM/build/CMakeFiles/Export/897c5f51db83d44d675063351cd4d9df/OpenREALMConfig.cmake:126 (message):
  The imported target "realm_core" references the file
     "/home/OpenREALM/build/lib/libopen_realm_core-1.0.0.a"
  but this file does not exist.  Possible reasons include:
  * The file was deleted, renamed, or moved to another location.
  * An install or uninstall procedure did not complete successfully.
  * The installation package was faulty and contained
     "/home/OpenREALM/build/CMakeFiles/Export/897c5f51db83d44d675063351cd4d9df/OpenREALMConfig.cmake"
  but not all the files it references.
Call Stack (most recent call first):
  OpenREALM_ROS1_Bridge/realm_ros/CMakeLists.txt:68 (find_package)
```
After rebuilding OpenREALM several times, I have verified that the /home/OpenREALM/build/lib/ directory does not contain the libopen_realm_core-1.0.0.a file. Could you please provide some guidance on how to resolve this issue? Thank you for your time and assistance.

eki-Shalll, 2023-04-07:
The file path that I found is:
```
root@4a99eb0ee43b:/# find / -name "libopen_realm_core-1.0.0.a"
/home/OpenREALM/build/modules/realm_core/libopen_realm_core-1.0.0.a
/usr/local/lib/libopen_realm_core-1.0.0.a

```
This may seem like a silly question, but I may need to modify the OpenREALMConfig.cmake file. I would greatly appreciate it if you could help me think it through.

eki-Shalll, 2023-04-08:
I resolved this issue using symbolic links, the codes are as follows:
```
mkdir -p /home/OpenREALM/build/lib
cd /usr/local/lib/
for libfile in libopen_realm_*.a libopen_realm_*.so; do
    ln -s "$(pwd)/${libfile}" /home/OpenREALM/build/lib/"${libfile}"
done
```

```
mkdir -p /home/OpenREALM/build/include
ln -s /usr/local/include/OpenREALM /home/OpenREALM/build/include/OpenREALM
```
And the new problem is :
```
In file included from /home/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/include/realm_ros/conversions.h:37:0,
                 from /home/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/src/realm_ros_lib/conversions.cpp:21:
/usr/local/include/OpenREALM/realm_core/frame.h:15:10: fatal error: realm_core/enums.h: No such file or directory
 #include <realm_core/enums.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/build.make:117: recipe for target 'OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/src/realm_ros_lib/conversions.cpp.o' failed
make[2]: *** [OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/src/realm_ros_lib/conversions.cpp.o] Error 1
In file included from /home/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/include/realm_ros/conversions.h:37:0,
                 from /home/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/include/realm_ros/grabber_exiv2_node.h:33,
                 from /home/catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/src/realm_ros_lib/grabber_exiv2_node.cpp:21:
/usr/local/include/OpenREALM/realm_core/frame.h:15:10: fatal error: realm_core/enums.h: No such file or directory
 #include <realm_core/enums.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/build.make:89: recipe for target 'OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/src/realm_ros_lib/grabber_exiv2_node.cpp.o' failed
make[2]: *** [OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/src/realm_ros_lib/grabber_exiv2_node.cpp.o] Error 1
CMakeFiles/Makefile2:3524: recipe for target 'OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/all' failed
make[1]: *** [OpenREALM_ROS1_Bridge/realm_ros/CMakeFiles/realm_ros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 85%] Linking CXX shared library /home/catkin_ws/devel/lib/librealm_rviz_plugin.so
[ 85%] Built target realm_rviz_plugin
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j10 -l10" failed
```
which I solved by adding :
`message(STATUS "catkin_INCLUDE_DIRS: ${catkin_INCLUDE_DIRS}")`
after
`find_package(catkin REQUIRED COMPONENTS ...)`

Now it seemed that I have installed OpenREALM correctly as far as I am concerned.

laxnpander, 2023-04-08:
@eki-Shalll Looks like you severely messed up your build somehow. Did you use the install scripts or followed the instructions in the read me? Normally OpenREALM will be installed to root and is automatically found by cmake using the /usr/local directory.

eki-Shalll, 2023-04-08:
I mistakenly created the "catkin_ws" folder in the wrong location since I'm using an Ubuntu 18.04 virtual machine in Docker. It's a little bit confusing and brought me some unexpected trouble. Anyway, thank you for making me notice my problem. Now I have fixed it in the right way.

closed 2023-04-08 by eki-Shalll

rebecca0011, 2024-09-13:
> Hello, @laxnpander! I am trying to integrate OpenVSLAM with OpenREALM. I have encountered an issue while compiling the OpenREALM project. The error message is as follows: [...]

I'm currently encountering the same problem, how did you solve it?

laxnpander, 2024-09-13:
@rebecca0011 Probably by reading my previous comment:

Use my openvslam fork

https://github.com/laxnpander/openvslam