laxnpander/OpenREALM #85: What is the rationale behind the thread spawing in PoseEstimation?

geoeo, 2023-05-31:
Looking at the code in pose_estimation.cpp, it seems 
1) Threads are spawned a lot regarding the GeoReferncer.
Why was this decision made? What is the benefit?

2) Furthermore, why is PoseEstimation the only class having its IO done in a separate thread?

Thanks!

laxnpander, 2023-06-01:
Hey,

this is code I wrote during my master thesis 5 years ago. I do think there is a lot of sketchy stuff in there I simply have no time to fix or rewrite. But anyway, to your questions:
1) Not sure how long the georeferencing takes, but I guess I didn't want to block the thread, therefore the spawning and detaching. Pose estimation is the most crucial part of the whole pipeline (good pose = good map), which is why there is a lot of optimisation I thought was necessary.
2) I guess same as 1), pose estimation is crucial. The other stages fall into place after that. I didn't really care if the map was slightly delayed. But if pose estimation is blocked for a few milliseconds, the visual SLAM has it tougher.

Cheers

geoeo, 2023-06-02:
Thanks for your quick reply. 

The reason im asking is that Im seeing the map generation being non-deterministic (at least with open v-slam) with no dropped frames, as far as logging is concerned. So im wondering if its the locking behavior in the pose estimation or if there are some random sampling procedures (e.g. RANSAC) going on in the vslam library

laxnpander, 2023-06-02:
@geoeo Jeah, pretty sure the latter is the case. OpenVSLAM has at least 3-4 threads running simultaneously. There is no way two executions are identical.

geoeo, 2023-06-04:
Thanks for the insight!. Will close this

closed 2023-06-04 by geoeo