laxnpander/OpenREALM #86: Run alex_reco.lanuch, can't get ply file

peakpang, 2023-06-01:
**Hi, thank you very much for open sourcing this project. Run alex_reco.lanuch, I have refer to issue 81, set save_dense_ply option to 1. But in /mosaicing/elevation/ply, I also can't find the ply file. I suspect that it is caused by the following error, if so, what solution does the author have? Thank you so much!**

`[ERROR] [1685621030.485793789]: PluginlibFactory: The plugin for class 'rviz_plugins/AerialMapDisplay' failed to load.  Error: According to the loaded plugin descriptions the class rviz_plugins/AerialMapDisplay with base class type rviz::Display does not exist. Declared types are  realm_rviz_plugin/GroundImage rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped`

**If it is caused by other reasons, you can refer to the following complete output：**

`SUMMARY
========

PARAMETERS
 * /realm_densification/config/id: alexa
 * /realm_densification/config/method: psl
 * /realm_densification/config/profile: alexa_reco
 * /realm_densification/stage/type: densification
 * /realm_densification/topics/input/frame: /realm/alexa/pose...
 * /realm_densification/topics/output: /realm/alexa/dens...
 * /realm_exiv2_grabber/config/id: alexa
 * /realm_exiv2_grabber/config/input: /home/nwpu/DataSe...
 * /realm_exiv2_grabber/config/profile: alexa_reco
 * /realm_exiv2_grabber/config/rate: 10.0
 * /realm_mosaicing/config/id: alexa
 * /realm_mosaicing/config/opt/output_directory: /home/nwpu/Projec...
 * /realm_mosaicing/config/profile: alexa_reco
 * /realm_mosaicing/stage/type: mosaicing
 * /realm_mosaicing/topics/input/frame: /realm/alexa/orth...
 * /realm_ortho_rectification/config/id: alexa
 * /realm_ortho_rectification/config/profile: alexa_reco
 * /realm_ortho_rectification/stage/type: ortho_rectification
 * /realm_ortho_rectification/topics/input/frame: /realm/alexa/surf...
 * /realm_ortho_rectification/topics/output: /realm/alexa/orth...
 * /realm_pose_estimation/config/id: alexa
 * /realm_pose_estimation/config/method: open_vslam
 * /realm_pose_estimation/config/profile: alexa_reco
 * /realm_pose_estimation/stage/master: True
 * /realm_pose_estimation/stage/type: pose_estimation
 * /realm_pose_estimation/topics/input/frame: /realm/alexa/input
 * /realm_pose_estimation/topics/output: /realm/alexa/pose...
 * /realm_surface_generation/config/id: alexa
 * /realm_surface_generation/config/profile: alexa_reco
 * /realm_surface_generation/stage/type: surface_generation
 * /realm_surface_generation/topics/input/frame: /realm/alexa/dens...
 * /realm_surface_generation/topics/output: /realm/alexa/surf...
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    realm_densification (realm_ros/realm_stage_node)
    realm_exiv2_grabber (realm_ros/realm_exiv2_grabber)
    realm_mosaicing (realm_ros/realm_stage_node)
    realm_ortho_rectification (realm_ros/realm_stage_node)
    realm_pose_estimation (realm_ros/realm_stage_node)
    realm_surface_generation (realm_ros/realm_stage_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [213155]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5d22e3a6-0074-11ee-a3a2-6b780e5a7741
process[rosout-1]: started with pid [213291]
started core service [/rosout]
process[realm_exiv2_grabber-2]: started with pid [213298]
process[realm_pose_estimation-3]: started with pid [213299]
process[realm_densification-4]: started with pid [213300]
process[realm_surface_generation-5]: started with pid [213301]
process[realm_ortho_rectification-6]: started with pid [213302]
process[realm_mosaicing-7]: started with pid [213303]
process[rviz-8]: started with pid [213316]
[ INFO] [1685621028.738182518]: STAGE_NODE [pose_estimation]: Loading stage settings from path:
[... 358 lines cut ...]
[ INFO] [1685621030.501536959]: Image #15, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0015.jpg
[ INFO] [1685621030.565497588]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621030.601790028]: Image #16, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0016.jpg
[ INFO] [1685621030.666816597]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621030.702237761]: Image #17, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0017.jpg
[ INFO] [1685621030.768643440]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621030.802509257]: Image #18, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0018.jpg
[ INFO] [1685621030.866828194]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621030.902764955]: Image #19, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0019.jpg
[ INFO] [1685621030.974340953]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621031.003088911]: Image #20, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0020.jpg
[ INFO] [1685621031.070429181]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621031.103286803]: Image #21, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0021.jpg
[ INFO] [1685621031.162288560]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621031.203551050]: Image #22, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0022.jpg
[ INFO] [1685621031.268727395]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621031.303812969]: Image #23, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0023.jpg
[ INFO] [1685621031.362252284]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621031.404070227]: Image #24, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0024.jpg
[ INFO] [1685621031.464091589]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621031.504318641]: Image #25, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0025.jpg
[ INFO] [1685621031.569974450]: STAGE_NODE [pose_estimation]: Received frame.
[ INFO] [1685621031.604570189]: Image #26, image Path: /home/nwpu/DataSet/02_OpenRealm/open_realm_edm_dataset/edm_big_overlap_50p/img_0026.jpg
[2023-06-01 20:03:51.637] [info] new map created with 59 points: frame 0 - frame 4
`

peakpang, 2023-06-01:
I found that saveElevationPointsToPLY() in pcl_export.cpp calls saveElevationPoints(), and then assert(map.exists(mask_layer_name)) in saveElevationPoints(); 
so the above output appears: No layer with name 'valid' available. 
Is it possible that this is the reason, then how to solve it?

peakpang, 2023-06-01:
Add two pictures：
[image]

[image]

laxnpander, 2023-06-01:
@peakpang Hmm, do you see results in rviz, e.g. the elevation map as image published? The rviz aerialmap plugin has nothing to do with it.

peakpang, 2023-06-02:
**Thank you so much for your reply！
The following is the picture of rviz：**
[image]

**I seem to have found out where it might be wrong. First, 1->2->3 is the function call. Then after step 3, the terminal output (shown in the next few pictures):**
```
terminate called after throwing an instance of 'std::out_of_range'
what(): No layer with name 'valid' available.
```

**As a result, the following 'Fill in the cloud data' did not continue to execute. Is there any way to solve it? Thank you!**

[image]
[image]
[image]
[image]

laxnpander, 2023-06-02:
@peakpang Jeah, the valid layer seems to be missing. Maybe I have removed it at some point, but not fixed the PLY export. I think you could try to replace 

```c
cv::Mat mask = map[mask_layer_name];
```

with 

```c
cv::Mat mask = (elevation == elevation);
```

This should check elevation for NaN values and mask those out.

laxnpander, 2023-06-02:
@peakpang Perfect! I'd merge a pull request if you create one.

peakpang, 2023-06-02:
@laxnpander ok. Thanks again!

laxnpander, 2023-06-02:
@peakpang Thank you! :)

closed 2023-06-02 by laxnpander