laxnpander/OpenREALM #89: An issue in reco.launch

DJ1756, 2023-06-23:
Hello, thank you for your nice work.
We use the cmd 'roslaunch realm_ros alexa_reco.launch' to run 2.5D mapping with visual SLAM and surface reconstruction (with the data you provided). But we can’t see 2.5D effect in the result (It looks like an orthographic image with bad resolution).
We want get the 2.5D effect like you presented in this video (where the 3D model of the house on the ground can be observed, even it's not good):
'https://www.youtube.com/watch?v=9MvPTHP0r0c'. 
How to do that? We hope for some help, thank you very much.

DJ1756, 2023-06-23:
We found this issue seems solved in #86, we will try that first, thank you again.

laxnpander, 2023-06-25:
@DJ1756 Depends, what do you want to see? The linked issue should only concern the .ply export. You are correct, that the point cloud visualisation in Rviz is disabled because it is bugged. But I have no time to fix it. The 2.5D should be used to correct visual distortion regardless. You can check whether or not it is working by displaying the elevation_map that is being published.