laxnpander/OpenREALM #91: How to debug OpenREALM by launching *.launch file

howz-chen, 2023-07-06:
Hi, laxnpander , I am new to C++, and I install the environment and run the demo successfully before. And now I want to add some function into it , but I am confused about how to debug info OpenREALM code, such as ~/OpenREALM/modules/realm_stages/src/pose_estimation.cpp， and now I can only launch alexa_gnss.launch with vscode and debug code which about ROS node. So, I want to ask you that when you develop this great project, how do you debug it? And I hope you can help me with more details, such as 
1. How to start the whole process and debug into source code(such as *.cpp mentioned above)
2. "OpenREALM main library", and if I want to do something different(such add functions), what need I do?

Looking forward to your reply, thanks in advance. @laxnpander

Kt-Janice, 2023-07-21:
Hi！Did you solve it? I wanted to call OpenREALM to do something else, but I couldn't just replace the dataset with my own photos.

howz-chen, 2023-07-21:
> Hi！Did you solve it? I wanted to call OpenREALM to do something else, but I couldn't just replace the dataset with my own photos.

Hi, I try to add some other function to whole project successful. Step as below:
1. Try to debug OpenREALM_ROS1_Bridge, and find corresponding function(maybe  exist in OpenREALM project), and familiar with flowchart.
2. Edit the source code (*.cpp in OpenREALM). This process is a bit tedious and requires patience to find the calling relationship and process flow. You can add some log info to help debug code. Especially attention the declaration of a function or variable.
3. When you finish your code, then compile OpenREALM first, and then compile OpenREALM_ROS1_Bridge(just as steps of installation in [this position](https://github.com/laxnpander/OpenREALM#:~:text=Build%20and%20install%20the%20main%20library) and [this position](https://github.com/laxnpander/OpenREALM#:~:text=%23%20Make%20sure%20you%20are%20in%20catkin_ws%2C%20not%20src.%20Then%20build%20it)), and you can launch the *.launch file to test your new function.

Finally, if you just want to replace your own dataset, just change the path [as the author said](https://github.com/laxnpander/OpenREALM#:~:text=PUT%20THE%20TEST%20DATASET%27S%20ABSOLUTE%20PATH%20HERE)

closed 2023-09-07 by laxnpander