laxnpander/OpenREALM #92: Failed to change visual SLAM method

lebronzzw, 2023-08-22:
I would like to ask about the use of ORBSLAM2 instead of Openvslam. I have successfully compiled and installed ORB_SLAM2 and modified the launch file, but still failed.

laxnpander, 2023-09-01:
@lebronzzw: Hey, you have to modify the following line in the CMakeLists.txt:
https://github.com/laxnpander/OpenREALM/blob/91e1e712ffc3d4c8b8b02b3a951d98a8d7817941/modules/realm_vslam/realm_vslam_base/CMakeLists.txt#L36

It's a bit of a hassle to dynamically link OpenVSLAM and ORBSLAM2 at runtime, because both use the same libraries in different versions. So better uncomment the one you want to use and comment the other.

lebronzzw, 2023-09-04:
Thank you for you reply and I have changed this CMakeList.txt. But there are another question like this：
/Home/zzw/OpenREALM/modules/realm_ Vslam/realm_ Vslam_ Base/include/realm_ Vslam_ Base/orb_ SLAM. h:17:10: fatal error: ORB_ SLAM2/System. h: No such file or directory
#Include<ORB_ SLAM2/System. h>
^~~~~~~~~~~~~~~~~~~~
Compilation terminated
So I think I should make some modifications to ORBSLAM2. But I don't know how to do it.

laxnpander, 2023-09-04:
Pull my fork: https://github.com/laxnpander/ORB_SLAM2
The important part is that it needs to be installable, otherwise cmake is not able to find orb slam 2 as a dependency. So pull the fork, make and make install it to your system.

lebronzzw, 2023-09-04:
Thank you for your reply and I have successfully compiled and installed ORBSLAM2 and openrealm. When I finish compiling openrealm_ Ws and modify the launch file to 
<param name="config/method" type="string" value="ORB_SLAM2"/>
 But it cannot run successfully with error like this:
[realm_pose_estimation-3] process has died [pid 115460, exit code -6, cmd /home/zzw/openrealm_ws/devel/lib/realm_ros/realm_stage_node __name:=realm_pose_estimation __log:=/home/zzw/.ros/log/c383aef8-4afe-11ee-a025-e17a78fb7451/realm_pose_estimation-3.log].
log file: /home/zzw/.ros/log/c383aef8-4afe-11ee-a025-e17a78fb7451/realm_pose_estimation-3*.log
How can I solve this problem？

laxnpander, 2023-09-04:
Hmm, the error is saying nothing really. Compilation with ORB SLAM2 enabled was successful yes? Did you install openrealm after this compilation with `make install` to your system as well? Otherwise the ros bridge will still use the outdated libs with openvslam.

lebronzzw, 2023-09-05:
I have read the output information of the terminal. When I use ORB_ When SLAM2 is used as a method, the terminal will display:
What (): Error: SLAM framework 'ORB_ SLAM2 'not found
I have recompiled and installed ORB_ SLAM2 and Openrealm, both running `make install`.

lebronzzw, 2023-09-07:
Thank you or your help and I have successfully switched the SLAM method to ORB_SLAM2. But it is difficult to complete the initialization (even if I modify orb_slam2_settings.yaml or ORB_SLAM2 source code). After running for a period of time, it will still appear `[realm_pose_estimation-3] process has died` .

laxnpander, 2023-09-07:
Hmm, hard to say. The respective error is thrown here:
https://github.com/laxnpander/OpenREALM/blob/91e1e712ffc3d4c8b8b02b3a951d98a8d7817941/modules/realm_vslam/realm_vslam_base/src/visual_slam_factory.cpp#L36

This is triggered when your settings file does not specify orb slam 2 as framework. This in turn comes from here:
https://github.com/laxnpander/OpenREALM_ROS1_Bridge/tree/master/realm_ros/profiles/alexa_noreco/pose_estimation/method
In this directory you need a settings file called "orb_slam2_settings.yaml". The content you can basically copy paste from the orbslam3 file. 

Once this is done you have to change this line to "orb_slam2":
https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/65a86d465b5ba0b5256f5c8f1bb8090ec3a027f8/realm_ros/launch/alexa_noreco.launch#L29

I know it's a bit complicated. Nowadays I guess I'd do it differently. But changing the whole vslam framework also wasn't intended to be used by others exhaustively.

laxnpander, 2023-10-13:
Reopen if issues persist.

closed 2023-10-13 by laxnpander

dubing-bit, 2023-10-24:
> 拉我的叉子： https: [//github.com/laxnpander/ORB_SLAM2](https://github.com/laxnpander/ORB_SLAM2) 重要的是它需要可安装，否则 cmake 无法找到 orb slam 2 作为依赖项。因此，拉出分支，制作并将其安装到您的系统中。
I have the same problem with ORB-SLAM3 but orb-slam doesn't seem to support make install , how can I fix it?

YznMur, 2023-11-10:
Hi @lebronzzw 
Did u face such a problem https://github.com/laxnpander/OpenREALM/issues/94#issue-1977337642 when compiling installable version of orb-slam2?

lebronzzw, 2023-11-10:
> Hi @lebronzzw @laxnpander @dubing-bit Did u face such a problem [#94 (comment)](https://github.com/laxnpander/OpenREALM/issues/94#issue-1977337642) when compiling installable version of orb-slam2?
I haven't encountered such a problem yet. I think this is indeed caused by g2o, you can try to solve this problem by installing g2o from OpenREALM/tools/install_deps.sh and then modifying CMakeList.txt. Or directly recompile g2o under ORBSLAM2/Thirdparty.

dubing-bit, 2023-11-15:
> Hi @lebronzzw @laxnpander @dubing-bit Did u face such a problem [#94 (comment)](https://github.com/laxnpander/OpenREALM/issues/94#issue-1977337642) when compiling installable version of orb-slam2?
> I [...]

I used orb-slam3 (https://github.com/laxnpander/ORB_SLAM3) to compile successfully, and the process has many other problems that can be solved by following the error prompts.

YznMur, 2024-06-27:
@laxnpander @dubing-bit @lebronzzw  Hi.
I am trying to change vslam method. pulled  https://github.com/laxnpander/ORB_SLAM3 and https://github.com/laxnpander/ORB_SLAM2. but got the same error (see #94 ). tried to directly recompile g2o under ORBSLAM2/Thirdparty but still got the same error. any ideas? thanks in advance