laxnpander/OpenREALM #93: Cannot receive frame in OrthoRectification or Mosaicing Stage

Leonhard-Euler-ai, 2023-09-22:
Hello! Why **can't** the **ortho rectification stage or mosaicing stage** **sometimes** receive the frames published in the previous stage. 
**Errors and logs** are below.

### One case (ortho rectification lost and cannot receive remaining frames published by surface generation)

The thread of the **ortho rectification node** prints:

> [ERROR] [1695377284.147772077]: a message of over a gigabyte was predicted in tcpros. that seems highly unlikely, so I'll assume protocol synchronization is lost.

Then the key log is:
> Timing [Publish]: 0 ms

Surface generation stage log:

> 2023-09-21 22:54:36.575 (4850.093s) [        FFBD9700]         stage_base.cpp:104      0| { Stage [surface_generation] statistics
2023-09-21 22:54:36.575 (4850.093s) [        FFBD9700]         stage_base.cpp:105      0| .   Total frames: 136 
2023-09-21 22:54:36.575 (4850.093s) [        FFBD9700]         stage_base.cpp:106      0| .   Processed frames: 135 
2023-09-21 22:54:36.575 (4850.093s) [        FFBD9700]         stage_base.cpp:107      0| .   Bad frames: 0 
2023-09-21 22:54:36.575 (4850.093s) [        FFBD9700]         stage_base.cpp:108      0| .   Dropped frames: 0 
2023-09-21 22:54:36.575 (4850.093s) [        FFBD9700]         stage_base.cpp:109      0| .   Fps in: 0.00 
2023-09-21 22:54:36.575 (4850.093s) [        FFBD9700]         stage_base.cpp:110      0| .   Fps out: 0.00 
2023-09-21 22:54:36.575 (4850.093s) [        FFBD9700]         stage_base.cpp:104      0| } 0.000 s: Stage [surface_generation] statistics
2023-09-21 22:54:46.178 (4859.695s) [Stage [surface_g] surface_generation.cpp:71       0| Timing [Compute DSM]: 33456 ms
2023-09-21 22:54:46.178 (4859.695s) [Stage [surface_g] surface_generation.cpp:80       0| Timing [Container Add]: 0 ms
2023-09-21 22:54:46.178 (4859.695s) [Stage [surface_g] surface_generation.cpp:82       0| Publishing frame for next stage...
2023-09-21 22:54:46.178 (4859.696s) [Stage [surface_g] surface_generation.cpp:87       0| After publishing, buffer has 1 frames
2023-09-21 22:54:46.178 (4859.696s) [Stage [surface_g] surface_generation.cpp:88       0| **Timing [Publish]: 0 ms**
2023-09-21 22:54:46.184 (4859.702s) [Stage [surface_g] surface_generation.cpp:93       0| Timing [Saving]: 6 ms
2023-09-21 22:54:46.194 (4859.712s) [Stage [surface_g] worker_thread_base.cpp:84       0| Timing [Total]: 33472 ms
2023-09-21 22:54:46.195 (4859.712s) [Stage [surface_g] surface_generation.cpp:53       0| **Processing frame #281...**
2023-09-21 22:54:46.195 (4859.712s) [Stage [surface_g] surface_generation.cpp:197      0| Frame is accurate and keyframe. Checking for dense information...
2023-09-21 22:54:46.195 (4859.712s) [Stage [surface_g] surface_generation.cpp:66       0| Surface assumption: ELEVATION.
2023-09-21 22:54:46.575 (4860.093s) [        FFBD9700]         stage_base.cpp:104      0| { Stage [surface_generation] statistics
2023-09-21 22:54:46.575 (4860.093s) [        FFBD9700]         stage_base.cpp:105      0| .   Total frames: 136 
2023-09-21 22:54:46.575 (4860.093s) [        FFBD9700]         stage_base.cpp:106      0| .   Processed frames: 136 
2023-09-21 22:54:46.575 (4860.093s) [        FFBD9700]         stage_base.cpp:107      0| .   Bad frames: 0 
2023-09-21 22:54:46.575 (4860.093s) [        FFBD9700]         stage_base.cpp:108      0| .   Dropped frames: 0 
2023-09-21 22:54:46.575 (4860.093s) [        FFBD9700]         stage_base.cpp:109      0| .   Fps in: 0.00 
2023-09-21 22:54:46.575 (4860.093s) [        FFBD9700]         stage_base.cpp:110      0| .   Fps out: 0.00 
2023-09-21 22:54:46.575 (4860.093s) [        FFBD9700]         stage_base.cpp:104      0| } 0.000 s: Stage [surface_generation] statistics
2023-09-21 22:54:56.575 (4870.093s) [        FFBD9700]         stage_base.cpp:104      0| { Stage [surface_generation] statistics
2023-09-21 22:54:56.575 (4870.093s) [        FFBD9700]         stage_base.cpp:105      0| .   Total frames: 136 
2023-09-21 22:54:56.575 (4870.093s) [        FFBD9700]         stage_base.cpp:106      0| .   Processed frames: 136 
2023-09-21 22:54:56.575 (4870.093s) [        FFBD9700]         stage_base.cpp:107      0| .   Bad frames: 0 
2023-09-21 22:54:56.575 (4870.093s) [        FFBD9700]         stage_base.cpp:108      0| .   Dropped frames: 0 
2023-09-21 22:54:56.575 (4870.093s) [        FFBD9700]         stage_base.cpp:109      0| .   Fps in: 0.00 
2023-09-21 22:54:56.575 (4870.093s) [        FFBD9700]         stage_base.cpp:110      0| .   Fps out: 0.10 
2023-09-21 22:54:56.575 (4870.093s) [        FFBD9700]         stage_base.cpp:104      0| } 0.000 s: Stage [surface_generation] statistics
2023-09-21 22:55:06.575 (4880.093s) [        FFBD9700]         stage_base.cpp:104      0| { Stage [surface_generation] statistics
2023-09-21 22:55:06.575 (4880.093s) [        FFBD9700]         stage_base.cpp:105      0| .   Total frames: 136 
2023-09-21 22:55:06.575 (4880.093s) [        FFBD9700]         stage_base.cpp:106      0| .   Processed frames: 136 
2023-09-21 22:55:06.575 (4880.093s) [        FFBD9700]         stage_base.cpp:107      0| .   Bad frames: 0 
2023-09-21 22:55:06.575 (4880.093s) [        FFBD9700]         stage_base.cpp:108      0| .   Dropped frames: 0 
2023-09-21 22:55:06.576 (4880.093s) [        FFBD9700]         stage_base.cpp:109      0| .   Fps in: 0.00 
2023-09-21 22:55:06.576 (4880.093s) [        FFBD9700]         stage_base.cpp:110      0| .   Fps out: 0.00 
2023-09-21 22:55:06.576 (4880.093s) [        FFBD9700]         stage_base.cpp:104      0| } 0.000 s: Stage [surface_generation] statistics
2023-09-21 22:55:16.576 (4890.094s) [        FFBD9700]         stage_base.cpp:104      0| { Stage [surface_generation] statistics
2023-09-21 22:55:16.576 (4890.094s) [        FFBD9700]         stage_base.cpp:105      0| .   Total frames: 136 
2023-09-21 22:55:16.576 (4890.094s) [        FFBD9700]         stage_base.cpp:106      0| .   Processed frames: 136 
2023-09-21 22:55:16.576 (4890.094s) [        FFBD9700]         stage_base.cpp:107      0| .   Bad frames: 0 
2023-09-21 22:55:16.576 (4890.094s) [        FFBD9700]         stage_base.cpp:108      0| .   Dropped frames: 0 
2023-09-21 22:55:16.576 (4890.094s) [        FFBD9700]         stage_base.cpp:109      0| .   Fps in: 0.00 
2023-09-21 22:55:16.576 (4890.094s) [        FFBD9700]         stage_base.cpp:110      0| .   Fps out: 0.00 
2023-09-21 22:55:16.576 (4890.094s) [        FFBD9700]         stage_base.cpp:104      0| } 0.000 s: Stage [surface_generation] statistics
2023-09-21 22:55:17.763 (4891.281s) [Stage [surface_g] surface_generation.cpp:71       0| **Timing [Compute DSM]: 31568 ms**
2023-09-21 22:55:17.764 (4891.281s) [Stage [surface_g] surface_generation.cpp:80       0| Timing [Container Add]: 0 ms
2023-09-21 22:55:17.764 (4891.281s) [Stage [surface_g] surface_generation.cpp:82       0| **Publishing frame for next stage...**
2023-09-21 22:55:17.764 (4891.281s) [Stage [surface_g] surface_generation.cpp:87       0| After publishing, buffer has 0 frames
2023-09-21 22:55:17.764 (4891.281s) [Stage [surface_g] surface_generation.cpp:88       0| **Timing [Publish]: 0 ms**
2023-09-21 22:55:17.769 (4891.287s) [Stage [surface_g] surface_generation.cpp:93       0| Timing [Saving]: 5 ms
[... 111 lines cut ...]
2023-09-22 17:43:15.447 ( 450.019s) [        FBDE5700]         stage_base.cpp:105      0| .   Total frames: 41 
2023-09-22 17:43:15.447 ( 450.020s) [        FBDE5700]         stage_base.cpp:106      0| .   Processed frames: 41 
2023-09-22 17:43:15.447 ( 450.020s) [        FBDE5700]         stage_base.cpp:107      0| .   Bad frames: 0 
2023-09-22 17:43:15.447 ( 450.020s) [        FBDE5700]         stage_base.cpp:108      0| .   Dropped frames: 0 
2023-09-22 17:43:15.447 ( 450.020s) [        FBDE5700]         stage_base.cpp:109      0| .   Fps in: 0.10 
2023-09-22 17:43:15.447 ( 450.020s) [        FBDE5700]         stage_base.cpp:110      0| .   Fps out: 0.10 
2023-09-22 17:43:15.447 ( 450.020s) [        FBDE5700]         stage_base.cpp:104      0| } 0.000 s: Stage [ortho_rectification] statistics

Mosaicing stage log:
> 2023-09-22 18:03:02.657 (  60.246s) [        C0C2F700]         stage_base.cpp:104      0| { Stage [mosaicing] statistics
2023-09-22 18:03:02.657 (  60.246s) [        C0C2F700]         stage_base.cpp:105      0| .   Total frames: 0 
2023-09-22 18:03:02.657 (  60.246s) [        C0C2F700]         stage_base.cpp:106      0| .   Processed frames: 0 
[... cut ...]
2023-09-22 18:03:12.657 (  70.246s) [        C0C2F700]         stage_base.cpp:107      0| .   Bad frames: 0 
2023-09-22 18:03:12.657 (  70.246s) [        C0C2F700]         stage_base.cpp:108      0| .   Dropped frames: 0 
2023-09-22 18:03:12.657 (  70.246s) [        C0C2F700]         stage_base.cpp:109      0| .   Fps in: 0.00 
2023-09-22 18:03:12.657 (  70.246s) [        C0C2F700]         stage_base.cpp:110      0| .   Fps out: 0.00 
2023-09-22 18:03:12.657 (  70.246s) [        C0C2F700]         stage_base.cpp:104      0| } 0.000 s: Stage [mosaicing] statistics

Leonhard-Euler-ai, 2023-09-22:
I need to publish a large orthoimage. How can I publish big data package?

laxnpander, 2023-10-13:
@Leonhard-Euler-ai That is a question that you should ask the ROS community. Not sure what goes wrong. 63 MPix sounds quite excessive for an incremental ortho update. The pipeline was not yet tested with such an image size.

laxnpander, 2023-11-23:
Closing until reopen.

closed 2023-11-23 by laxnpander