laxnpander/OpenREALM #94: Building g2o from installable ORB_SLAM2 error

YznMur, 2023-11-04:
Hi, I am trying to change SLAM method to ORB_SLAM2, I pulled your installable version: https://github.com/laxnpander/ORB_SLAM2 and tried to build it with ORB_SLAM2/build.sh and faced this error:
```
root@yazan-System-Product-Name:/home/trainer/ORB_SLAM2/Thirdparty/g2o/build# make -j4
[  3%] Building CXX object CMakeFiles/ORB_SLAM2_g2o.dir/g2o/types/types_sba.cpp.o
[  6%] Building CXX object CMakeFiles/ORB_SLAM2_g2o.dir/g2o/types/types_six_dof_expmap.cpp.o
[  9%] Building CXX object CMakeFiles/ORB_SLAM2_g2o.dir/g2o/types/types_seven_dof_expmap.cpp.o
[ 12%] Building CXX object CMakeFiles/ORB_SLAM2_g2o.dir/g2o/core/hyper_graph_action.cpp.o
In file included from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:37:0,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.cpp:27:
/home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/openmp_mutex.h:30:10: fatal error: ../../config.h: No such file or directory
 #include "../../config.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:37:0,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_six_dof_expmap.h:37,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_six_dof_expmap.cpp:27:
/home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/openmp_mutex.h:30:10: fatal error: ../../config.h: No such file or directory
 #include "../../config.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:37:0,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_sba.h:30,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_sba.cpp:27:
/home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/openmp_mutex.h:30:10: fatal error: ../../config.h: No such file or directory
 #include "../../config.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/core/optimizable_graph.h:37:0,
                 from /home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/core/hyper_graph_action.cpp:28:
/home/trainer/ORB_SLAM2/Thirdparty/g2o/g2o/core/openmp_mutex.h:30:10: fatal error: ../../config.h: No such file or directory
 #include "../../config.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/ORB_SLAM2_g2o.dir/build.make:103: recipe for target 'CMakeFiles/ORB_SLAM2_g2o.dir/g2o/types/types_seven_dof_expmap.cpp.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2_g2o.dir/g2o/types/types_seven_dof_expmap.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/ORB_SLAM2_g2o.dir/build.make:89: recipe for target 'CMakeFiles/ORB_SLAM2_g2o.dir/g2o/types/types_six_dof_expmap.cpp.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2_g2o.dir/g2o/types/types_six_dof_expmap.cpp.o] Error 1
CMakeFiles/ORB_SLAM2_g2o.dir/build.make:75: recipe for target 'CMakeFiles/ORB_SLAM2_g2o.dir/g2o/types/types_sba.cpp.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2_g2o.dir/g2o/types/types_sba.cpp.o] Error 1
CMakeFiles/ORB_SLAM2_g2o.dir/build.make:117: recipe for target 'CMakeFiles/ORB_SLAM2_g2o.dir/g2o/core/hyper_graph_action.cpp.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2_g2o.dir/g2o/core/hyper_graph_action.cpp.o] Error 1
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/ORB_SLAM2_g2o.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2_g2o.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2
```
Any ideas?
by the way I didnt install g2o from ```OpenREALM/tools/install_deps.sh```, should I?
Thanks

laxnpander, 2023-11-23:
@YznMur Yes, you should use the G2O version shipped with Orb slam. I think they modified some parts, so it might not be compatible.

YznMur, 2023-11-27:
@laxnpander sorry I am a bit confused
should I clone this orbslam2 version (uninstallable) ``` https://github.com/raulmur/ORB_SLAM2``` or your one (installable) ```https://github.com/laxnpander/ORB_SLAM2```?

laxnpander, 2024-04-12:
That is a long waiting time. You should install the one from my GitHub.

laxnpander, 2024-05-05:
Reopen if necessary.

closed 2024-05-05 by laxnpander

YznMur, 2024-06-27:
Hi @laxnpander .
I am using your modified orbslam2 and orbslam3 from the start. still getting the same error:

```root@yazan-System-Product-Name:/home/trainer/ORB_SLAM3/Thirdparty/g2o/build# make -j4
[  6%] Building CXX object CMakeFiles/ORB_SLAM3_g2o.dir/g2o/types/types_sba.cpp.o
[  6%] Building CXX object CMakeFiles/ORB_SLAM3_g2o.dir/g2o/types/types_six_dof_expmap.cpp.o
[  9%] Building CXX object CMakeFiles/ORB_SLAM3_g2o.dir/g2o/types/types_seven_dof_expmap.cpp.o
[ 12%] Building CXX object CMakeFiles/ORB_SLAM3_g2o.dir/g2o/core/hyper_graph_action.cpp.o
In file included from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:37:0,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.cpp:27:
/home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/openmp_mutex.h:30:10: fatal error: ../../config.h: No such file or directory
 #include "../../config.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:37:0,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_six_dof_expmap.h:37,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_six_dof_expmap.cpp:27:
/home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/openmp_mutex.h:30:10: fatal error: ../../config.h: No such file or directory
 #include "../../config.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:37:0,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_sba.h:30,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_sba.cpp:27:
/home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/openmp_mutex.h:30:10: fatal error: ../../config.h: No such file or directory
 #include "../../config.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/core/optimizable_graph.h:37:0,
                 from /home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/core/hyper_graph_action.cpp:28:
/home/trainer/ORB_SLAM3/Thirdparty/g2o/g2o/core/openmp_mutex.h:30:10: fatal error: ../../config.h: No such file or directory
 #include "../../config.h"
          ^~~~~~~~~~~~~~~~
```

Any ideas?
Thanks in advance