laxnpander/OpenREALM #95: roslaunch realm_ros failed

ChoiiYill, 2023-11-18:
When I try : **roslaunch realm_ros alexa_noreco.launch**
And then it prompts：
RLException: [alexa_noreco.launch] is neither a launch file in package [realm_ros] nor is [realm_ros] a launch file name
The traceback for the exception was written to the log file
**What do I need to do to fix this?**

ChoiiYill, 2023-11-19:
> Hello, when I try : **roslaunch realm_ros alexa_noreco.launch** And I encounter with this： RLException: [alexa_noreco.launch] is neither a launch file in package [realm_ros] nor is [realm_ros] a launch file name The traceback for the exception was written to the log file **How to fix this? Thanks.**

Sorry, I am a fledgling of ROS develop.
I have solved this problem by setting environment variables. 
1. Open bashrc by: **sudo vim ~/.bashrc**
Add this two lines to the end:
_source /home/pc/app/OpenREALM/catkin_ws/devel/setup.sh
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/pc/app/OpenREALM/catkin_ws/_
2. Save and restart bashrc: **source ~/.bashrc**

closed 2023-11-19 by ChoiiYill

hfwuyanzu666, 2023-11-19:
Do you have the following errors when running nocore.launch and core.launch?
[realm_pose_estimation-3] process has died [pid 16691, exit code -11,  cmd /home/yanzu/openrealm/openrealm_ros_bridge/catkin_ws/devel/lib/realm_ros/realm_stage_node  __name:=realm_pose_estimation  __log:=/home/yanzu/.ros/log/8156f674-86f0-11ee-b3fc-d6499dadd82b/realm_pose_estimation-3.log].
log file: /home/yanzu/.ros/log/8156f674-86f0-11ee-b3fc-d6499dadd82b/realm_pose_estimation-3*.log

I didn't know how to solve it. I tried similar problems and solutions that other people had come up with before and it didn't work. I would appreciate it if you know how to solve it

ChoiiYill, 2023-11-20:
> Do you have the following errors when running nocore.launch and core.launch? [realm_pose_estimation-3] process has died [pid 16691, exit code -11, cmd [...]

Yes, I've met this problem when I use Ubuntu20.04, OpenCV-4. I solve it by using Ubuntu18.04 and OpenCV-3.4.
I think it might cause by **cv_bridge**, which you can refer to this issue. (issue #22 ) https://github.com/laxnpander/OpenREALM/issues/22#issuecomment-1151097095

hfwuyanzu666, 2023-12-05:
Thank you very much for your valuable answer. I went to deal with other things in the middle of the project, which delayed the project for a while. I tried to match the cv_bridge version with the opencv version, and the running result was still the same as before. The [realm_pose_estimation-3] process has died [pid 10931, exit code 1] error still occurs.
I let it run all the way down to the state in the image below and nothing else changed
[image]

hfwuyanzu666, 2023-12-05:
The following is where the program is reporting errors
[image]