laxnpander/OpenREALM #96: An issue in roslaunch realm_ros noreco and reco

hfwuyanzu666, 2023-11-19:
Thanks for maintaining the OpenRealm project. When I ran gnss.launch, it was normal, but when I ran nocore.launch and core.launch, the following error occurred:
[realm_pose_estimation-3] process has died [pid 16691, exit code -11,  cmd /home/yanzu/openrealm/openrealm_ros_bridge/catkin_ws/devel/lib/realm_ros/realm_stage_node  __name:=realm_pose_estimation  __log:=/home/yanzu/.ros/log/8156f674-86f0-11ee-b3fc-d6499dadd82b/realm_pose_estimation-3.log].
log file: /home/yanzu/.ros/log/8156f674-86f0-11ee-b3fc-d6499dadd82b/realm_pose_estimation-3*.log
I didn't know how to solve it. I tried similar problems and solutions that other people had come up with before and it didn't work.

laxnpander, 2023-11-23:
@hfwuyanzu666 This debug information is not sufficient to help you. I need more, either full terminal output or a stack trace.

hfwuyanzu666, 2023-12-05:
Thank you very much for your valuable answer. I went to deal with other things in the middle of the project, which delayed the project for a while. I tried to match the cv_bridge version with the opencv version, and the running result was still the same as before. The [realm_pose_estimation-3] process has died [pid 10931, exit code 1] error still occurs.
I let it run all the way down to the state in the image below and nothing else changed
[image]
The following is where the program is reporting errors
[image]

hfwuyanzu666, 2023-12-06:
Thank you very much for your maintenance. This is my complete terminal output when running the roslaunch realm_ros alexa_noreco.launch command. Looking forward to your reply

hfwuyanzu666, 2023-12-06:
Thank you very much for your maintenance. This is my complete terminal output when running the roslaunch realm_ros alexa_noreco.launch command. Looking forward to your reply！
[output.txt](https://github.com/laxnpander/OpenREALM/files/13577171/output.txt)

laxnpander, 2023-12-07:
@hfwuyanzu666 Hmm, hard to tell. It looks like the pose_estimation is not processing anything. Can you check via ```rostopic hz /realm/alexa/img``` if there is data coming out of the exiv_grabber_node?

hfwuyanzu666, 2023-12-08:
> @hfwuyanzu666 Hmm, hard to tell. It looks like the pose_estimation is not processing anything. Can you check via `rostopic hz /realm/alexa/img` if there is data coming out of the exiv_grabber_node?

Thank you very much for your reply. Now I have solved the problem that “the [realm_pose_estimation-3] process has died“. Encountering new problems is the same as Issues [#71.](https://github.com/laxnpander/OpenREALM/issues/71) I'm currently trying to run openvslam alone to see if it's a problem with openvslam.

000skysky000, 2023-12-12:
@hfwuyanzu666 I encountered the same issue. Have you managed to resolve it? I'm attempting to rerun it. I'd appreciate it if you could share how you solved it.

1005452649, 2023-12-13:
@hfwuyanzu666 hello, I encountered the same issue. Have you managed to resolve it? I'm attempting to rerun it. I'd appreciate it if you could share how you solved it.

hfwuyanzu666, 2023-12-14:
Sorry, I haven't solved it yet, but we can add a contact way for more communication and solve the problem together. My QQ is 2690405895

normal-crayon, 2024-02-22:
Hi @laxnpander and @hfwuyanzu666 Ive encountered with the same issue. have you managed to resolve it by any chance ?

normal-crayon, 2024-02-22:
#67

normal-crayon, 2024-02-22:
@67

normal-crayon, 2024-02-22:
I think the #67 is the solution... Changing the vocabulary is atleast not hanging my system. But now i am encountering this error 
```[ INFO] [1708624186.053782911]: STAGE_NODE [surface_generation]: Received output directory, set to:
        /catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/output/24-02-22_17-49-45
[ INFO] [1708624186.054274057]: STAGE_NODE [mosaicing]: Detected stage type: 'mosaicing'
2024-02-22 17:49:46.056 (   0.462s) [Stage [pose_esti]    pose_estimation.cpp:278   WARN| No tracking.
2024-02-22 17:49:46.056 (   0.462s) [Stage [pose_esti]    pose_estimation.cpp:174      0| Timing [Tracking]: 45 ms
2024-02-22 17:49:46.056 (   0.462s) [Stage [pose_esti] worker_thread_base.cpp:84       0| Timing [Total]: 45 ms
[ INFO] [1708624186.057217756]: STAGE_NODE [mosaicing]: : Loading camera from path:
        /catkin_ws/src/OpenREALM_ROS1_Bridge/realm_ros/profiles/alexa_noreco/camera/calib.yaml
```

laxnpander, 2024-03-30:
@normal-crayon I don't see any error? Looks okay to me? Just not tracking for whatever reason.

laxnpander, 2024-05-05:
Reopen if issues persists.

closed 2024-05-05 by laxnpander

hfwuyanzu666, 2024-09-20:
> > @hfwuyanzu666 Hmm, hard to tell. It looks like the pose_estimation is not processing anything. Can you check via `rostopic hz /realm/alexa/img` if there is data coming out of the exiv_grabber_node?
> 
> Thank you very much for your reply. Now I have solved the problem that “the [realm_pose_estimation-3] process has died“. Encountering new problems is the same as Issues [#71.](https://github.com/laxnpander/OpenREALM/issues/71) I'm currently trying to run openvslam alone to see if it's a problem with openvslam. output.txt is what the terminal displays.The following picture is the error displayed by the terminal. [output.txt](https://github.com/laxnpander/OpenREALM/files/13610422/output.txt) ![2023-12-08 14-36-22 的屏幕截图](https://private-user-images.githubusercontent.com/105181434/288985845-6c7dc7d1-afaa-44c4-aa25-d871723c1bd7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY4MTg3NTEsIm5iZiI6MTcyNjgxODQ1MSwicGF0aCI6Ii8xMDUxODE0MzQvMjg4OTg1ODQ1LTZjN2RjN2QxLWFmYWEtNDRjNC1hYTI1LWQ4NzE3MjNjMWJkNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMFQwNzQ3MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMDkxY2NjMmZmZjI1NzVhM2NhZDllYzExNWM4ZTZkYWRjZGU2NzZkNTdmMGVlMGZlODI0MjI4MTRjNGMzOTM4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.qqKnNil8m8tQ--IreXleiy2SO_jBLf4MJjLduGfZBLs)

hello