laxnpander/OpenREALM #97: My own dataset

zhangjd1029, 2023-12-05:
Thank you very much for your work.

Does OpenRealm have to run data with GPS information? I have some unmanned aerial vehicle high-altitude images taken by myself, but they do not contain GPS information. Can they be operated?

If GPS data is necessary, can the code be modified to run without relying on GPS data?

Looking forward to your reply。

laxnpander, 2023-12-07:
I don't think it can. GNSS is used to estimate the scale and properly project the images. It could work in theory, but practically the pipeline is not designed around such a case. Are you sure your images have no GNSS meta tags? Usually they are hidden in the jpegs. DJI for example shouldn't be an issue to process.

laxnpander, 2024-03-30:
Without further information there is nothing I can help with here.

closed 2024-03-30 by laxnpander