laxnpander/OpenREALM #98: A question about how to get the georeference

9435LV, 2023-12-17:
Hello, I greatly admire your ability to do such a great job.

I would like to ask you, if I want to create my own dataset, how do I obtain the georeference parameter in the georeference.yaml. 
Or rather, can I use Alexa without changing this parameter_ Use reco. launch to run my dataset

laxnpander, 2024-01-20:
Hey, georeference is computed from the GNSS information in the image meta tags. If those are not provided, there is no georeference. There was once a function to provide an external reference through the georeference.yaml, but this is not currently maintained.

9435LV, 2024-01-24:
It is an honor for you to reply to my message. Do you mean that the  #georeference.yaml file cannot truly function in the program?
georeference.yaml 
%YAML:1.0
transformation_w2g: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 85.147685, 37.630008, -2.445093, 583499.057912, 37.462956, -85.102441, -5.121097, 5806387.668580, -4.303824, 3.698813, -92.951171, 97.517175, 0.0, 0.0, 0.0, 1.0 ]

laxnpander, 2024-01-24:
No, it's something I used during my thesis. It's a feature that might not be present anymore. Georeferencing still works using the exiv tags of the images, but you cannot provide a known georeference through the yaml.

9435LV, 2024-01-28:
Thank you very much for your answer. It has solved my question and I sincerely appreciate your outstanding work. In addition, the Chinese New Year is approaching, and I wish you a happy new year in advance.  :)

Aishop119, 2024-02-12:
Hello, what is the effect of your implementation on your own dataset? Is the performance achieved comparable to the dataset collected by the author?

laxnpander, 2024-03-30:
@Aishop119 Open an issue if you have a specific question.

closed 2024-03-30 by laxnpander