laxnpander/OpenREALM #101: Test dataset - OrthoMosaic output

marleyshan21, 2024-01-23:
Hi!

I was running the test dataset in 2D mapping with visual SLAM mode. 
I am seeing disparity in the results when compared to the video of OpenRealm.
[image]
[image]

I tried playing around primarily with nrof_features for the orb extractor and ground sampling distance.
What are the other main parameters that i need to tune? 
The images are for the default parameters that are present in the repo. 

@laxnpander

laxnpander, 2024-01-24:
@marleyshan21 I'd rather load the output gtiff into apps like QGIS and check there. I am pretty sure some of the static transformers is publishing a wrong transformation or the satellite map uses a wrong transformation. The output data is unaffected of that, just a visualisation issue. Not sure if I find the time fix it.

marleyshan21, 2024-01-24:
Ah okay ✌️!
I will check it out on QGIS.

I will check out this static transform part. 
If you could guide me on where this static transform part exists in the code, i could check it out and try fixing it. 

Lemme know!

laxnpander, 2024-01-25:
If it's on OpenREALM side it is somewhere here:

https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/master/realm_ros/src/realm_ros_lib/stage_node.cpp

marleyshan21, 2024-01-25:
Using your fork of rviz_satellite solved it!
https://github.com/laxnpander/rviz_satellite

[image]

Based on your commit, it looks like an issue with the frame being lookedup to determine the transformation. Your fork uses the one needed by OpenRealm

Thanks for the help @laxnpander!

closed 2024-01-25 by marleyshan21

laxnpander, 2024-01-25:
Sure, glad I could help.