laxnpander/OpenREALM #102: The performance is not very satisfactory when using my own dataset.

Aishop119, 2024-02-03:
Hello, thank you very much for your outstanding contribution. I would like to ask, after adding latitude, longitude, and altitude information exiv2 tags to the images in my own dataset, and modifying the camera parameter configuration in calib.yaml, what other modifications do I need to make? 
Currently, I have only made these changes, but the results are not satisfactory. 
I am looking forward to your reply, thank you.

Aishop119, 2024-02-21:
I use the DJI Phantom 4 RTK with a frame rate set at 10fps and a frame size of 1920x1080. The front and rear overlap rate is 0.995, and the left and right overlap rate is 0.78. The shooting is conducted on a farm.
[image]

Aishop119, 2024-02-21:
And the mosaic result as this
[image]
[image]

laxnpander, 2024-03-30:
@Aishop119 Looks like tracking was not really good. I've seen similar behaviour in other datasets and usually this happens when the camera calibration is not good. How did you calibrate your setup? Do you think you can improve it?

laxnpander, 2024-05-05:
Closed until reopen.

closed 2024-05-05 by laxnpander