laxnpander/OpenREALM #105: No tracking on my dataset

lyy-1997, 2024-02-23:
Thanks a lot for your work.
I have successfully run gnss,noreo and reco using both the provided test data set and my own (**1014x760 resolution, 126m flight altitude**). However, when I used a dataset with a resolution of **6144x4096 and a flight altitude of 171m**, I was never able to complete 3D mapping.Even if the camera parameters are perfectly correct, there are still a few things that are strange.
1.No tracking has been displayed since the beginning.
2.Even though the image has obvious features, there are almost no feature points in rviz.
Since I have previously successfully run the data set of 1014x760 resolution (the resolution is similar to the test data set provided), I am guessing that the resolution of 6144x4096 May be causing the problem? Or was it flight altitude?If so, how should I configure it correctly?THANKS!
By the way, I only changed the camera parameter configuration in profile/*/camera/calib.yaml, and the resolution Settings were consistent with the image size.

[image]

SUMMARY
========

PARAMETERS
 * /realm_densification/config/id: alexa
 * /realm_densification/config/method: dummy
 * /realm_densification/config/profile: SHRA_reco
 * /realm_densification/stage/type: densification
 * /realm_densification/topics/input: /realm/alexa/pose...
 * /realm_densification/topics/output: /realm/alexa/dens...
 * /realm_exiv2_grabber/config/id: alexa
 * /realm_exiv2_grabber/config/input: /home/ubuntu/data...
 * /realm_exiv2_grabber/config/profile: SHRA_reco
 * /realm_exiv2_grabber/config/rate: 0.5
 * /realm_mosaicing/config/id: alexa
 * /realm_mosaicing/config/profile: SHRA_reco
 * /realm_mosaicing/stage/type: mosaicing
 * /realm_mosaicing/topics/input: /realm/alexa/orth...
 * /realm_ortho_rectification/config/id: alexa
 * /realm_ortho_rectification/config/profile: SHRA_reco
 * /realm_ortho_rectification/stage/type: ortho_rectification
 * /realm_ortho_rectification/topics/input: /realm/alexa/surf...
 * /realm_ortho_rectification/topics/output: /realm/alexa/orth...
 * /realm_pose_estimation/config/id: alexa
 * /realm_pose_estimation/config/method: orb_slam2
 * /realm_pose_estimation/config/profile: SHRA_reco
 * /realm_pose_estimation/stage/master: True
 * /realm_pose_estimation/stage/type: pose_estimation
 * /realm_pose_estimation/topics/input: /realm/alexa/input
 * /realm_pose_estimation/topics/output: /realm/alexa/pose...
 * /realm_surface_generation/config/id: alexa
 * /realm_surface_generation/config/profile: SHRA_reco
 * /realm_surface_generation/stage/type: surface_generation
 * /realm_surface_generation/topics/input: /realm/alexa/dens...
 * /realm_surface_generation/topics/output: /realm/alexa/surf...
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    realm_densification (realm_ros/realm_stage_node)
    realm_exiv2_grabber (realm_ros/realm_exiv2_grabber)
    realm_mosaicing (realm_ros/realm_stage_node)
    realm_ortho_rectification (realm_ros/realm_stage_node)
    realm_pose_estimation (realm_ros/realm_stage_node)
    realm_surface_generation (realm_ros/realm_stage_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [9005]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bc3e05e4-d1f9-11ee-9032-24418c641e46
process[rosout-1]: started with pid [9865]
started core service [/rosout]
/opt/ros/melodic/lib/python2.7/dist-packages/roslib/packages.py:470: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if resource_name in files:
process[realm_exiv2_grabber-2]: started with pid [9929]
process[realm_pose_estimation-3]: started with pid [9957]
process[realm_densification-4]: started with pid [9981]
process[realm_surface_generation-5]: started with pid [10000]
[... 404 lines cut ...]
[realm_pose_estimation-3] killing on exit
[realm_exiv2_grabber-2] killing on exit
2024-02-23 11:16:03.725 (  59.996s) [        6A2D39C0] worker_thread_base.cpp:124      0| Thread 'Stage [surface_generation]' received finish request...
2024-02-23 11:16:03.733 (  59.980s) [        AA27A9C0] worker_thread_base.cpp:124      0| Thread 'Stage [ortho_rectification]' received finish request...
2024-02-23 11:16:03.734 (  60.026s) [        59149700]         stage_base.cpp:122      0| FPS in: 0.500000, out: 0.000000
2024-02-23 11:16:03.738 (  60.030s) [        976269C0] worker_thread_base.cpp:124      0| Thread 'Stage [pose_estimation]' received finish request...
2024-02-23 11:16:03.740 (  60.026s) [        26D659C0] worker_thread_base.cpp:124      0| Thread 'Stage [densification]' received finish request...
2024-02-23 11:16:03.744 (  60.030s) [        E9089700]         stage_base.cpp:122      0| FPS in: 0.000000, out: 0.000000
2024-02-23 11:16:03.746 (  59.980s) [        F02679C0] worker_thread_base.cpp:124      0| Thread 'Stage [mosaicing]' received finish request...
2024-02-23 11:16:03.753 (  60.039s) [        E8888700] worker_thread_base.cpp:96       0| Thread 'Stage [densification]' finished!
2024-02-23 11:16:03.754 (  60.025s) [        1FDE5700]         stage_base.cpp:122      0| FPS in: 0.000000, out: 0.000000
2024-02-23 11:16:03.757 (  60.029s) [        1F5E4700] worker_thread_base.cpp:96       0| Thread 'Stage [surface_generation]' finished!
2024-02-23 11:16:03.770 (  60.017s) [        5FBD9700]         stage_base.cpp:122      0| FPS in: 0.000000, out: 0.000000
2024-02-23 11:16:03.776 (  60.023s) [        5F3D8700] worker_thread_base.cpp:96       0| Thread 'Stage [ortho_rectification]' finished!
2024-02-23 11:16:05.499 (  61.791s) [        10ED6700]    pose_estimation.cpp:192   WARN| No tracking.
2024-02-23 11:16:05.536 (  61.828s) [        10ED6700] worker_thread_base.cpp:91       0| Thread 'Stage [pose_estimation]' has processed data. Time elapsed: 2.32 [s]
2024-02-23 11:16:07.114 (  63.406s) [        116D7700]    pose_estimation.cpp:514      0| Publishing pose of frame #26...
2024-02-23 11:16:07.536 (  63.828s) [        10ED6700] worker_thread_base.cpp:96       0| Thread 'Stage [pose_estimation]' finished!
2024-02-23 11:16:07.536 (  63.828s) [        976269C0] worker_thread_base.cpp:124      0| Thread 'Publisher [pose_estimation]' received finish request...
2024-02-23 11:16:09.129 (  65.421s) [        116D7700] worker_thread_base.cpp:96       0| Thread 'Publisher [pose_estimation]' finished!
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

laxnpander, 2024-03-30:
@lyy-1997 Hey, welcome to the dark arts of tuning OpenREALM. I am pretty sure your image resolution is way too high.I'd start with significantly smaller images. You can resize them automatically using the configs in the respective profile, e.g.

https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/65a86d465b5ba0b5256f5c8f1bb8090ec3a027f8/realm_ros/profiles/alexa_reco/pose_estimation/method/orb_slam3_settings.yaml#L6

So set resizing to 0.2 and see if the results improve. You can also increase nrof features if you have the processing power to back it up.

lyy-1997, 2024-04-01:
Thanks for your reply.
I've switched to a lower resolution dataset and got the desired results. So I think it's the resolution of the dataset.
When I set resizing to 0.2, do I need to adjust the camera parameters in calib.yam for the dataset with a high resolution? If so, how should I set the camera parameters?I guese, keep other parameters unchanged, only adjust the width and height of the camera parameters to 0.2 times the original?
By the way, does the current version successfully support ORBSLAM3, because I saw that you mentioned ORBSLAM3 in your reply?

laxnpander, 2024-05-05:
Reopen if issues persists.

closed 2024-05-05 by laxnpander

gityeezy, 2024-07-28:
你好 @lyy-1997 ，你自己录制的数据集是怎么做的，你的图片制作好后，除了照片分辨率外，代码中还需要修改哪些内容？