laxnpander/OpenREALM #109: Question about 2.5D mapping in alexa_reco.launch

ZackStone233, 2024-04-29:
Hello, thank you for maintaining OpenREALM. I noticed that when running alexa_reco.launch, Rviz still displays a 2D map. I found on your forum that you mentioned temporarily suspending the release of 3D grids. Therefore, I would like to ask if you could provide some guidance on publishing and visualizing 3D grids?

ZackStone233, 2024-05-09:
Thank you very much for your response. I will try to debug that part of the code accordingly.

ZackStone233, 2024-05-14:
Hello,

I would like to inquire once again regarding the visualization of 2.5D mapping in the `alexa_reco.launch` file within Rviz. Could you please specify which plugin is utilized for this visualization? Additionally, may I know which topic the visualization data is sourced from?

laxnpander, 2024-05-14:
@ZackStone233: According to this line it is a marker in ROS, if you mean the mesh:

https://github.com/laxnpander/OpenREALM_ROS1_Bridge/blob/65a86d465b5ba0b5256f5c8f1bb8090ec3a027f8/realm_ros/src/realm_ros_lib/stage_node.cpp#L205

ZackStone233, 2024-05-17:
Thank you very much for your response!

closed 2024-08-05 by laxnpander