laxnpander/OpenREALM #121: My own datasets

IgnorAnsel, 2025-07-29:
​​Why Does My Data Show This Behavior?​​

[image]

WhovianZyh, 2026-03-27:
> [@WhovianZyh](https://github.com/WhovianZyh) Hello, first of all, I would like to thank you very much for your dataset. I would like to ask if the camera intrinsic parameters of this dataset need to be modified? If necessary, could you please provide the intrinsic parameter data of the camera for this dataset? My email address is [john5541107@163.com](mailto:john5541107@163.com)

Regarding the camera intrinsic parameters: I don't know if they need to be modified either. I haven't changed any other configuration files in the project — I just used this dataset with the default settings.

OrduVictor, 2026-04-01:
@John554590 One thing i noticed when using @WhovianZyh dataset for the noreco test was that in the initial mosaiced images, say about after 15 images have been stitched, the pipeline seems to perform quite accurately, although the pose estimation is very wobbly. The errors begin when the aircraft turns, yawing motion, producing the warped effect on the lower part of the final mosaic. Afterwards, everything just falls apart.

About the camera calibration, I verify my camera intrinsic parameters are accurate by running perspective transformations on a number of image pairs. I tried reducing my dataset resolution 4032x3024 by a scale factor of 0.2, but the SLAM stage still basically is all NO tracking warnings. The gnss test performs horribly on my dataset too, similar to the result you have.